lm-sensors/lm-sensors #298: Gigabyte B550M PRO-P, missing fan information.

Stmrllr, 2020-12-12:
Arch Linux, latest kernel (5.9.13), AMD 3900X, Gigabyte B550M PRO-P, 2x16GB Corsair Vengeance 3200MHZ

Everything was working fine with my last motherboard (MSI B450i Gaming Plus AC), but I can't get any motherboard fan RPMs to show. I think I'm missing something else too from the output. I ran sensors-detect as sudo and first tried with default settings, and then 'yes/YES' to all. This is my output:

`k10temp-pci-00c3
Adapter: PCI adapter
Vcore:         1.42 V  
Vsoc:          1.09 V  
Tctl:         +61.1 C  
Tdie:         +61.1 C  
Tccd1:        +41.5 C  
Tccd2:        +42.0 C  
Icore:        16.00 A  
Isoc:         13.50 A  

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8 C  (crit = +20.8 C)

amdgpu-pci-0a00
Adapter: PCI adapter
vddgfx:      693.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3200 RPM)
edge:         +46.0 C  (crit = +110.0 C, hyst = -273.1 C)
                       (emerg = +115.0 C)
junction:     +46.0 C  (crit = +105.0 C, hyst = -273.1 C)
                       (emerg = +110.0 C)
mem:           +0.0 C  (crit = +105.0 C, hyst = -273.1 C)
                       (emerg = +110.0 C)
power1:       11.00 W  (cap = 130.00 W)

nvme-pci-0100
Adapter: PCI adapter
Composite:    +37.9 C  (low  =  -0.1 C, high = +74.8 C)
                       (crit = +79.8 C)
`

karypid, 2020-12-20:
I have the Gigabyte B450M DSH3 (rev 1.0) and am getting the same behaviour. Should I raise a separate issue, or would they be using the same sensor chip? How can we help identify what the name of the chip is? Perhaps booting into Windows and looking at device manager? Where would we go about finding that information?

Stmrllr, 2020-12-20:
I actually got it working. I followed instructions from https://wiki.archlinux.org/index.php/lm_sensors#Gigabyte_B250/Z370/B450M_motherboards, only difference is I used "force_id=0x8628".

I'm still missing coretemps though. Bashtop shows same temp for every core, so it's not lm_sensors related anyways.

abucodonosor, 2020-12-21:
@Stmrllr 

There is no 'coretemp' for Ryzen CPUs, that is for Intel.

Your CPU/CCD temps are there.