lm-sensors/lm-sensors #300: x570 Aorus Master / Ryzen 9 5900x / RTX 3080

mitjans, 2020-12-20:
Hi! Just finished building my first rig, but after running `sensors-detect` as root, the output of `sensors` give me:

```
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:            N/A

k10temp-pci-00c3
Adapter: PCI adapter
temp1:         +0.0°C  (high = +70.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)

it8792-isa-0a60
Adapter: ISA adapter
in0:           1.78 V  (min =  +0.00 V, max =  +2.78 V)
in1:         600.00 mV (min =  +0.00 V, max =  +2.78 V)
in2:         992.00 mV (min =  +0.00 V, max =  +2.78 V)
+3.3V:         1.68 V  (min =  +0.00 V, max =  +2.78 V)
in4:           1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:           1.18 V  (min =  +0.00 V, max =  +2.78 V)
in6:           2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:          1.67 V  (min =  +0.00 V, max =  +2.78 V)
Vbat:          1.55 V
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +39.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

nvme-pci-0100
Adapter: PCI adapter
Composite:    +47.9°C  (low  =  -5.2°C, high = +83.8°C)
                       (crit = +87.8°C)
```

Why are the fans RPM 0? Where's the temp info for my cpu cores? Can lm_sensors detect nvidia gpu temps and fans?

Thanks!

abucodonosor, 2020-12-21:
> Where's the temp info for my cpu cores?

You'll need kernel >= 5.10 for that to show up.

DrissiReda, 2020-12-25:
I just updated my machine to 5.10.2 and I still can't see my cpu temps:

My setup:

Msi MAG B550M mortar
ryzen 5800X
nvidia GT420

Ignore the gpu it's a server and it's only there for displaying the terminal for first time install/debugging.

Relevant part of sensors-detect
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd592
```
<details>
<summary> output of sensors-detect </summary>

```bash

# sensors-detect revision $Revision$
# System: Micro-Star International Co., Ltd. MS-7C94 [1.0]
# Board: Micro-Star International Co., Ltd. MAG B550M MORTAR (MS-7C94)
# Kernel: 5.10.2-051002-generic x86_64
# Processor: AMD Ryzen 7 5800X 8-Core Processor (25/33/0)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
[... 3101 lines cut ...]
Probing for `Maxim MAX1617'...                              No
Probing for `Maxim MAX1617A'...                             No
Probing for `Maxim MAX1668'...                              No
Probing for `Maxim MAX1805'...                              No
Probing for `Maxim MAX1989'...                              No
Probing for `Maxim MAX6655/MAX6656'...                      No
Probing for `TI THMC10'...                                  No
Probing for `National Semiconductor LM84'...                No
Probing for `Genesys Logic GL523SM'...                      No
Probing for `Onsemi MC1066'...                              No
Probing for `Maxim MAX1618'...                              No
Probing for `Maxim MAX1619'...                              No
Probing for `National Semiconductor LM82/LM83'...           No
Probing for `Maxim MAX6654'...                              No
Probing for `Maxim MAX6690'...                              No
Probing for `Maxim MAX6680/MAX6681'...                      No
Probing for `Maxim MAX6695/MAX6696'...                      No
Probing for `Texas Instruments TMP400'...                   No
Probing for `National Semiconductor LM95231'...             No
Probing for `National Semiconductor LM95233'...             No
Probing for `National Semiconductor LM95241'...             No
Probing for `Analog Devices ADM1029'...                     No
Probing for `ITE IT8712F'...                                No
Client found at address 0x2c
Probing for `Myson MTP008'...

abucodonosor, 2020-12-25:
@DrissiReda 

>I just updated my machine to 5.10.2 and I still can't see my cpu temps:

Well, you can, can't you?:

```
k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +34.1°C <-- there they are.
Tdie:         +34.1°C

```

However, if you wish to see the TCCD temps there is no such support in 5.10.x yet. I've sent a patch to Guenter a few days ago
to address hat for 5.11. Please note 5.11 will remove voltage, etc support so all you'll see will be Tctl,Tdie and Tccd from 5.11
on.

If you want to apply something for 5.10.x for k10temp you can play with this patch:

https://raw.githubusercontent.com/frugalware/frugalware-current/master/source/base/kernel/k10temp-fix-ZEN2-desktop-add-ZEN3-desktop.patch

DrissiReda, 2020-12-25:
Thanks for the link, and yes I know, but as you might have guessed I'm looking for the CCD temps. Is there a specific reason why 5.11 will remove the rest? and will it be worth it for me to go to 5.11 ?

abucodonosor, 2020-12-25:
@DrissiReda 

5.11 is not even rc1 :) So not use it yet.

Well, the reason is that is near impossible to maintain without AMD supporting that, which they don't.
There is always the out-of-tree zenpower modules, and Guenter may take a separate voltage module,
if someone writes & maintains it ofc :).

Just apply the patch to 5.10.x and you should have your CCD temps and voltage etc back for now.

DocMAX, 2020-12-27:
Ryzen 5950x here, same problem. Kernel 5.9.14.
```
k10temp-pci-00c3
Adapter: PCI adapter
temp1:         +0.0°C  (high = +70.0°C)

```

DrissiReda, 2020-12-27:
> 
> 
> Ryzen 5950x here, same problem. Kernel 5.9.14.
> 
> ```
> k10temp-pci-00c3
> Adapter: PCI adapter
> temp1:         +0.0°C  (high = +70.0°C)
> ```

if you move to kernel 5.10.2 you will get the temp instead of 0°C, but you won't be able to get the CCD temps though. But it might still be worth it for you.

worzel910, 2021-01-02:
You need this adding to your boot parms to see both sensor chip outputs.

`acpi_enforce_resources=lax`

You should then see the following 

```
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +32.0°C  

it8688-isa-0a40
Adapter: ISA adapter
in0:         960.00 mV (min =  +0.00 V, max =  +3.06 V)
in1:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.20 V  (min =  +0.00 V, max =  +3.06 V)
in5:         912.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           1.37 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.05 V  
fan1:         716 RPM  (min =    0 RPM)
fan2:         465 RPM  (min =    0 RPM)
fan3:         363 RPM  (min =    0 RPM)
fan4:        1614 RPM  (min =    0 RPM)
fan5:        1445 RPM  (min =    0 RPM)
temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +38.0°C  (low  =  +0.0°C, high = -125.0°C)  sensor = thermistor
temp6:        +49.0°C  (low  = -16.0°C, high = +90.0°C)  sensor = thermistor
intrusion0:  ALARM

nvme-pci-0400
Adapter: PCI adapter
Composite:    +37.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +37.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +39.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)

amdgpu-pci-0c00
Adapter: PCI adapter
vddgfx:        1.09 V  
fan1:           0 RPM  (min =    0 RPM, max = 3850 RPM)
edge:         +38.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +63.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +35.0°C  (crit = +94.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
power1:      185.00 W  (cap = 250.00 W)

it8792-isa-0a60
Adapter: ISA adapter
in0:           1.79 V  (min =  +0.00 V, max =  +2.78 V)
in1:         676.00 mV (min =  +0.00 V, max =  +2.78 V)
in2:         992.00 mV (min =  +0.00 V, max =  +2.78 V)
+3.3V:         3.38 V  (min =  +0.00 V, max =  +5.56 V)
in4:           1.80 V  (min =  +0.00 V, max =  +2.78 V)
in5:           1.19 V  (min =  +0.00 V, max =  +2.78 V)
in6:           2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:          3.36 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:          3.14 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

zenpower-pci-00c3
Adapter: PCI adapter
Tdie:         +35.0°C  (high = +95.0°C)
Tctl:         +35.0°C  

nvme-pci-0100
Adapter: PCI adapter
Composite:    +36.9°C  (low  = -273.1°C, high = +72.8°C)
                       (crit = +75.8°C)
Sensor 1:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +39.9°C  (low  = -273.1°C, high = +65261.8°C)
```
it8688-isa-0a40 and it8792-isa-0a60 are the sensor chips on the motherboard. 

This using kernel 5.9.16

Am looking for a sensors.conf file for use with this board.

mitjans, 2021-01-03:
I can confirm that Tctl and Tdie are now correctly displayed.

A couple of questions though:

In `/sys/class/thermal/` there are only 2 zones for this motherboard, with types:

- acpitz
- iwlwifi_1

Is that correct?

The temperature read from `acpitz-acpi-0` is always 16.8°C. I can see that @worzel910 has the same temperature, why is that?

Also, fans RPM are still 0...

TheBigK02, 2021-07-06:
I have the same issue with 5.11.0-22-generic. I have the boot paramters acpi_enforce_resources=lax but still have no fan speed while the mainboard fanspeed utility has Fan speeds.

Any updates on this or things I can try?

jared-mackey, 2021-07-14:
In the same boat. Really looking for EC_TEMP_1 measures as well.

Giger22, 2023-02-11:
> I have the same issue with 5.11.0-22-generic. I have the boot paramters acpi_enforce_resources=lax but still have no fan speed while the mainboard fanspeed utility has Fan speeds.
> 
> Any updates on this or things I can try?

https://gist.github.com/bakman2/e801f342aaa7cade62d7bd54fd3eabd8
https://forum.level1techs.com/t/configuring-lm-sensors-on-x570-aorus-master-to-get-cpu-temps-fans-etc/166170/5