lm-sensors/lm-sensors #301: Wrong Core Number in the Output

cyprix, 2020-12-22:
**HW:**
Motherboard: Asus WS-C621E-SAGE
CPUs: 2 x Xeon Platinum 24-core ES Samples QL1K
Number of CPU Threads: 96
`grep ^processor /proc/cpuinfo  | wc -l
96`

**OS**
Kernel: 5.4.0-58-generic
System: Ubuntu 20.04.1 64-bit
lm-sensors packages versions:
```
libsensors5:amd64    1:3.6.0-2ubuntu1           amd64        library to read temperature/voltage/fan sensors
libsensors5:i386     1:3.6.0-2ubuntu1           i386         library to read temperature/voltage/fan sensors
lm-sensors           1:3.6.0-2ubuntu1           amd64        utilities to read temperature/voltage/fan sensors
```

 Sensors command output shows only 30+30 ( total: 60 )cores, but should be 96 as I think:
```
sensors
coretemp-isa-0001
Adapter: ISA adapter
Package id 1:  +48.0°C  (high = +81.0°C, crit = +91.0°C)<br>
Core 0:        +47.0°C  (high = +81.0°C, crit = +91.0°C)
Core 1:        +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 2:        +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 3:        +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 4:        +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 5:        +48.0°C  (high = +81.0°C, crit = +91.0°C)
Core 8:        +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 9:        +45.0°C  (high = +81.0°C, crit = +91.0°C)
Core 10:       +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 11:       +45.0°C  (high = +81.0°C, crit = +91.0°C)
Core 12:       +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 13:       +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 16:       +47.0°C  (high = +81.0°C, crit = +91.0°C)
Core 17:       +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 18:       +45.0°C  (high = +81.0°C, crit = +91.0°C)
Core 19:       +45.0°C  (high = +81.0°C, crit = +91.0°C)
Core 20:       +47.0°C  (high = +81.0°C, crit = +91.0°C)
Core 21:       +47.0°C  (high = +81.0°C, crit = +91.0°C)
Core 24:       +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 25:       +46.0°C  (high = +81.0°C, crit = +91.0°C)
Core 26:       +47.0°C  (high = +81.0°C, crit = +91.0°C)
Core 27:       +45.0°C  (high = +81.0°C, crit = +91.0°C)
Core 28:       +45.0°C  (high = +81.0°C, crit = +91.0°C)
Core 29:       +46.0°C  (high = +81.0°C, crit = +91.0°C)
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +45.0°C  (high = +81.0°C, crit = +91.0°C)
Core 0:        +45.0°C  (high = +81.0°C, crit = +91.0°C)
Core 1:        +43.0°C  (high = +81.0°C, crit = +91.0°C)
Core 2:        +43.0°C  (high = +81.0°C, crit = +91.0°C)
Core 3:        +43.0°C  (high = +81.0°C, crit = +91.0°C)
Core 4:        +42.0°C  (high = +81.0°C, crit = +91.0°C)
Core 5:        +45.0°C  (high = +81.0°C, crit = +91.0°C)
Core 8:        +43.0°C  (high = +81.0°C, crit = +91.0°C)
Core 9:        +42.0°C  (high = +81.0°C, crit = +91.0°C)
Core 10:       +42.0°C  (high = +81.0°C, crit = +91.0°C)
Core 11:       +42.0°C  (high = +81.0°C, crit = +91.0°C)
Core 12:       +43.0°C  (high = +81.0°C, crit = +91.0°C)
Core 13:       +44.0°C  (high = +81.0°C, crit = +91.0°C)
Core 16:       +44.0°C  (high = +81.0°C, crit = +91.0°C)
Core 17:       +43.0°C  (high = +81.0°C, crit = +91.0°C)
Core 18:       +42.0°C  (high = +81.0°C, crit = +91.0°C)
Core 19:       +42.0°C  (high = +81.0°C, crit = +91.0°C)
Core 20:       +42.0°C  (high = +81.0°C, crit = +91.0°C)
Core 21:       +44.0°C  (high = +81.0°C, crit = +91.0°C)
Core 24:       +43.0°C  (high = +81.0°C, crit = +91.0°C)
Core 25:       +43.0°C  (high = +81.0°C, crit = +91.0°C)
Core 26:       +42.0°C  (high = +81.0°C, crit = +91.0°C)
Core 27:       +43.0°C  (high = +81.0°C, crit = +91.0°C)
Core 28:       +43.0°C  (high = +81.0°C, crit = +91.0°C)
Core 29:       +43.0°C  (high = +81.0°C, crit = +91.0°C)```

abucodonosor, 2020-12-25:
@cyprix 

coretemp module is maintained in the kernel tree. You may want to poke the developers there.

abucodonosor, 2020-12-25:
:~/kernel/linux$ scripts/get_maintainer.pl ./drivers/hwmon/coretemp.c
Fenghua Yu <fenghua.yu@intel.com> (maintainer:CORETEMP HARDWARE MONITORING DRIVER)
Jean Delvare <jdelvare@suse.com> (maintainer:HARDWARE MONITORING)
Guenter Roeck <linux@roeck-us.net> (maintainer:HARDWARE MONITORING)
linux-hwmon@vger.kernel.org (open list:CORETEMP HARDWARE MONITORING DRIVER)
linux-kernel@vger.kernel.org (open list)

romanstingler, 2021-01-21:
same here with 2x Xeon Platinum 8260

cyprix, 2021-01-26:
Hi @romanstingler can you provide the /proc/cpuinfo output? 
I find that  the lm-sensors display only the HW cores (That's right)
And I think the core number obtained from the procfs ( cpuinfo ).
As you can see the cores groups by 6 ( Core 0-6,8-13,16-21,24-29). Similar to the Xeon E5-2650L v3.
In my thoughts, it is not the lm-sensors bug or issue, due that it provides the information from procfs.

romanstingler, 2021-02-01:
I know that sensors just displays physical cores.

cpuinfo shows me 
**cpu cores	: 24** 
which is true per cpu

the cores do not group by 6 (where do you take the information)!

do you mean  in lscpu
```
NUMA node0 CPU(s):               0-23,48-71
NUMA node1 CPU(s):               24-47,72-95
```

Linux takes the physical cores first and then the logical!

The 2650Lv3 has 12 physical cores and
maybe your numa nodes have separated the cores according to your finding but normally the Numa nodes are per CPU (except very large core count). 

[cpuinfo.txt](https://github.com/lm-sensors/lm-sensors/files/5903507/cpuinfo.txt)

cyprix, 2021-02-01:
Hi @romanstingler 
Sorry, I wrote incorrect.
Please run this command
`grep "core id" cpuinfo.txt | sort  | uniq | sort -V`
And you will see that core id groups with 6 (0,1,2,3,4,5) and next group starts from 8

romanstingler, 2021-02-15:
blackout@Workstation ~/Videos % grep "core id" cpuinfo.txt | sort -V | uniq
core id		: 0
core id		: 1
core id		: 2
core id		: 3
core id		: 4
core id		: 5
core id		: 8
core id		: 9
core id		: 10
core id		: 11
core id		: 12
core id		: 13
core id		: 16
core id		: 17
core id		: 18
core id		: 19
core id		: 20
core id		: 21
core id		: 24
core id		: 25
core id		: 26
core id		: 27
core id		: 28
core id		: 29

romanstingler, 2021-03-22:
and why was is correctly displayed months ago ??