lm-sensors/lm-sensors #303: ASUS X570 Crosshair VIII Dark Hero / Ryzen 5950X

petasis, 2020-12-29:
Hi all,

I am trying to get temperatures on this board, but the output is really sparse:
```
# sensors
nvme-pci-0100
Adapter: PCI adapter
Composite:    +33.9°?C  (low  =  -5.2°?C, high = +83.8°?C)
                       (crit = +87.8°?C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +33.0°?C  

k10temp-pci-00c3
Adapter: PCI adapter
temp1:         +0.0°?C  (high = +70.0°?C)
```

Ok, I know I need kernel 5.10 for the cpu temp to work, but what about the rest?

The board has a [nuvoton nct6798D](https://www.techpowerup.com/review/asus-rog-crosshair-viii-dark-hero-c8h/7.html) chip.

The output of sensors-detect is:
```
# sensors-detect
# sensors-detect version 3.6.0
# Board: ASUSTeK COMPUTER INC. ROG CROSSHAIR VIII DARK HERO
# Kernel: 5.9.16-200.fc33.x86_64 x86_64
# Processor: AMD Ryzen 9 5950X 16-Core Processor (25/33/0)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no):
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT6798D Super IO Sensors'                   Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
[... 210 lines cut ...]
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)
Client found at address 0x77
Probing for `Asus Mozart-2'...                              No

Next adapter: SMBus PIIX4 adapter port 2 at 0b00 (i2c-1)
Do you want to scan it? (YES/no/selectively):

Next adapter: SMBus PIIX4 adapter port 1 at 0b20 (i2c-2)
Do you want to scan it? (YES/no/selectively): 

Now follows a summary of the probes I have just done.
Just press ENTER to continue:

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

Driver `nct6775':
  * ISA bus, address 0x290
    Chip `Nuvoton NCT6798D Super IO Sensors' (confidence: 9)
```

nct6775 kernel module is loaded, I can see it in `lsmod`.

veldenb, 2021-01-30:
Have you tried adding acpi_enforce_resources=lax as a kernel parameter?
See https://github.com/lm-sensors/lm-sensors/issues/220#issuecomment-707762085

ncoder-1, 2021-02-20:
I have a ASUS X570 Crosshair VIII Dark Hero and confirm that adding `acpi_enforce_resources=lax` to my kernel parameters worked:

```
nct6798-isa-0290
Adapter: ISA adapter
in0:                        1.31 V  (min =  +0.00 V, max =  +1.74 V)
in1:                        1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.74 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                      592.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                        1.08 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.25 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                      896.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                       8.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                      88.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                       1.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                     888.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                        0 RPM  (min =    0 RPM)
fan2:                     1347 RPM  (min =    0 RPM)
fan3:                      815 RPM  (min =    0 RPM)
fan4:                        0 RPM  (min =    0 RPM)
fan5:                        0 RPM  (min =    0 RPM)
fan6:                        0 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +37.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +44.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +21.0°C    sensor = thermistor
AUXTIN1:                  +127.0°C    sensor = thermistor
AUXTIN2:                  +104.0°C    sensor = thermistor
AUXTIN3:                   +32.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +44.0°C  
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
PCH_CHIP_TEMP:              +0.0°C  
PCH_CPU_TEMP:               +0.0°C  
intrusion0:               ALARM
intrusion1:               ALARM
beep_enable:              disabled
```

petasis, 2021-02-20:
Yes, adding this acpi_enforce_resources=lax does work, although I am not sure all sensors are reported.

I have also created a configuration file for the dark hero:

```
chip "nct6798-*"
    # For the Asus ROG CROSSHAIR VIII Dark Hero.
    
    # Fan headers
    label fan1 "Chassis Fan 1"
    
    label fan2 "CPU/CPU OPT Fan"
        set fan2_min 200
    
    label fan3 "Chassis Fan 2"
    label fan4 "Chassis Fan 3"
    label fan5 "H_AMP"
    label fan6 "W_PUMP+"
    label fan7 "AIO PUMP"
    
    # Voltages
    label in0 "Vcore"
        set in0_min 0.2
        set in0_max 1.5
    
    label in1 "AVCC +5V"
        compute in1 @ * 5, @ / 5
        set in1_min  5 * 0.95
        set in1_max  5 * 1.05
    
    label in2 "AVSB"
        set in2_min  3.3 * 0.95
        set in2_max  3.3 * 1.05
    
    label in3 "3VCC"
        set in3_min  3.3 * 0.95
        set in3_max  3.3 * 1.05
    
    label in4 "+12V"
        compute in4 @ * 7, @ / 7
        # compute in4 @*((2*28/10)+1),  @/(2*(28/10)+1)
        set in4_min  12 * 0.95
        set in4_max  12 * 1.05
    
    label in5 "Chipset"
        compute in5 @ * 2, @ / 2
        set in5_min  0.2
        set in5_max  1.5
    
    label in6 "CPU SOC"
        set in6_min  0.2
        set in6_max  1.5
    
    label in7 "3VSB +3.3V"
        set in7_min  3.3 * 0.95
        set in7_max  3.3 * 1.05
    
    label in8 "Vbat"
        set in8_min  3.3 * 0.95
        set in8_max  3.3 * 1.05
    
    label in9 "1.8 PLL"
        compute in9 @ * 2, @ / 2
        set in9_min  1.8 * 0.95
        set in9_max  1.8 * 1.05
    
    label in10 "VDDP"
        compute in10 @ * 2, @ / 2
        set in10_max  1.1
    
    label in11 "VIN6"
        compute in11 @ * 2, @ / 2
        set in11_max  1.1
    
    label in12 "1.8V_ALW"
        compute in12 @ * 1.8, @ / 1.8
        set in12_min  1.8 * 0.95
        set in12_max  1.8 * 1.05
    
    label in13 "DRAM"
        set in13_min 1.20
        set in13_max 1.45
    
    label in14 "1.8_RUN"
        compute in14 @ * 2, @ / 2
        set in14_min  1.8 * 0.95
        set in14_max  1.8 * 1.05
    
    # Temperatures
    label temp1 "Motherboard"
        set temp1_type      4
        set temp1_max      60
        set temp1_max_hyst 55
    label temp2 "CPU"
        set temp2_type      4
        set temp2_max      70
        set temp2_max_hyst 65
    label temp7 "CPU Package"
        set temp7_type      4
        set temp7_max      70
        set temp7_max_hyst 65
        set temp7_max_crit 80
    
    # Always show zeros
    ignore temp3
    ignore temp4
    ignore temp5
    ignore temp6
    ignore temp8
    ignore temp9
    ignore temp10
    
    # The motherboard doesn't seem to have them physically connected or present
    ignore intrusion0
    ignore intrusion1
```

With the above configuartion, I get this output:

```
nct6798-isa-0290
Adapter: ISA adapter
Vcore:           944.00 mV (min =  +0.20 V, max =  +1.50 V)
AVCC +5V:          5.00 V  (min =  +4.76 V, max =  +5.24 V)
AVSB:              3.41 V  (min =  +3.14 V, max =  +3.47 V)
3VCC:              3.33 V  (min =  +3.14 V, max =  +3.47 V)
+12V:             12.21 V  (min = +11.42 V, max = +12.60 V)
Chipset:           1.20 V  (min =  +0.21 V, max =  +1.50 V)
CPU SOC:           1.08 V  (min =  +0.20 V, max =  +1.50 V)
3VSB +3.3V:        3.41 V  (min =  +3.14 V, max =  +3.47 V)
Vbat:              3.25 V  (min =  +3.14 V, max =  +3.47 V)
1.8 PLL:           1.79 V  (min =  +1.71 V, max =  +1.89 V)
VDDP:             80.00 mV (min =  +0.00 V, max =  +1.10 V)
VIN6:            128.00 mV (min =  +0.00 V, max =  +1.10 V)
1.8V_ALW:          1.84 V  (min =  +1.71 V, max =  +1.89 V)
DRAM:              1.38 V  (min =  +1.20 V, max =  +1.45 V)
1.8_RUN:           1.79 V  (min =  +1.71 V, max =  +1.89 V)
Chassis Fan 1:      0 RPM  (min =    0 RPM)
CPU/CPU OPT Fan: 2380 RPM  (min =  200 RPM)
Chassis Fan 2:      0 RPM  (min =    0 RPM)
Chassis Fan 3:   1467 RPM  (min =    0 RPM)
H_AMP:              0 RPM  (min =    0 RPM)
W_PUMP+:            0 RPM  (min =    0 RPM)
AIO PUMP:           0 RPM  (min =    0 RPM)
Motherboard:      +25.0°?C  (high = +60.0°?C, hyst = +55.0°?C)  sensor = thermistor
CPU:              +34.0°?C  (high = +70.0°?C, hyst = +65.0°?C)  sensor = thermistor
CPU Package:      +38.0°?C  
beep_enable:     disabled
```

Eraseri, 2021-04-19:
@petasis How do I put in your configuration? I have the same mobo

petasis, 2021-04-20:
You can just copy the contents in a new file, inside /etc/sensors.d. In my Fedora 33 system, I named the file `AsusCrosshairVIIIDarkHero` and just placed it in  /etc/sensors.d.
[AsusCrosshairVIIIDarkHero.txt](https://github.com/lm-sensors/lm-sensors/files/6341330/AsusCrosshairVIIIDarkHero.txt)

markusressel, 2021-08-15:
I tried both `acpi_enforce_resources=lax` and adding the configuration file proposed by @petasis (thx, btx), but I still only get this output:

```
amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      862.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3300 RPM)
edge:         +47.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
junction:     +49.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +46.0°C  (crit = +105.0°C, hyst = -273.1°C)
                       (emerg = +110.0°C)
power1:       27.00 W  (cap = 186.00 W)

nvme-pci-0700
Adapter: PCI adapter
Composite:    +41.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +41.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +40.9°C  (low  = -273.1°C, high = +65261.8°C)

nvme-pci-0600
Adapter: PCI adapter
Composite:    +48.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +48.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +56.9°C  (low  = -273.1°C, high = +65261.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +46.0°C  

nvme-pci-0e00
Adapter: PCI adapter
Composite:    +46.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +46.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +53.9°C  (low  = -273.1°C, high = +65261.8°C)
```

The kernel module is loaded:

```
> lsmod | grep nct6775
nct6775                73728  0
hwmon_vid              16384  1 nct6775
```

I ran `sudo sensors-detect` multiple times throughout the testing, nothing changed.

`dmesg` can see the chip though:

```
> sudo dmesg | grep nct6775
[   18.279002] nct6775: Found NCT6798D or compatible chip at 0x2e:0x290
```

Not sure whats missing?

markusressel, 2021-08-19:
Huh, I don't know what changed, but now I get the expected output:

```
sensors                                                           0 (11:50.981) < 00:33:45
amdgpu-pci-0e00
Adapter: PCI adapter
vddgfx:      887.00 mV 
fan1:         561 RPM  (min =    0 RPM, max = 3300 RPM)
edge:         +56.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
junction:     +60.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +56.0°C  (crit = +105.0°C, hyst = -273.1°C)
                       (emerg = +110.0°C)
power1:       29.00 W  (cap = 186.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +67.0°C  
Tdie:         +67.0°C  
Tccd1:        +56.8°C  

nct6798-isa-0290
Adapter: ISA adapter
Vcore:           888.00 mV (min =  +0.20 V, max =  +1.50 V)
AVCC +5V:          5.04 V  (min =  +4.76 V, max =  +5.24 V)
AVSB:              3.39 V  (min =  +3.14 V, max =  +3.47 V)
3VCC:              3.31 V  (min =  +3.14 V, max =  +3.47 V)
+12V:             12.26 V  (min = +11.42 V, max = +12.60 V)
Chipset:           1.18 V  (min =  +0.21 V, max =  +1.50 V)
CPU SOC:           1.08 V  (min =  +0.20 V, max =  +1.50 V)
3VSB +3.3V:        3.39 V  (min =  +3.14 V, max =  +3.47 V)
Vbat:              3.34 V  (min =  +3.14 V, max =  +3.47 V)
1.8 PLL:           1.78 V  (min =  +1.71 V, max =  +1.89 V)
VDDP:             16.00 mV (min =  +0.00 V, max =  +1.10 V)
VIN6:            208.00 mV (min =  +0.00 V, max =  +1.10 V)
1.8V_ALW:          1.84 V  (min =  +1.71 V, max =  +1.89 V)
DRAM:              1.35 V  (min =  +1.20 V, max =  +1.45 V)
1.8_RUN:           1.78 V  (min =  +1.71 V, max =  +1.89 V)
Chassis Fan 1:      0 RPM  (min =    0 RPM)
CPU/CPU OPT Fan: 1050 RPM  (min =  200 RPM)
Chassis Fan 2:    701 RPM  (min =    0 RPM)
Chassis Fan 3:    390 RPM  (min =    0 RPM)
H_AMP:            389 RPM  (min =    0 RPM)
W_PUMP+:            0 RPM  (min =    0 RPM)
AIO PUMP:           0 RPM  (min =    0 RPM)
Motherboard:      +44.0°C  (high = +60.0°C, hyst = +55.0°C)  sensor = thermistor
CPU:               -0.5°C  (high = +70.0°C, hyst = +65.0°C)  sensor = thermistor
CPU Package:      +58.5°C  
beep_enable:     disabled

nvme-pci-0100
Adapter: PCI adapter
Composite:    +52.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +52.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +52.9°C  (low  = -273.1°C, high = +65261.8°C)

nvme-pci-0400
Adapter: PCI adapter
Composite:    +53.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +53.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +54.9°C  (low  = -273.1°C, high = +65261.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +41.0°C  

nvme-pci-0500
Adapter: PCI adapter
Composite:    +29.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +29.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +47.9°C  (low  = -273.1°C, high = +65261.8°C)
```

zelifcam, 2022-01-05:
Does this configuration eventually find its way into the code base?  Is there a repo for similar fixes for other chips? I feel like this is one of those "problems" that have solutions but not a good one, especially for new users trying to test their hardware.