lm-sensors/lm-sensors #307: Missing support for nuvoton npcd379

moro1111, 2021-01-13:
I am trying to get lm-sensors to work on my HP ProDesk. I think the missing kernel driver for my IO chip causes the problem, due to the line
```
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x1c11
```
but i am not sure. I am using a NUVOTON npcd379h chip. 
[image]

After a first google result i did not found any datasheets for this chip, but i also have no idea where to look.
Is there any way to fix this problem? Or is there probably a quite similar chip, so i could try to force that driver?
Or any other ideas how to tackle the problem?

Just in case, the full result of `sensors-detect`

```
$ sudo sensors-detect
# sensors-detect version 3.6.0
# System: Hewlett-Packard HP ProDesk 490 G1 MT
# Board: Hewlett-Packard 18EB
# Kernel: 5.8.0-36-generic x86_64
# Processor: Intel(R) Core(TM) i5-4570 CPU @ 3.20GHz (6/60/3)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): y
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): y
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x1c11
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): y
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): y
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): y
Using driver `i2c-i801' for device 0000:00:1f.3: Intel Lynx Point (PCH)

Next adapter: SMBus I801 adapter at f040 (i2c-0)
Do you want to scan it? (YES/no/selectively): y
Client found at address 0x50
Handled by driver `at24' (already loaded), chip type `spd'
    (note: this is probably NOT a sensor chip!)
Client found at address 0x51
Handled by driver `at24' (already loaded), chip type `spd'
    (note: this is probably NOT a sensor chip!)
Client found at address 0x52
Handled by driver `at24' (already loaded), chip type `spd'
    (note: this is probably NOT a sensor chip!)
Client found at address 0x53
Handled by driver `at24' (already loaded), chip type `spd'
    (note: this is probably NOT a sensor chip!)

Next adapter: i915 gmbus ssc (i2c-1)
Do you want to scan it? (yes/NO/selectively): y
y

Next adapter: i915 gmbus vga (i2c-2)
Do you want to scan it? (yes/NO/selectively): Client found at address 0x49
Probing for `National Semiconductor LM75'...                No
Probing for `National Semiconductor LM75A'...               No
Probing for `Dallas Semiconductor DS75'...                  No
Probing for `National Semiconductor LM77'...                No
Probing for `Analog Devices ADT7410/ADT7420'...             No
Probing for `Maxim MAX6642'...                              No
Probing for `Texas Instruments TMP435'...                   No
Probing for `National Semiconductor LM73'...                No
Probing for `National Semiconductor LM92'...                No
Probing for `National Semiconductor LM76'...                No
Probing for `Maxim MAX6633/MAX6634/MAX6635'...              No
Probing for `NXP/Philips SA56004'...                        No
Probing for `SMSC EMC1023'...                               No
Probing for `SMSC EMC1043'...                               No
Probing for `SMSC EMC1053'...                               No
Probing for `SMSC EMC1063'...                               No

Next adapter: i915 gmbus panel (i2c-3)
Do you want to scan it? (yes/NO/selectively): y

Next adapter: i915 gmbus dpc (i2c-4)
Do you want to scan it? (yes/NO/selectively): y

Next adapter: i915 gmbus dpb (i2c-5)
Do you want to scan it? (yes/NO/selectively): y

Next adapter: i915 gmbus dpd (i2c-6)
Do you want to scan it? (yes/NO/selectively): y

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

To load everything that is needed, add this to /etc/modules:
#----cut here----
# Chip drivers
coretemp
#----cut here----
If you have some drivers built into your kernel, the list above will
contain too many modules. Skip the appropriate ones!
```
and the detected sensores, where both fans are missing
```
$ sensors
acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +105.0°C)
temp2:        +29.8°C  (crit = +105.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +25.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +22.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:        +26.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:        +20.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:        +24.0°C  (high = +80.0°C, crit = +100.0°C)
```

deepcoder, 2021-02-25:
I too would like to get fan support on a HP Desktop, specific model:
Hewlett-Packard HP ProDesk 600 G1 DM
sensors-detect only finds the Intel digital thermal sensor and then finds this sensor that it does not recognize :

```
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x1c11
```

Thanks for your great work!

wildstray, 2022-01-15:
Me too! HP 260 G2...

Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x1c11

liyan2015, 2025-07-26:
Same chip with missing fan speed. Found a reverse engineered datasheet [here](https://doc.coreboot.org/_sources/superio/nuvoton/npcd378.md.txt), which has been tested [here](https://dokk.org/documentation/coreboot/4.22.01/mainboard/hp/compaq_8300_usdt/).