lm-sensors/lm-sensors #312: Use /etc/modules-load.d instead of systemd service

Mek101, 2021-02-22:
Hello, I'm one of the four users of void linux. 
Given that void linux uses runit instead of systemd, lm_sensors [doesn't work](https://github.com/void-linux/void-packages/issues/27525); and I would ask it's possible to move to use a config file with the modules to load in `/etc/modules-load.d` instead of a systemd service.
Such a change shouldn't change compatibility with other distros (at worst, it would require to change the directory name), wouldn't require the admin to enable the service after a `sensors-detect` and removes an otherwise unneeded dependency

thegarlynch, 2021-04-16:
This here please

Sigma-One, 2021-12-06:
+1 from me
Most likely also applies to other non-systemd distros unless they do custom patches

Samson7, 2024-02-23:
Is there some news about this issue? Any hope that this will be done? I tired of loud fans on Void.

tylerdq, 2025-07-28:
I also have loud fans on Void and would like to enable this service.