lm-sensors/lm-sensors #320: Add support for ITE IT8987

Bogdan107, 2021-03-23:
Please, add support to IT8987 chip.

Laptop info:
```
$ inxi -M
Machine:   Type: Laptop System: HP product: HP Pavilion Gaming Laptop 15-ec1xxx v: N/A serial: <superuser/root required> 
           Mobo: HP model: 87B1 v: 31.21 serial: <superuser/root required> UEFI: AMI v: F.20 date: 11/04/2020 
```
sensors-detect info:
```
Do you want to scan for Super I/O sensors? (YES/no): y
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x8987
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
```
isadump info:
```
# isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 89 87 01 01 00 00 00 00 00 00 00 00 01 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
modprobe && dmesg info:
```
# modprobe it87                                                                                                                                                                                                          
modprobe: ERROR: could not insert 'it87': No such device
# dmesg | grep it87                                                                                                                                                                                                      
[68399.883557] it87: Unsupported chip (DEVID=0x8987)
```

Expreiments with _modprobe -r it87 && modprobe it87 force_id=0xXXXX_ (as described in [https://github.com/a1wong/it87]) was not help: in all variants of XXXX, _sensors-detect && sensors_ do not show CPU-temp or CPU cooler speed.

Please, add support to IT8987 chip.

Datasheet for IT8987 attached.
Datasheet was received officially from itesupport@ite.com.tw
[IT8987_B_V0.3.1_102513.pdf](https://github.com/lm-sensors/lm-sensors/files/6190220/IT8987_B_V0.3.1_102513.pdf)

gatalon, 2021-03-26:
My device is hades canyon use IT8987 chip, got no fanspeeds at all, I can take all test job if you want.

Bogdan107, 2021-03-26:
> My device is hades canyon use IT8987 chip, got no fanspeeds at all, I can take all test job if you want.

Yes, I want this.

My laptop has 2 coolers inside.
BIOS has 2 modes for coolers: "Always On" and "Automatic".
Coolers started with system power on and stopped after a few seconds.
1-st cooler start automatic when CPU load.
2-nd cooler stops after system startup and start only when GPU load.

Coolers speeds showed neither "sensors", no "nvidia-smi" tools (nvidia-smi shows GPU temperature).

vit9696, 2021-03-27:
IT8987 is not a usual chip, and you cannot "support" it as it does not provide FAN information to the host. More proper explanation is given here: https://github.com/acidanthera/bugtracker/issues/395#issuecomment-808728953.

For some boards with this chip and several others you can bruteforce the EC RAM area as explained [here](https://github.com/acidanthera/bugtracker/issues/395#issuecomment-808761101). For NUCs in particular we provided the reference implementation here with @joedmru: https://github.com/acidanthera/VirtualSMC/commit/3b17b820271acef25560d0de6a9f9583874868cf. Feel free to use it but provide us with updates if you add support for more hardware using similar EC sensors (:

RAMChYLD, 2022-11-17:
According to [this](https://www.phoronix.com/news/Star-Labs-Fwupd-Support) Phoronix article, Star Labs are releasing some laptops with the it8987 chipset support and is apparently making inroads to getting the chip working in Linux. Can anyone get in contact with them and see if they’d open source their code?