lm-sensors/lm-sensors #325: Found unknown chip with ID 0xb732 - Nuvoton NCT6683D

jorgelaranjo, 2021-04-01:
On a Lenovo P700 running `5.11.10-arch1-1` running `sensors-detect` bring the following

```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xb732
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
```

Base on the specsheet from the manufacturer I'd guess this
`Found unknown chip with ID 0xb732` corresponds to the `NCT6683D` chip

[image]

The module does exist in the system but does not load

`modprobe: ERROR: could not insert 'nct6683': No such device`
Not sure why happens. 

The system is booting with `acpi_enforce_resources=lax`

JiangXL, 2021-10-31:
I can confirm this issue in Thinkstation P500 with Fedora 34.

jorgelaranjo, 2021-10-31:
> I can confirm this issue in Thinkstation P500 with Fedora 34.

Have you found a solution for it or not really?

JiangXL, 2021-10-31:
Sorry, I can't find the solution too.

hightowe, 2023-04-12:
I can confirm this issue on a Thinkstation P360 running Linux Mint 21.1.

a10kiloham, 2023-11-20:
@hightowe did you ever discover what the 0x0111 corresponds to module wise?

hightowe, 2023-11-20:
> @hightowe did you ever discover what the 0x0111 corresponds to module wise?

@a10kiloham - unfortunately, no...

a10kiloham, 2023-11-20:
@hightowe it seems there's a patch for it but i've not yet figured out how to try it
https://patchwork.kernel.org/project/linux-watchdog/patch/20230808174631.243777-1-dober6023@gmail.com/

hightowe, 2023-11-20:
@a10kiloham - that looks like a watchdog timer driver, to me, not lmsensors support.

a10kiloham, 2023-11-20:
@hightowe ah that makes sense. confusing there's no module anywhere for this series of chip...

jorgelaranjo, 2023-12-09:
> @hightowe it seems there's a patch for it but i've not yet figured out how to try it https://patchwork.kernel.org/project/linux-watchdog/patch/20230808174631.243777-1-dober6023@gmail.com/

https://www.kernel.org/doc/html/v4.15/process/applying-patches.html

a10kiloham, 2023-12-09:
@jorgelaranjo please let us know if you get it working!

jorgelaranjo, 2023-12-09:
> @jorgelaranjo please let us know if you get it working!

Not planning on using the P700 anymore, moved into a M75Q gen 2, E14 gen4 and my trusted X270