lm-sensors/lm-sensors #332: lm-sensors does not work with 5.10.X on XPS 10th gen

psammarco, 2021-04-16:
Hello,

I have recently noticed that PWM and lm_sensors do fail to work on my XPS 10th gen under kernel 5.10.X and Gentoo, specifically gentoo-sources-5.10.27.

I do think, and I may be wrong, that the kernel recognizes the laptop fans, however for some reason lm_sensors fails to detect it. I have even tried mainline and latest stable kernel from kernel.org, though it made no difference.

I am using the same configuration as per kernel 5.4.97, where lm_sensors and fans do work just fine.

I was wondering if this is a known issue or if the stack has changed and I need to do something else to get it working on kernel 5.10.X.

I am attaching some logs that will hopefully give you a better idea of the scenario described. I apologize for the amount of logs and I do appreciate your help.

Best Regards,
Pietro Sammarco
[lm_sensors_logs.zip](https://github.com/lm-sensors/lm-sensors/files/6323995/lm_sensors_logs.zip)

Wer-Wolf, 2021-05-11:
In dmesg, you can find the following:
[   21.004938] dell_smm_hwmon: not running on a supported Dell system.
[   21.004940] dell_smm_hwmon: vendor=Dell Inc., model=XPS 13 7390, version=1.7.0

Since dell_smm_hwmon is likely responsible for fan handling, pwmconfig/fancontrol is unable to control them anymore.
I think this issue was introduced in the linux kernel in the commit "hwmon: (dell-smm) Use one DMI match for all XPS models", which might prevent dell_smm_hwmon from recognizing you model now.

Better open an issue on https://bugzilla.kernel.org/, because this seems to be a kernel issue.

psammarco, 2021-05-12:
Thanks @Wer-Wolf. Just raised https://bugzilla.kernel.org/show_bug.cgi?id=213039.

psammarco, 2021-05-12:
@farzad447 unrelated and no.

Wer-Wolf, 2021-05-14:
I jsut had an idea, try loading the module with ignore_dmi=1

psammarco, 2021-05-15:
made no difference unfortunately.

Wer-Wolf, 2021-05-15:
If the error message is still appearing thats normal.
Pls load the module with
`modprobe -r dell-smm-hwmon`
`modprobe dell-smm-hwmon ignore_dmi=1 force=1`

What does `dmesg` and `sensors` say now?

Wer-Wolf, 2021-05-15:
Can you give me the output of `pwmconfig` and `sensors` on kernel 5.4?

Wer-Wolf, 2021-05-15:
This also means that i was wrong on my first assumption, you may need to delete you bugreport on bugzilla.
Sorry

psammarco, 2021-05-16:
> Can you give me the output of `pwmconfig` and `sensors` on kernel 5.4?
> Because after looking at the dmesg and sensor logs of kernel 5.10, its seems that the kernel is indeed detecting the model.
> However it reports a constant fan speed of 0 RPM, so pwmconfig ignores both fans.
> I have a similar problem on my Latitude C600 with pwmconfig ignoring my fan, try to use i8kutils.

```
# pwmconfig version 3.6.0
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

Found the following devices:
   hwmon0 is AC
   hwmon1 is BAT0
   hwmon2 is coretemp
   hwmon3 is dell_smm
   hwmon4 is iwlwifi_1

Found the following PWM controls:
   hwmon3/pwm1           current value: 0
   hwmon3/pwm2           current value: 0

Giving the fans some time to reach full speed...
Found the following fan sensors:
   hwmon3/fan1_input     current speed: 0 ... skipping!
   hwmon3/fan2_input     current speed: 0 ... skipping!

There are no working fan sensors, all readings are 0.
Make sure you have a 3-wire fan connected.
You may also need to increase the fan divisors.
See doc/fan-divisors for more information.
admin@gentoo ~ $ 
```

```
admin@gentoo ~ $ sensors
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +45.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +65.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +65.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +62.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +65.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +64.0°C  (high = +100.0°C, crit = +100.0°C)

dell_smm-virtual-0
Adapter: Virtual device
fan1:           0 RPM
fan2:           0 RPM

BAT0-acpi-0
Adapter: ACPI interface
in0:           7.67 V  
curr1:         2.37 A  

admin@gentoo ~ $ 
```

Do you think this to be a userspace bug at this point? Will i8kutils work well with laptop-mode-tools? Thanks

Wer-Wolf, 2021-05-16:
Im now confused.
Under 5.4 and 5.10, pwmconfig fails to detect the fans because of its (in my opinion) strange design.
But if pwmconfig fails to detect them under both kernels, what made you think fancontrol worked on 5.4?

The reason for pwmconfig ignoring you fans is that they dont report a inalid speed (0 RPM).
But the fans can still be controled, its just an issue of pwmconfig and fancontrol requiring speed reporting even if its not neccesary.

If i8kutils will work well with laptop mode tools? Idk :(
After all, both fancontrol and i8kutils are just simple scripts reading temps and setting the fans.
See [https://wiki.archlinux.org/title/Fan_speed_control](url) for setup.