lm-sensors/lm-sensors #333: No individual CPU core temperatures reported on Ryzen 9 5950X / ASUS Prime B550-Plus

locque, 2021-05-02:
This is the output from sensors after running sensors-detect and allowing all checks:
```
k10temp-pci-00c3
Adapter: PCI adapter
temp1:         +0.0°C  (high = +70.0°C)

nouveau-pci-0400
Adapter: PCI adapter
GPU core:    850.00 mV (min =  +0.82 V, max =  +1.21 V)
fan1:        1140 RPM
temp1:        +36.0°C  (high = +95.0°C, hyst =  +3.0°C)
                       (crit = +105.0°C, hyst =  +5.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)
power1:       19.31 W  (crit = 139.61 mW)
```
Why are the individual core temperatures not being displayed?

OS : Ubuntu Server 20.04
CPU : AMD Ryzen 9 5950X
RAM : 2x 32GB Corsair Vengeance LPX DDR4-3600 CL18-22-22-42
MBD : Asus Prime B550-Plus
SSD : 2x 2TB Corsair MP600
HDD : 12TB WD My Book over USB
PSU : Corsair HX750

winkee01, 2021-05-30:
same here, can't detect CPU temperature

01bbae, 2022-05-01:
Mine too. Running 5950x with Asus Proart x570 on Proxmox.