lm-sensors/lm-sensors #342: Wrong fan speed on ASUS Pro WS WRX80E-SAGE SE WIFI with Nuvoton NCT 6798D-R

bachandi, 2021-05-31:
sensors does not read the fan speeds correctly on my  ASUS Pro WS WRX80E-SAGE SE WIFI motherboard. It only reports 0 rpm for all fans. nct6775 kernel module is loaded and the other temperatures are read fine. I tried to set ``acpi_enforce_resources=lax``  but it didn't change anything on the rpms.

System:
Ubuntu 21.04
5.11.0-17-generic
sensors version 3.6.0 with libsensors version 3.6.0

Kernel module is loaded:
```
dmesg | grep nct6775
[    5.450173] nct6775: Found NCT6798D or compatible chip at 0x2e:0x290
```

sensors output with fan1 to fan7 at 0 rpm which is not correct: 
```
amd_energy-isa-0000
Adapter: ISA adapter
Ecore000:     11.29 kJ
Ecore001:     11.18 kJ
Ecore002:     11.11 kJ
Ecore003:     11.14 kJ
Ecore004:     11.10 kJ
Ecore005:     11.10 kJ
Ecore006:     11.17 kJ
Ecore007:     11.08 kJ
Ecore008:     10.83 kJ
Ecore009:     10.89 kJ
Ecore010:     10.81 kJ
Ecore011:     10.87 kJ
Ecore012:     10.93 kJ
Ecore013:     11.01 kJ
Ecore014:     10.91 kJ
Ecore015:     10.99 kJ
Ecore016:     12.47 kJ
Ecore017:     12.39 kJ
Ecore018:     12.41 kJ
Ecore019:     12.39 kJ
Ecore020:     12.27 kJ
Ecore021:     12.27 kJ
Ecore022:     12.25 kJ
Ecore023:     12.27 kJ
Ecore024:     10.66 kJ
Ecore025:     10.67 kJ
Ecore026:     10.67 kJ
Ecore027:     10.69 kJ
Ecore028:     10.71 kJ
Ecore029:     10.73 kJ
Ecore030:     10.62 kJ
Ecore031:     10.66 kJ
Esocket0:    565.75 kJ

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +74.8°C  
Tdie:         +74.8°C  
Tccd1:        +72.2°C  
Tccd3:        +71.8°C  
Tccd5:        +74.2°C  
Tccd7:        +66.5°C  

nvme-pci-2d00
Adapter: PCI adapter
Composite:    +37.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +37.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +40.9°C  (low  = -273.1°C, high = +65261.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:            N/A  

nct6798-isa-0290
Adapter: ISA adapter
in0:                   288.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                   1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                     3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                     3.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                     1.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                   592.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                   336.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                     3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                     3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                   880.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                  192.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                  152.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                    1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                  328.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                  312.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                     0 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
fan6:                     0 RPM  (min =    0 RPM)
fan7:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +29.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                 +58.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +76.5°C    sensor = thermistor
AUXTIN1:                +82.0°C    sensor = thermistor
AUXTIN2:                +82.0°C    sensor = thermistor
AUXTIN3:                +86.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C  
intrusion0:            OK
intrusion1:            ALARM
beep_enable:           disabled
```

Did anyone face this issue with a similar chip or motherboard?

vazmuten, 2021-06-06:
Absolutely the same behavior on my B550 Taichi:

nct6798-isa-0290
Adapter: ISA adapter
in0:                   704.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                     1.66 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                     3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                     3.26 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                     1.83 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                     1.09 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                     1.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                     3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                     3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                   904.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                    1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                  624.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                    1.08 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                  912.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                  904.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                     0 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
fan6:                     0 RPM  (min =    0 RPM)
fan7:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +46.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                 +55.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                 +8.5°C    sensor = thermistor
AUXTIN1:                -61.0°C    sensor = thermistor
AUXTIN2:                +13.0°C    sensor = thermistor
AUXTIN3:                +31.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C
PCH_CHIP_TEMP:           +0.0°C
PCH_CPU_TEMP:            +0.0°C
PCH_MCH_TEMP:            +0.0°C
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

bachandi, 2021-07-21:
Did anyone come up with a solution?

erwincoumans, 2022-04-07:
Also issues with fans with the same motherboard. In 'automatic' fan control, fans seem to run at 60% to 100%. In manual override mode, fan speed seems inconsistent, sometimes too fast, sometimes off, independent of the temperatures.

nerdroychan, 2022-09-09:
Running a B550 Taichi and having the same issue as @vazmuten. Looks like a driver issue.

nerdroychan, 2022-09-10:
CC @vazmuten

I did some research today and I found that the B550 Taichi motherboard actually uses a Nuvoton 6683D chip for sensors. However, when booting up the kernel both `nct6683` and `nct6775` modules are loaded (this can be seen in `dmesg`). So the solution is to remove the wrong kmod and make sure the correct one is loaded. I just created a config file under `/etc/modprobe.d` with the following content and then the fan speed output can be read correctly:

```
blacklist nct6775
options nct6683 force=1
```

The `force=1` is necessary to make it work correctly. This is documented at: https://michael.stapelberg.ch/posts/2021-07-10-linux-25gbit-internet-router-pc-build/.

I'm not sure if this is the case for the ASUS WRX80 motherboard. Maybe it's worth a shot.

arkitoure, 2023-02-21:
@bachandi What ever came of this? We have a few workstations on the same MB and fan control is 0. All utilities show n/a or 0 RPM and there seems to be no BIOS PWM control on temp changes. Common solution I keep getting is to tap into IPMI but this is not ideal on a non-server use workstation. It must be some option in the BIOS and/or jumper on the board?

bachandi, 2023-02-26:
@arkitoure I am still using IPMI to monitor fan speed. Fan control is working and can be set via the BMC web interface of the Motherboard. There you can specify individual fan speed curves depending either CPU temperature or an external temperature sensor.

arkitoure, 2023-03-04:
Thanks @bachandi, yes, we figured this out right after this message. Didn't even know IPIMI was on and active on the network as there is no bold/obvious indication of this from the manual. Checked the BIOS for IP's and configured everything as needed.

There seems to be a firmware **bug** also, for whomever else might read this. If the logs get filled up with warnings and errors the fan speed and temps will stop reporting and you can potentially get thermal runaway.

So anyone using this MB please be careful and check the IPMI logs, clear them and set temp thresholds within the range of your fan specifications.

We are using a 200, 140 and 120 sized Noctua Chromax fans, each with their own specific min and max RPM's. Make sure your thresholds are setup for the fans you have in the correct RPM range or the logs will fill up again and stop reporting this important data. Defaults are for industrial RPM's so this is why the out the box setup fills the logs before you even get to configure it.

Sword-Smith, 2023-03-27:
Same behavior on ` ASUS Pro WS WRX80E-SAGE SE WIFI` with kernel `5.19.0-38-generic`. Getting a reading of 0 RPM but in the BIOS I can see sensible readings of the fan speed from the `CHA_FAN` values.

Any advice would be very nice so that my computer can be a bit more quiet.

Willian-Zhang, 2024-06-01:
Same here with `Linux sage 5.15.0-107-generic` and `[    9.026988] nct6775: Found NCT6798D or compatible chip at 0x2e:0x290`