lm-sensors/lm-sensors #349: sensors service wakes up gGPU (amdgpu)

Strohhutpat, 2021-06-29:
Since Kernel 5.6 lm-sensors wakes up my dGPU from sleep,
the issue is on 5.6, 5.8 and 5.11

without lm-sensors:
> cat /sys/kernel/debug/vgaswitcheroo/switch 
0:DIS: :DynOff:0000:02:00.0
1:IGD:+:Pwr:0000:00:02.0

with lm-sensors:
> cat /sys/kernel/debug/vgaswitcheroo/switch 
0:DIS: :DynPwr:0000:02:00.0
1:IGD:+:Pwr:0000:00:02.0

> amdgpu-pci-0200
Adapter: PCI adapter
vddgfx:      900.00 mV 
edge:         +38.0°C  (crit = +94.0°C, hyst = -273.1°C)
power1:        3.11 W  (cap =  25.00 W)

on Kernel 5.4 the dGPU is always on sleep/standby with running lm-sensors

for example "sensors coretemp-isa-0000" doesnt wake up the dGPU

> dmesg | grep amdgpu
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-5.11.0-22-generic root=UUID=157485c3-2bb7-48d4-9cbc-1007f1204df3 ro rootflags=subvol=@ scsi_mod.use_blk_mq=1 quiet splash psmouse.synaptics_intertouch=0 amdgpu.runpm=1 amdgpu.dc=1
[    0.064984] Kernel command line: BOOT_IMAGE=/vmlinuz-5.11.0-22-generic root=UUID=157485c3-2bb7-48d4-9cbc-1007f1204df3 ro rootflags=subvol=@ scsi_mod.use_blk_mq=1 quiet splash psmouse.synaptics_intertouch=0 amdgpu.runpm=1 amdgpu.dc=1
[    1.597627] [drm] amdgpu kernel modesetting enabled.
[    1.597854] amdgpu: Topology: Add CPU node
[    1.598002] amdgpu 0000:02:00.0: amdgpu: Trusted Memory Zone (TMZ) feature not supported
[    1.667247] amdgpu 0000:02:00.0: amdgpu: Fetched VBIOS from ATRM
[    1.667250] amdgpu: ATOM BIOS: BR36111.001
[    1.685859] amdgpu 0000:02:00.0: amdgpu: VRAM: 2048M 0x000000F400000000 - 0x000000F47FFFFFFF (2048M used)
[    1.685862] amdgpu 0000:02:00.0: amdgpu: GART: 256M 0x000000FF00000000 - 0x000000FF0FFFFFFF
[    1.686058] [drm] amdgpu: 2048M of VRAM memory ready
[    1.686060] [drm] amdgpu: 3072M of GTT memory ready.
[    1.688191] amdgpu: hwmgr_sw_init smu backed is polaris10_smu
[    2.182551] amdgpu 0000:02:00.0: amdgpu: SE 2, SH per SE 1, CU per SH 5, active_cu_number 10
[    2.185389] amdgpu 0000:02:00.0: amdgpu: Using ATPX for runtime pm
[    2.185650] [drm] Initialized amdgpu 3.40.0 20150101 for 0000:02:00.0 on minor 1

valeri2000, 2021-07-02:
I have the same problem - on kernel 5.4 "sensors" doesn't even show the amd dgpu if it is not used which is expected, but on 5.10, 5.11, 5.12 there is a slight delay in the output of the command as if it wakes the dgpu and it shows it with temperature, power, etc regardless of whether it is being used or not.

Strohhutpat, 2021-07-08:
I also noticed the delay of the output command (Kernel 5.11)
(with disabled sensors service)
with cat /sys/kernel/debug/vgaswitcheroo/switch
you can see the wake up of the dGPU

flaviobn, 2022-08-08:
`sensors` is waking up AMD dGPU (kernel 5.18). This wastes battery in modern hybrid laptops (iGPU/dGPU).

Strohhutpat, 2023-09-19:
No new updates?