lm-sensors/lm-sensors #353: Gigabyte X570 Aorus Master not detected sensors correctly incorrect config

spstarr, 2021-07-14:
Sensors detect output:

{code}
# sensors-detect version 3.6.0
# System: Gigabyte Technology Co., Ltd. X570 AORUS MASTER [-CF]
# Kernel: 5.12.14-300.fc34.x86_64 x86_64
# Processor: AMD Ryzen 9 5950X 16-Core Processor (25/33/0)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 19h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
/dev/port: Operation not permitted

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): YES
/dev/port: Operation not permitted

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): YES
/dev/port: Operation not permitted

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): YES    
Using driver `i2c-piix4' for device 0000:00:14.0: AMD KERNCZ SMBus
Module i2c-dev loaded successfully.

Next adapter: SMBus PIIX4 adapter port 0 at 0b00 (i2c-0)
Do you want to scan it? (YES/no/selectively): YES
Client found at address 0x4a
Probing for `National Semiconductor LM75'...                No
Probing for `National Semiconductor LM75A'...               No
Probing for `Dallas Semiconductor DS75'...                  No
Probing for `National Semiconductor LM77'...                No
[... 100 lines cut ...]
Now follows a summary of the probes I have just done.
Just press ENTER to continue:  

Driver `adm1021':
  * Bus `NVIDIA i2c adapter 1 at b:00.0'
    Busdriver `nvidia', I2C address 0x2b
    Chip `Maxim MAX1617' (confidence: 3)

Driver `lm75':
  * Bus `SMBus PIIX4 adapter port 1 at 0b20'
    Busdriver `i2c_piix4', I2C address 0x4f
    Chip `Dallas Semiconductor DS75' (confidence: 3)

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no): 
Unloading i2c-dev... OK
{code}

It sure is not a lm75...

I have noted the `it87` modprobe option force_id=0x8728 which gives more sensors, but there is still no correct sensors config generated.

Any solutions?

spstarr, 2021-07-14:
With the module forced I see:

```
it8728-isa-0a40
Adapter: ISA adapter
in0:           1.01 V  (min =  +0.00 V, max =  +3.06 V)
in1:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.04 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.20 V  (min =  +0.00 V, max =  +3.06 V)
in5:         924.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           1.38 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.17 V  
fan1:        1227 RPM  (min =    0 RPM)
fan2:         531 RPM  (min =    0 RPM)
fan3:         494 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +31.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +39.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = disabled
intrusion0:  OK

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:            N/A  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +35.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +35.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +35.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)

it8728-isa-0a60
Adapter: ISA adapter
in0:           1.98 V  (min =  +0.38 V, max =  +1.60 V)
in1:         744.00 mV (min =  +0.60 V, max =  +0.88 V)
in2:           1.09 V  (min =  +1.09 V, max =  +1.21 V)
in3:          60.00 mV (min =  +0.00 V, max =  +3.06 V)
in4:           1.98 V  (min =  +1.92 V, max =  +2.04 V)
in5:           1.31 V  (min =  +1.57 V, max =  +1.74 V)
in6:           3.06 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.67 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.53 V  
fan1:         501 RPM  (min =    0 RPM)
fan2:         516 RPM  (min =    0 RPM)
fan3:         470 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

hidpp_battery_0-hid-3-8
Adapter: HID adapter
in0:           4.17 V  

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +39.5°C  
Tdie:         +39.5°C  
Tccd1:        +36.0°C  
Tccd2:        +32.8°C  

nvme-pci-0100
Adapter: PCI adapter
Composite:    +34.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +34.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +34.9°C  (low  = -273.1°C, high = +65261.8°C)
```

jared-mackey, 2021-07-14:
Getting same results. Would love to see EC_TEMP_1 in detected by sensors.

Wer-Wolf, 2021-12-22:
With secure boot enabled, sensors-detect cannot probe for certain superio chips, since ioport access is disabled.
Maybe you can try and temporally disable secure boot?

spstarr, 2022-01-04:
With secure boot disabled, I didn't try but will. even if it isn't detecting it. Do we have a valid config for this board?

frankcrawford, 2022-02-13:
The problem you are seeing is that there are two different chips on this M/B an ITE8792E and an ITE8688E, and they do present different info (e.g. one has 3 fans and 3 temps, the other has 5 fans and 6 temps).

I've got a git repo (https://github.com/frankcrawford/it87) that I have tried to capture recent chips based on earlier work by others., and you really need this to reasonable values.

However I am never certain if it is correct, and would love others to check out the values.

This is what I get:

```
it8792-isa-0a60
Adapter: ISA adapter
in0:           1.79 V  (min =  +0.00 V, max =  +2.78 V)
in1:         600.00 mV (min =  +0.00 V, max =  +2.78 V)
in2:         981.00 mV (min =  +0.00 V, max =  +2.78 V)
+3.3V:         3.36 V  (min =  +0.00 V, max =  +5.56 V)
in4:           1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:           1.18 V  (min =  +0.00 V, max =  +2.78 V)
in6:           2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:          3.36 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:          3.29 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +47.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +39.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +47.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +47.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +54.9°C  (low  = -273.1°C, high = +80.8°C)
                       (crit = +80.8°C)
Sensor 1:     +54.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +75.8°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)

it8688-isa-0a40
Adapter: ISA adapter
in0:           1.42 V  (min =  +0.00 V, max =  +3.06 V)
in1:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:         996.00 mV (min =  +0.00 V, max =  +3.06 V)
in5:         912.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           1.22 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.34 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.24 V  
fan1:        2872 RPM  (min =    0 RPM)
fan2:        1054 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:        2156 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +32.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +71.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +52.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +48.0°C  (low  =  +0.0°C, high = -125.0°C)  sensor = thermistor
temp6:        +58.0°C  (low  = -16.0°C, high = +90.0°C)  sensor = thermistor
intrusion0:  OK

k10temp-pci-00c3
Adapter: PCI adapter
Core0 Temp:   +71.1°C  
Core1 Temp:   +59.8°C  
Core1 Temp:   +52.0°C  

nvme-pci-0100
Adapter: PCI adapter
Composite:    +51.9°C  (low  = -273.1°C, high = +80.8°C)
                       (crit = +80.8°C)
Sensor 1:     +51.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +75.8°C  (low  = -273.1°C, high = +65261.8°C)
```