lm-sensors/lm-sensors #357: All JSON output values converted to Fahrenheit regardless of units when both -j and -f are specified

Alives, 2021-07-31:
When `-f` is specified along with `-j`, all output is converted to fahrenheit regardless of whether it's a temperature unit or not.

Both `-j` and `-f` specified:
```
      "fan1":{
         "fan1_input": 1106.600,
         "fan1_min": 32.000,
         "fan1_alarm": 32.000,
         "fan1_beep": 32.000,
         "fan1_pulses": 35.600
      },
```
Only `-j` specified:
```
      "fan1":{
         "fan1_input": 613.000,
         "fan1_min": 0.000,
         "fan1_alarm": 0.000,
         "fan1_beep": 0.000,
         "fan1_pulses": 2.000
      },
```

cpackham-atlnz, 2021-08-13:
Also affects raw output `sensors -u`

HorlogeSkynet, 2021-08-28:
You're right, (partially) my bad when I proposed #196 for missing Fahrenheit conversion. I hope your PR will get reviewed and merged.

mlmatlock, 2023-12-30:
Still here...

eremem, 2024-05-14:
Any chance on integrating the existing patch into a release version any time soon?