lm-sensors/lm-sensors #368: Got some Error outputs after installing lm-sensors.

Ezmanatvancouver, 2021-09-29:
installed lm-sensors with ubuntu running on AMD 3960x CPU and Asus ROG STRIX motherboard. Also, ran sensor-detect with all possible modules installed. i re-install ubuntu a few times and tried same process just in case i did something wrong that i didnt notice. but resulted the same. here is my output, hoping someone could give me some advise on how to fix those. Thanks

nvme-pci-2100
Adapter: PCI adapter
Composite:    +38.9°C  (low  =  -0.1°C, high = +74.8°C)
                       (crit = +79.8°C)
ERROR: Can't get value of subfeature temp2_min: I/O error
ERROR: Can't get value of subfeature temp2_max: I/O error
Sensor 1:    -216.2°C  (low  =  +0.0°C, high =  +0.0°C)
ERROR: Can't get value of subfeature temp3_min: I/O error
ERROR: Can't get value of subfeature temp3_max: I/O error
Sensor 2:    -234.2°C  (low  =  +0.0°C, high =  +0.0°C)
ERROR: Can't get value of subfeature temp4_min: I/O error
ERROR: Can't get value of subfeature temp4_max: I/O error
Sensor 3:    -234.2°C  (low  =  +0.0°C, high =  +0.0°C)

Wer-Wolf, 2021-12-22:
No worry, it seems that your nvme drive does not support reporting of min. /max. temperature, and returns an IO-Error instead.
Its your drive which seems to advertise but not support reading of min./max. temperature, maybe you can check if your drive runs the latest firmware?

Wer-Wolf, 2021-12-22:
Also the remaining values look quite strange, maybe a firmware update could help.

steve-pingpong, 2022-10-08:
I get 
```
                       (crit = +79.8°C)
ERROR: Can't get value of subfeature temp3_min: I/O error
ERROR: Can't get value of subfeature temp3_max: I/O error
Sensor 2:     +32.9°C  (low  =  +0.0°C, high =  +0.0°C)
ERROR: Can't get value of subfeature temp4_min: I/O error
ERROR: Can't get value of subfeature temp4_max: I/O error
Sensor 3:     +43.9°C  (low  =  +0.0°C, high =  +0.0°C)
ERROR: Can't get value of subfeature temp5_min: I/O error
ERROR: Can't get value of subfeature temp5_max: I/O error
Sensor 4:     +30.9°C  (low  =  +0.0°C, high =  +0.0°C)
ERROR: Can't get value of subfeature temp6_min: I/O error
ERROR: Can't get value of subfeature temp6_max: I/O error
Sensor 5:     +30.9°C  (low  =  +0.0°C, high =  +0.0°C)
ERROR: Can't get value of subfeature temp7_min: I/O error
ERROR: Can't get value of subfeature temp7_max: I/O error
Sensor 6:     +33.9°C  (low  =  +0.0°C, high =  +0.0°C)

```
so my values are good but I sure wish I could hide the min and max errors.

Wer-Wolf, 2022-10-08:
Edit sensors.conf and hide the min and max attributes

luquino386, 2023-02-18:
Hi, I have the same problem, but I can't find a sensors.conf with the nvme device.
this is my output

luca@pc-sala:~$ sensors
k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +45.1°C  

amdgpu-pci-0800
Adapter: PCI adapter
vddgfx:           N/A  
vddnb:            N/A  
edge:         +45.0°C  

acpitz-acpi-0
Adapter: ACPI interface
nvme:         +16.8°C  (crit = +20.8°C)

it8686-isa-0a40
Adapter: ISA adapter
CPU Vcore:     696.00 mV (min =  +0.00 V, max =  +3.06 V)
+3.3v:           2.05 V  (min =  +0.00 V, max =  +3.06 V)
+12v:            2.04 V  (min =  +0.00 V, max =  +3.06 V)
+5v:             2.03 V  (min =  +0.00 V, max =  +3.06 V)
CPU Vcore SOC: 876.00 mV (min =  +0.00 V, max =  +3.06 V)
CPU Vddp:      912.00 mV (min =  +0.00 V, max =  +3.06 V)
DRAM A/B:        1.22 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:            3.29 V  (min =  +0.00 V, max =  +6.12 V)
Battery:         3.05 V  
CPU_FAN:       1967 RPM  (min =   10 RPM)
SYS_FAN:       1991 RPM  (min =    0 RPM)
MOBO:           +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
Chipset:        +51.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
CPU Socket:     +45.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
PCI-EX16:       +19.0°C  (low  =  +0.0°C, high = +127.0°C)  sensor = thermistor
VRM MOS:        +46.0°C  (low  =  +0.0°C, high = -120.0°C)  sensor = thermistor
vSOC MOS:       +44.0°C  (low  =  +0.0°C, high = -120.0°C)  sensor = thermistor
intrusion0:    ALARM

nvme-pci-0700
Adapter: PCI adapter
Composite:    +36.9°C  (low  =  -0.1°C, high = +69.8°C)
                       (crit = +84.8°C)
ERROR: Can't get value of subfeature temp2_min: I/O error
ERROR: Can't get value of subfeature temp2_max: I/O error
Sensor 1:     +54.9°C  (low  =  +0.0°C, high =  +0.0°C)
ERROR: Can't get value of subfeature temp3_min: I/O error
ERROR: Can't get value of subfeature temp3_max: I/O error
Sensor 2:     +42.9°C  (low  =  +0.0°C, high =  +0.0°C)

I found these conf files

luca@pc-sala:~$ ls /etc/sensors.d
acpitz-acpi-0  it8686-isa-0a40

luca@pc-sala:~$ cat /etc/sensors.d/it8686-isa-0a40 
chip "it8686-isa-0a40"
  label temp1 "MOBO"
  label temp2 "Chipset"
  label temp3 "CPU Socket"
  label temp4 "PCI-EX16"
  label temp5 "VRM MOS"
  label temp6 "vSOC MOS"
  label in0 "CPU Vcore"
  label in1 "+3.3v"
  label in2 "+12v"
  label in3 "+5v"
  label in4 "CPU Vcore SOC"
  label in5 "CPU Vddp"
  label in6 "DRAM A/B"
  label in7 "3VSB"
  label in8 "Battery"
  label fan1 "CPU_FAN"
  label fan2 "SYS_FAN"

  compute temp3 @+15,@+15
  compute in1 @*1.65,@*1.65
  compute in2 @*6,@*6
  compute in3 @*2.5,@*2.5
luca@pc-sala:~$ 

and

luca@pc-sala:~$ cat /etc/sensors.d/acpitz-acpi-0 
chip "acpitz-acpi-0"

label temp1 "nvme"
luca@pc-sala:~$ 

luca@pc-sala:~$ 
luca@pc-sala:~$ ls /etc/sensors*
/etc/sensors3.conf      (this is very large)

in none of those conf files there is the label  nvme-pci-0700

How can I fix it?

Wer-Wolf, 2023-02-18:
Create a new config file for your nvme device?

Wer-Wolf, 2023-02-18:
Use `ignore` statements to block certain attributes like temp1_max, etc.

Maybe you can also create a bugreport on bugzilla.kernel.org if the issues cannot be resolved with a firmware update.

Ezmang, 2023-02-20:
i found out it was the ssd.

luquino386, 2023-02-20:
> Create a new config file for your nvme device?

HI Wolf, can you help me in creating it?

luquino386, 2023-02-21:
> > Create a new config file for your nvme device?
> 
> HI Wolf, can you help me in creating it?

Ok, I created a new conf file for nvme-pci-0700, it works because I can change the label name of temp1. But the errors persist.

luquino386, 2023-02-21:
> 

Can you, please, give me an example of `ignore` statement?

Wer-Wolf, 2023-02-21:
See Pull-request #424 for an example on how to use those statenents.

luquino386, 2023-02-21:
Ok, I tried some tricks but no luck, the error persist, finally I made a workaround and I get the temp from a conky so I don't see the errors. Sorry but I have no time to spend on it.
Thank you for your help and for you work.

ATLief, 2023-03-01:
@luquino386 I think I found at least a partial solution. If the error mentions "temp3_max" then add an "ignore" statement for "temp3" (you can't seem to only disable "temp3_max" etc).

In your particular case, add the following to /etc/sensors.d/whatever.conf:
```
chip "nvme-pci-2100"
    ignore temp2
    ignore temp3
    ignore temp4
```

luquino386, 2023-03-01:
> @luquino386 I think I found at least a partial solution. If the error mentions "temp3_max" then add an "ignore" statement for "temp3" (you can't seem to only disable "temp3_max" etc).
>
> In your [...]

Fine, that works, thx a lot ATLief.
I tried something similar, but my error was that I wrote `ignore temp2_min` and so on, one line per each min and max.

itoffshore, 2025-05-19:
the fix above worked for me too - adding a note to help others searching for a similar error

```
 /usr/lib/gdm-x-session [ERR] lmsensor: Cannot get value of subfeature temp1_input: Kernel interface error.
```
I was using `psensor` for temperature monitoring in XFCE - which gave me the name of the "chip" under the details for `temp1` in `psensor`

* Creating an override file `/etc/sensors.d/acpitz-acpi-0.conf` solved it with:
```
chip "acpitz-acpi-0"
  ignore temp1
  ignore temp2
```

DeboBurro, 2025-12-23:
It works but also removes all the information.
```
nvme-pci-0200
Adapter: PCI adapter
Composite:    +52.9°C  (low  = -20.1°C, high = +84.8°C)
                       (crit = +94.8°C)
ERROR: Can't get value of subfeature temp2_min: I/O error
ERROR: Can't get value of subfeature temp2_max: I/O error
Sensor 1:     +70.8°C  (low  =  +0.0°C, high =  +0.0°C)
```
after I set up the configuration
```
nvme-pci-0200
Adapter: PCI adapter
```
is there a way to only remove the error but preserve the sensor temperature reading?