lm-sensors/lm-sensors #373: How to PR hwmon? ITE IT8686E - Help an aspiring dev

bai-yi-bai, 2021-12-05:
Hello,
I'm interested in adding support for the "Gigabyte Technology Co., Ltd. X299 AORUS Gaming 3 Pro"... but I have never created a pull request for something in the Linux kernel before, so I could use some assistance. I have enough Linux knowledge that I feel I could do this.

### Situation
I have installed several PWM-capable fans in my system and need to add support for this fan controller since my SATA SSDs are reaching 66 C under heavy writes. This results in several warnings in the Unraid Web GUI and the system appears to stall and fall to 0 MB/s. My guess is that the drives throttle to allow the temperature to fall before starting to write again. The motherboard BIOS does spin up the CPU fan as load increases, but of course, it doesn't monitor the SATA SSDs. My goal is to get my fans to speed up and then slow down.

I am running 5.10.28-Unraid x86_64. I want to use the Dynamix Autofan plug-in and have installed Perl support to detect the sensors (https://wiki.unraid.net/Setting_up_CPU_and_board_temperature_sensing), however, when I run 'sensors-detect' from the command line, I get this output:

`Probing for Super-I/O at 0x2e/0x2f
Trying family `ITE'...                                      Yes
Found `ITE IT8686E Super IO Sensors'                        Success!
    (address 0xa40, driver `to-be-written')
Probing for Super-I/O at 0x4e/0x4f
Trying family `ITE'...                                      Yes
Found `ITE IT8792E Super IO Sensors'                        Success!
    (address 0xa60, driver `it87')`

- I can confirm that the IT8686E is not listed in the it87.c file
https://github.com/torvalds/linux/blob/master/drivers/hwmon/it87.c

- I cannot locate a datasheet for the ITE IT8686E

- While writing this post, I found a solution (in Chinese): force the system to use the 8628 driver. ('acpi_enforce_resources = lax' in the configuration and then run "sudo modprobe it87 force_id=0x8628") 
https://post.smzdm.com/p/a270z6g2/
The impacted motherboard seems to be a Gigabyte Z370M D3H-CF

I am about to try this solution... however I don't feel it's ideal. Plus, I'm not sure if it captures all the capabilities of the 8686E. I would like to attempt to build my own it87.c file... but again, I've never done this before, all I need is a push in the right direction.

bai-yi-bai, 2021-12-05:
Follow-up
The solution worked, but still, native support would be better.

Before only the coretemp-isa-0000 temperatures would be reported.
```
sensors
it8628-isa-0a60
Adapter: ISA adapter
in0:           1.96 V  (min =  +0.00 V, max =  +3.06 V)
in1:         156.00 mV (min =  +0.00 V, max =  +3.06 V)
in2:           1.14 V  (min =  +0.00 V, max =  +3.06 V)
+5V:           3.70 V  (min =  +0.00 V, max =  +6.12 V)
in4:           1.15 V  (min =  +0.00 V, max =  +3.06 V)
in5:         960.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           3.06 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.67 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.55 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +38.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp4:       -128.0°C  
temp5:        -93.0°C  
temp6:         -2.0°C  
intrusion0:  ALARM

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +58.0°C  (high = +76.0°C, crit = +86.0°C)
Core 0:        +45.0°C  (high = +76.0°C, crit = +86.0°C)
Core 1:        +44.0°C  (high = +76.0°C, crit = +86.0°C)
Core 2:        +58.0°C  (high = +76.0°C, crit = +86.0°C)
Core 3:        +47.0°C  (high = +76.0°C, crit = +86.0°C)
Core 4:        +48.0°C  (high = +76.0°C, crit = +86.0°C)
Core 8:        +47.0°C  (high = +76.0°C, crit = +86.0°C)
Core 9:        +47.0°C  (high = +76.0°C, crit = +86.0°C)
Core 10:       +46.0°C  (high = +76.0°C, crit = +86.0°C)
Core 11:       +44.0°C  (high = +76.0°C, crit = +86.0°C)
Core 16:       +46.0°C  (high = +76.0°C, crit = +86.0°C)
Core 17:       +45.0°C  (high = +76.0°C, crit = +86.0°C)
Core 18:       +42.0°C  (high = +76.0°C, crit = +86.0°C)
Core 19:       +47.0°C  (high = +76.0°C, crit = +86.0°C)
Core 20:       +47.0°C  (high = +76.0°C, crit = +86.0°C)
Core 24:       +46.0°C  (high = +76.0°C, crit = +86.0°C)
Core 25:       +46.0°C  (high = +76.0°C, crit = +86.0°C)
Core 26:       +46.0°C  (high = +76.0°C, crit = +86.0°C)
Core 27:       +45.0°C  (high = +76.0°C, crit = +86.0°C)

it8628-isa-0a40
Adapter: ISA adapter
in0:           1.82 V  (min =  +0.00 V, max =  +3.06 V)
in1:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.06 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.22 V  (min =  +0.00 V, max =  +3.06 V)
in5:         900.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           1.24 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.41 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.19 V  
fan1:        1726 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:        1002 RPM  (min =    0 RPM)
fan4:         730 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +43.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +56.0°C  (low  =  +0.0°C, high = +80.0°C)  sensor = disabled
temp4:        +38.0°C  
temp5:        +47.0°C  
temp6:        +34.0°C  
intrusion0:  ALARM
```

Wer-Wolf, 2021-12-22:
You must make sure that modprobe loads the it87 mode with "force_id=0x8628" on every boot, better have a look at /etc/modprobe .d/ (https://linux.die.net/man/5/modprobe.d).
Also as long as no datasheet is available, the kernel devs can only resort to reverse engineering, which is difficult.

frankcrawford, 2022-02-13:
If you want to match the correct modules, you can try my it87 modules, which is based on previous work by others at https://github.com/frankcrawford/it87

It includes definitions for the IT8686E and IT8792E.

However, you will need to learn how to handle an out of tree modules, and I would suggest using dkms (config included).

bai-yi-bai, 2022-02-22:
Thank you Frank. When I have some time, I will dedicate it to investigating this.
The real questions are:
1. How has the company ITC been able to skirt by for 20 years without actively contributing to open source movements.
2. How do I pass this off to the maintainers of Arch Linux or Unraid.

frankcrawford, 2022-02-23:
@bai-yi-bai Well, to answer your two questions:
1) I'm not happy with it either, but there is no requirement for them to support open source. and
2) I'm exploring how to get the updates into the kernel, the current maintainer has sort of abandoned it87.

Pythonic-Rainbow, 2023-09-16:
Yep the Chinese post works. You can also set the module parameter on boot by passing it as a command line options.
[Arch Wiki as a reference](https://wiki.archlinux.org/title/Kernel_module#Using_kernel_command_line)