lm-sensors/lm-sensors #375: Question about sensors_get_detected_chips and supported subfeatures

bou6, 2021-12-07:
using the command `sensors` I am getting the following output which is nice : 
`Adapter: ISA adapter
Immediate temperature for site 0:  +73.0 C
Average temperature for site 0:    +72.0 C
Immediate temperature for site 1:  +73.0 C
Average temperature for site 1:    +72.0 C
Immediate temperature for site 2:  +72.0 C
Average temperature for site 2:    +72.0 C`

Looking at `sysfs` I have follwoing files:
`ls /sys/class/hwmon/hwmon0/
device  name  of_node  power  subsystem  temp1_input  temp1_label  temp2_input  temp2_label  temp3_input  temp3_label  temp4_input  temp4_label  temp5_input  temp5_label  temp6_input temp6_label  uevent`.

till Now everything is Ok, I also checked the driver Implementation and datasheet of the Soc, the soc has 3 temperature sites and temp<site_nb>, temp<site_nb+1> corresponds to Immediate and average temperature.

Now I have a c application where I call the library, following is my code:
`
int sensors_ret=0;
int idx_chips=0;
int index_int=0;

sensors_ret = sensors_init(NULL);
if (sensors_ret != 0)
{
	printf("sensors_init returns 0x%0x \n", sensors_ret);
	return -1;
}

const sensors_chip_name* chip_names_pst = sensors_get_detected_chips(NULL,&idx_chips);
if (NULL == chip_names_pst)
{
	printf("sensors_get_detected_chips return NULL List\n");
	return -1;
}
printf("sensors_get_detected_chips returns a valid list idx_chips = 0x%0x\n", idx_chips);
`
The problem is I am getting a number of chips = 1, I expected having 3, did I miss something ?
My second questions is why there is no subfeature `SENSORS_SUBFEATURE_TEMP_AVERAGE` and `SENSORS_SUBFEATURE_TEMP_IMMEDIATE` in `sensors_subfeature_type`, that confuses me because using sensors is correct it detects those 2 subfeatures but they are not part of the Library API

Wer-Wolf, 2021-12-22:
"Immediate temperature for site X:" and "Average temperature for site X:" are the labels for the temperatures 0 - 6.
Also each sensor chip has its own entry under /sys/class/hwmon/hwmonX, so your libsensors only detecting a single sensor chip is totally fine, what you want is the number of features of this particular sensor chip, which is 3 (each temperature sensor channel is a SENSORS_FEATURE_TEMP). Each sensor channel has in your case SENSORS_SUBFEATURE_TEMP_INPUT, while the label can be retrieved with sensors_get_label(const sensors_chip_name *name, const sensors_feature *feature).

Wer-Wolf, 2021-12-22:
Tbh the labeling of the temperature sensors is because there are no standardized  attributes for "average temperature", so the driver uses the labels as a sort of workaround if i understand correctly.

Whats the name of this sensor chip (content of /sys/class/hwmon/hwmon0/name)?

bou6, 2022-01-28:
The name of the sensor is `s32tmu`, I just take a look at how lmsensor is looking for the temperature subfeatures, lmsensor is considering the name of the files in `/sys/class/hwmon/hwmon0/` and comparing them to the names in `temp_matches`.
You just mentioned "standardized attributes", any hints for that, I am pretty curious about it?

Wer-Wolf, 2022-01-28:
All standardized attributes are documented in https://www.kernel.org/doc/Documentation/hwmon/sysfs-interface,
but by using libsensors, you should not worry about this.

First, you should populate a sensors_chip_name matching the chipname you are searching for: s32tmu:
sensors_chip_name match ={
	.prefix = "s32tmu",
	.bus = {
                .type = SENSORS_BUS_TYPE_ANY,
                .SENSORS_BUS_NR_ANY,
        },
	.addr = SENSORS_CHIP_NAME_ADDR_ANY,
};

Wer-Wolf, 2022-01-28:
Then you should use sensors_get_detected_chips(&match, &idx_chips),
which returns to you the first instance of a sensor chip called "s32tmu"
or NULL if no matching chip was found.

Wer-Wolf, 2022-01-28:
Now you can use sensors_get_feature(&<sensor chip instance>, &nr) to iterate over all sensor features the chip offers you.
You should check if the returned feature is NULL since this would indicate that there are no more features to iterate over.
Since i suppose your interested in the sensor features representing temperature sensors, you can check if
feature->type equals SENSORS_FEATURE_TEMP to filter unnecessary features.

Wer-Wolf, 2022-01-28:
As you will see, the chip should offer you 6 features with the SENSORS_FEATURE_TEMP type.
You can read the temperature sensor label with sensors_get_label(&chip, &feature) and the temperature input subfeature with
sensors_get_subfeature(&chip, &feature, SENSORS_SUBFEATURE_TEMP_INPUT) and
sensors_get_value(&chip, subfeature->nr, &value).

Wer-Wolf, 2022-01-28:
See here for some information regarding the functions:
https://linux.die.net/man/3/libsensors