lm-sensors/lm-sensors #376: Error CPU number detected in intel 12nd alder lake

asas1asas200, 2021-12-21:
I found my htop temp only display on CPU number 0, 4, 8, 12, and I run `sensors` command it output as below:

```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +25.0°C  (high = +80.0°C, crit = +100.0°C)
Core 4:        +23.0°C  (high = +80.0°C, crit = +100.0°C)
Core 8:        +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 12:       +24.0°C  (high = +80.0°C, crit = +100.0°C)
Core 16:       +26.0°C  (high = +80.0°C, crit = +100.0°C)
Core 20:       +22.0°C  (high = +80.0°C, crit = +100.0°C)
Core 28:       +23.0°C  (high = +80.0°C, crit = +100.0°C)
Core 29:       +23.0°C  (high = +80.0°C, crit = +100.0°C)
Core 30:       +23.0°C  (high = +80.0°C, crit = +100.0°C)
Core 31:       +23.0°C  (high = +80.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +105.0°C)

nvme-pci-e100
Adapter: PCI adapter
Composite:    +33.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +33.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +44.9°C  (low  = -273.1°C, high = +65261.8°C)
```

And my hardware list:
+ CPU: i5-12600KF
+ Board: ASUS Z690-PLUS D4

lm_sensor version: sensors version 3.6.0+git with libsensors version 3.6.0+git

Wer-Wolf, 2021-12-22:
The coretemp driver seems to get confused over which core has which number.
Whats your kernel version?

asas1asas200, 2021-12-22:
My kernel: `5.15.10-arch1-1`

Btw I use arch.

Wer-Wolf, 2021-12-23:
Maybe you should file a bugreoprt on bugzilla.kernel.org

Wer-Wolf, 2022-01-02:
I found this article, maybe it can help:
https://wiki.archlinux.org/title/lm_sensors#Example_3._Renumbering_cores_for_multi-CPU_systems