lm-sensors/lm-sensors #378: GPU Fan always 0 rpm for AMD Rx 6600 xt

Moelf, 2022-01-21:
similar to:
- https://github.com/lm-sensors/lm-sensors/issues/134

during furmark, lm_sensors:
```
~> sensors
amdgpu-pci-0e00
Adapter: PCI adapter
vddgfx:      825.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3500 RPM)
edge:         +70.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +82.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +76.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
power1:      125.00 W  (cap = 130.00 W)
```

however fan is not 0 (I can physically see it), and:
```
cat /sys/class/drm/card0/device/hwmon/hwmon4/pwm1_enable
2
~> cat /sys/class/drm/card0/device/hwmon/hwmon4/pwm1
112
```

worth mentioning the GPU does have 0 rpm at idle, in that case:
```
~> cat /sys/class/drm/card0/device/hwmon/hwmon4/pwm1
0
```

Wer-Wolf, 2022-01-28:
Please note that the RPM of the fan is stored in fanX_input, not in pwm1.
pwmX_enable containing "2" means your fan speed is controlled by the video card,
and pwmX containing "112" means your fan is driven at 112/255 speed by the video card.
Maybe the fan does not report the rpm back to the card, so "0" is always displayed?
Whats the output of the following command:
cat /sys/class/drm/card0/device/hwmon/hwmon4/fan*_input
?

Moelf, 2022-10-23:
```
~> cat /sys/class/drm/card0/device/hwmon/hwmon6/fan1_input
0
```

Wer-Wolf, 2022-10-23:
If fan1_input does always 0 rpm even if the fan is spinning, then it could be possible that the fan does not report its current rpm value.
Do system monitoring tools on windows like hwinfo64 display the fan rpm? If not, then the fan does not support reporting of its rpm value.

CiscoDeTours, 2024-05-11:
Mobile version of the RX 6600 on my HP Laptop has the same problem:

~~~
~$ sensors

hp-isa-0000
Adapter: ISA adapter
fan1:        2406 RPM
fan2:        2213 RPM

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +40.5°C  

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:       18.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3300 RPM)
edge:         +41.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +41.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
mem:          +40.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
PPT:           5.00 W  (cap =  90.00 W)
~~~

`hp-isa-0000` shows the value while `amdgpu-pci-0300` remains fixed at zero.

furkanonder, 2024-07-02:
The Alpha 15 B5EEK RX6600M has the same problem.

```
k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +63.8°C

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      712.00 mV
fan1:           0 RPM  (min =    0 RPM, max = 3300 RPM)
edge:         +57.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +57.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
mem:          +54.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
PPT:           6.00 W  (cap =  65.00 W)

BAT1-acpi-0
Adapter: ACPI interface
in0:          16.21 V
curr1:         0.00 A

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +49.0°C

amdgpu-pci-0700
Adapter: PCI adapter
vddgfx:        1.45 V
vddnb:       962.00 mV
edge:         +55.0°C
PPT:          33.00 W

nvme-pci-0600
Adapter: PCI adapter
Composite:    +39.9°C  (low  =  -0.1°C, high = +86.8°C)
                       (crit = +89.8°C)
Sensor 1:     +39.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +60.0°C
```