lm-sensors/lm-sensors #380: IT8587

giovannicaligaris, 2022-01-25:
Any chance of providing support for 0x8587 ITE? All Clevo laptops use this to control the fan.

Wer-Wolf, 2022-01-28:
Afaik, its hard or even impossible to gain access to the datasheets of such ITE chips,
and without them there will be likely no support.
Sorry

Wer-Wolf, 2022-01-28:
Also please note that support for individual hardware is an issue regarding the drivers inside the linux kernel.

giovannicaligaris, 2022-01-28:
I found this online, does it help
[IT8587.pdf](https://github.com/lm-sensors/lm-sensors/files/7960002/IT8587.pdf)
?

Source: https://repowebapps.blogspot.com/2019/10/kbc-ite-it8587-schematic.html

Wer-Wolf, 2022-01-28:
That is a circuit plan, not a Datasheet, sry.

viktorxda, 2022-08-07:
Here is a datasheet for the: [IT5576E-128.pdf](https://github.com/lm-sensors/lm-sensors/files/9276079/IT5576E-128.pdf)
It seems IT5570/IT5576/IT8528/IT8587 are somewhat similar.

junocomp, 2022-10-07:
any updates on this?

Vitamin-X, 2023-03-14:
the  tuxedo-control-center got support for it... may take a look into thier sources?

ntzb, 2023-04-16:
@Vitamin-X which chip do you mean exactly?
also can please share how you know that tuxedo has support for it? where did you see it?

Vitamin-X, 2023-05-10:
i got an IT5570 in my notebook and its controllable by the Tuxedo software

junocomp, 2023-05-10:
@Vitamin-X any chance of making a it5570e-dkms for this and I can try to test it?

Vitamin-X, 2023-05-10:
the tuxedo-keyboard pakage got one its part of thier controll center 

https://github.com/tuxedocomputers/tuxedo-keyboard

junocomp, 2023-05-10:
But that's the entire tuxedo-keyboard package with it5570e.

ntzb, 2023-05-10:
interesting..
I have it5571 which might be a related/similar chip.
should installing the `tuxedo_keyboard` module help make the EC/sensor detectable? I tried but it didn't seem to make a difference (it might be due to it not being it5571 but it5570e though)

Vitamin-X, 2023-05-10:
yea up here the fans are also not shown in lm-sensors

if you got an Clevo notebook, have you tryed the full tuxedo controll software? 
its not that bad

btw:
when you search on google for your notebook and add  site:e-weekly.co.uk you can find the service manual with schematics to see whats all connected to the Embedded Controller

thats the one of my notebook https://www.e-weekly.co.uk/download/RnD/DRIVERS/CLEVO/PCxxHxx/ESM7.pdf

ntzb, 2023-05-10:
I don't have a clevo or tuxedo hardware, it's an ASUS laptop.
seems like schematics won't help advancing any support in lm-sensors for IT557X devices.

Vitamin-X, 2023-05-10:
oh the person switched haha i ment the other one 

but do asus notebooks uses these ITE  chips  too?
so far i notice that asus i way more friendly against linux... maybe they got some similar stuff

ntzb, 2023-05-10:
nothing linux friendly from asus.
some of their laptops use ITE chips. 
mine uses IT5570 unfortunately.

junocomp, 2023-05-10:
https://github.com/a1wong/it87

Vitamin-X, 2023-05-11:
annoying the creator ? haha

but what exacly do they need to support them? currently rewriting that system76 EC flash and i could provide some details.

never user lm-sensors much, and came just randomly by here, cuz google found it while i was searching for that chip

ntzb, 2023-05-11:
I'm not sure what's needed, but if tuxedo/coreboot got it working with their clients, it seems doable for lm-sensors.
I tried poking around other forks like here https://github.com/frankcrawford/it87/issues/8 but it seems it557x are too different from the it87 devices to get any help.

Vitamin-X, 2023-05-12:
https://github.com/Vitamin-X/ec/blob/master/src/board/system76/common/smfi.c#L355

some code running inside the IT5570 to communicate with the PCH ... it executes some commands these also include getFanSpeed

and it seems like it runs on SPI

[image]
[image]