lm-sensors/lm-sensors #388: Intel(R) Xeon(R) CPU E5-2670 v2

ahmaddxb, 2022-02-22:
It detects 12 cores on a 10 core CPU, anyone know why this is?
https://ark.intel.com/content/www/us/en/ark/products/75275/intel-xeon-processor-e52670-v2-25m-cache-2-50-ghz.html

root@dellr720xd-pve:~# sensors
coretemp-isa-0001
Adapter: ISA adapter
Package id 1:  +41.0°C  (high = +85.0°C, crit = +95.0°C)
Core 0:        +40.0°C  (high = +85.0°C, crit = +95.0°C)
Core 1:        +32.0°C  (high = +85.0°C, crit = +95.0°C)
Core 2:        +39.0°C  (high = +85.0°C, crit = +95.0°C)
Core 3:        +38.0°C  (high = +85.0°C, crit = +95.0°C)
Core 4:        +39.0°C  (high = +85.0°C, crit = +95.0°C)
Core 8:        +41.0°C  (high = +85.0°C, crit = +95.0°C)
Core 9:        +36.0°C  (high = +85.0°C, crit = +95.0°C)
Core 10:       +41.0°C  (high = +85.0°C, crit = +95.0°C)
Core 11:       +37.0°C  (high = +85.0°C, crit = +95.0°C)
Core 12:       +35.0°C  (high = +85.0°C, crit = +95.0°C)

ahmaddxb, 2022-02-22:
Sorry it is correctly showing 10 cores in total but the numbering is wrong core 5 to 7 are missing and instead has added them to then end from 10 to 12

shimakazevr, 2022-12-03:
i can also confirm this
dell r720 duel E5-2670 v2 (10c20t)
https://i.imgur.com/LhlxOSM.png