lm-sensors/lm-sensors #392: I can't use sensors to show my fan speed and to regulate it（Inter11th N5095）

zailushang2008, 2022-04-15:
**Problem description:**
I can't use sensors to show my fan speed and to regulate it.

**My system:**
Linux ontheroad 4.4.180+ #42218 SMP Mon Oct 18 19:17:56 CST 2021 x86_64 GNU/Linux synology_apollolake_918+

**My motherboard:**
https://ark.intel.com/content/www/cn/zh/ark/products/212322/intel-celeron-processor-n5095-4m-cache-up-to-2-90-ghz.html
http://www.maxsun.com.cn/2021/1109/5624.html

**The mainboard supports PWM speed regulation（CPU Fan and SYS Fan）：**
[image]
[image]

**I used this method to install sensors on my Synology:**
mkdir /temp
cd /temp
wget http://ipkg.nslu2-linux.org/feeds/optware/syno-i686/cross/unstable/syno-i686-bootstrap_1.2-7_i686.xsh
chmod +x syno-i686-bootstrap_1.2-7_i686.xsh
sh syno-i686-bootstrap_1.2-7_i686.xsh
ipkg update
ipkg install lm-sensors

**Run the command:**
root@ontheroad:~# sensors
coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +47.0 C  (high = +105.0 C, crit = +105.0 C)
Core 0:         +44.0 C  (high = +105.0 C, crit = +105.0 C)
Core 1:         +44.0 C  (high = +105.0 C, crit = +105.0 C)
Core 2:         +44.0 C  (high = +105.0 C, crit = +105.0 C)
Core 3:         +44.0 C  (high = +105.0 C, crit = +105.0 C)

root@ontheroad:~# sensors-detect
-ash: /opt/sbin/sensors-detect: /usr/bin/perl: bad interpreter: No such file or directory

zailushang2008, 2022-04-15:
root@ontheroad:/# sensors-detect
# sensors-detect revision 5946 (2011-03-23 11:54:44 +0100)
# System: Intel Synoden
# Board: Intel N5095-AIO

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor'...                   No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8613
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor'...                   No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): YES
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): YES
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): YES
Found unknown SMBus adapter 8086:5ad4 at 0001:00:1f.1.
Found unknown SMBus adapter 8086:4da3 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.

Next adapter: i915 gmbus dpc (i2c-0)
Do you want to scan it? (YES/no/selectively): YES

Next adapter: i915 gmbus dpb (i2c-1)
Do you want to scan it? (YES/no/selectively): YES

Next adapter: i915 gmbus dpd (i2c-2)
Do you want to scan it? (YES/no/selectively): YES

Next adapter: DPDDC-A (i2c-3)
Do you want to scan it? (YES/no/selectively): YES

Now follows a summary of the probes I have just done.
Just press ENTER to continue:

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to generate /etc/sysconfig/lm_sensors? (YES/no): YES
Copy prog/init/lm_sensors.init to /etc/init.d/lm_sensors
for initialization at boot time.
You should now start the lm_sensors service to load the required
kernel modules.

Wer-Wolf, 2022-04-19:
It seems that you mainboard contains an unknown superio chip from ITE.
Since ITE usually does not provide datasheets of their superio chips, we cannot extend the ite87 driver to support newer chip ids.
Sorry :(

closed 2022-04-24 by zailushang2008