lm-sensors/lm-sensors #393: I hope Asrock H310CM ATX/AC will be supported

zailushang2008, 2022-04-24:
https://www.asrock.com/mb/Intel/H310CM-ITXac/index.asp

Wer-Wolf, 2022-04-26:
If in doubt, run `sensors-detect`. It will usually detect supported chips on your mainboard.
I think your mainboard is supported thru the nct6775 driver.

zailushang2008, 2022-05-12:
Only CPU temperature can be displayed

Wer-Wolf, 2022-05-12:
Did `sensors-detect` detect any sensor chips?
What kernel version fo you use?

Wer-Wolf, 2022-05-12:
Because most drivers need to be loaded manually.

zailushang2008, 2022-05-12:
**Linux ontheroad 4.4.180+ #42218 SMP Mon Oct 18 19:17:56 CST 2021 x86_64 GNU/Linux synology_apollolake_918+**

# sensors-detect revision 5946 (2011-03-23 11:54:44 +0100)
# System: To Be Filled By O.E.M. Synoden
# Board: ASRock H310CM-ITX/ac

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor'...                   No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd121
    (logical device B has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor'...                   No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:5ad4 at 0001:00:1f.1.
Found unknown SMBus adapter 8086:a2a3 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.

Next adapter: i915 gmbus dpc (i2c-0)
Do you want to scan it? (YES/no/selectively): 

Next adapter: i915 gmbus dpb (i2c-1)
Do you want to scan it? (YES/no/selectively): 

Next adapter: i915 gmbus dpd (i2c-2)
Do you want to scan it? (YES/no/selectively): 

Next adapter: DPDDC-D (i2c-3)
Do you want to scan it? (YES/no/selectively): 

Next adapter: DPDDC-E (i2c-4)
Do you want to scan it? (YES/no/selectively): 

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no): 
Copy prog/init/lm_sensors.init to /etc/init.d/lm_sensors
for initialization at boot time.
You should now start the lm_sensors service to load the required
kernel modules.

Wer-Wolf, 2022-05-12:
Your kernel (4.4) and sensors-detect (from 2011) are outdated. You should try newer versions since they support more sensor chips.

Wer-Wolf, 2022-05-12:
Also sensors-detect finds an unknown chip, which might be supported by newer versions.

zailushang2008, 2022-05-12:
Thank you for your help, let me have a try. I can't update the kernel because of Synology,but I can try to update the software version

zailushang2008, 2022-05-13:
I updated the software and made new progress, but I still couldn't detect the fan speed.

`root@ontheroad:~# sensors-detect
# sensors-detect version 3.6.0
# Board: ASRock H310CM-ITX/ac
# Kernel: 4.4.180+ x86_64
# Processor: Intel(R) Core(TM) i3-8100 CPU @ 3.60GHz (6/158/11)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT6793D Super IO Sensors'                   Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): YES
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): YES
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): YES
Using driver `i2c-i801' for device 0001:00:1f.1: Broxton (SOC)
Module i2c-i801 loaded successfully.
Using driver `i2c-i801' for device 0000:00:1f.4: Kaby Lake (PCH)

Next adapter: i915 gmbus dpc (i2c-0)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: i915 gmbus dpb (i2c-1)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: i915 gmbus dpd (i2c-2)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: DPDDC-D (i2c-3)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: DPDDC-E (i2c-4)
Do you want to scan it? (yes/NO/selectively): YES

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `nct6775':
  * ISA bus, address 0x290
    Chip `Nuvoton NCT6793D Super IO Sensors' (confidence: 9)

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no): YES
Copy prog/init/lm_sensors.service to /usr/lib/systemd/system
and run 'systemctl enable lm_sensors.service'
for initialization at boot time.
Unloading i2c-i801... failed

root@ontheroad:~# sensors
coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +46.0 C  (high = +80.0 C, crit = +100.0 C)
Core 0:         +46.0 C  (high = +80.0 C, crit = +100.0 C)
Core 1:         +45.0 C  (high = +80.0 C, crit = +100.0 C)
Core 2:         +46.0 C  (high = +80.0 C, crit = +100.0 C)
Core 3:         +44.0 C  (high = +80.0 C, crit = +100.0 C)`

Wer-Wolf, 2022-05-13:
The kernel is likely to old to support the sensor chip on your motherboard.

zailushang2008, 2022-05-13:
I guess I didn't load the NCT6775 driver module. No one compiles driver to support DSM7.0.1 and neither do I, thanks for your help

closed 2022-05-18 by zailushang2008