lm-sensors/lm-sensors #400: EC ITE IT5571 detected as 'unknown chip'

wharnisch, 2022-06-12:
ZOTAC ZBOX CI331 NANO uses EC ITE5571 as I/O chip.

sensors-detect 3.6.0 on Ubuntu 22.04 says when run on CI331:

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x5571

Any chances to get this fixed?

Wer-Wolf, 2022-06-13:
If you can find a datasheet for this superio chip, then support could be added to the it87 driver.
If however no public datasheet is available, then it will be next to impossible to fix that.

intractabilis, 2022-11-07:
My understanding that it is equivalent to IT8571. Here are the specs: https://dl.iranfso.com/DataSheet/IC/ITE/13161912-IT8570E.pdf.

bbkr, 2022-11-29:
Worth noting is that GPD Win Max 2 netbok uses very similar chip:

```
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x5570
```

I can test support if IT557x will be added to lm_sensors.

starsforeveryone, 2022-12-29:
I have an ASUS Expertbook B5 Flip with the same IT5571 sensor chip, an IT5571-VG128 to be exact. Since ASUS doesn't offer any fan control via UEFI, and doesn't support Linux, and has a very annoying default fan curve (to say the least!), and didn't respond to my polite E-Mails, I'm getting fed up about the whole situation.

Therefore, I took the liberty to write ITE (the manufacturer of the chip) an E-Mail, asking politely if they could make the datasheet available to the Linux Kernel driver team. I gave them the it87 maintainer's E-Mail address and put him in CC for the mail I sent. Maybe this will help us.

intractabilis, 2022-12-29:
Control of fans in ASUS is performed via ACPI WMI object. There is a driver in Linux kernel that supports it: `drivers/platform/x86/asus-wmi.c`. It is actively updated, find it in the git repository, the git history has authors emails, you can try talking to them.

ntzb, 2023-03-31:
> I have an ASUS Expertbook B5 Flip with the same IT5571 sensor chip, an IT5571-VG128 to be exact. Since ASUS doesn't offer any fan control via UEFI, and doesn't support Linux, and has a very annoying [...]

have you heard anything back?
I have a new laptop, TP3402, and the fan noise is driving me crazy!

Pyrestone, 2024-02-21:
I have the same chip on a Gigabyte / Aorus 15P:
```txt
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x5571
```

Did anyone ever get fan speed control running on this?

Can I somehow help write a driver for this by inspecting/snooping how the windows driver communicates with the chip?
If so, can someone point me in the right direction on how I were to go about this? (what bus to snoop, common programs to do that, ...)

Wer-Wolf, 2024-02-21:
Can you share the output of `acpidump`?

ntzb, 2024-02-21:
Hi, not at home right now, but it might be possible.
I succeeded with my ASUS motherboard, check this out:
https://github.com/zeule/asus-ec-sensors/issues/38
It depends on the acpi implementation of the motherboard maker.

Pyrestone, 2024-02-21:
@ntzb  Thanks for the suggestion, I'll try to look into that as well.

Wer-Wolf, 2024-02-21:
> > Can you share the output of `acpidump`?
> 
> @Wer-Wolf
> 
> just running `sudo acpidump` ? or do I need specific arguments for the address of the chip?

Just the command.

Wer-Wolf, 2024-02-21:
Gigabyte seems to have an ACPI WMI interface for exposing sensor data. Someone needs to reverse-engineer the Gigabyte application which uses this WMI interface.

Pyrestone, 2024-02-21:
Anything I can do to help with that?
I have a working dual-boot with the gigabyte fan control software installed. is there any way I can capture the relevant ACPI (or MMIO or whatever it is using) calls to that chip to figure out how it does that?

TBH I'm more interested in setting the fan pwm values than reading the fan RPMs or temperature, since my cpu and GPU give me plenty of APIs to read temperature already.
@Wer-Wolf 
Did you find any pointers on how to do that with acpi_call or similar?

slavaandrejev, 2024-02-21:
> Someone needs to reverse-engineer the Gigabyte application which uses this WMI interface

Below are the lists for Gigabyte Aero 16 SKU P86YE5.

## Get methods
```
[WMI, dynamic: ToInstance, provider("WmiProv"), Locale("MS\\0x409"), Description("Gigabyte WMI Get method"), guid("{ABBC0F6F-8EA1-11d1-00A0-C90629100000}")]
class GB_WMIACPI_Get
{
	[key, read] string InstanceName;
	[read] boolean Active;
	[WmiMethodId(70), Implemented, read, write, Description("Get CPU Fan Duty")] void GetCPUFanDuty([out, Description("Data")] uint8 Data);
	[WmiMethodId(71), Implemented, read, write, Description("Get GPU Fan Duty")] void GetGPUFanDuty([out, Description("Data")] uint8 Data);
	[WmiMethodId(72), Implemented, read, write, Description("Get Battery Cycle Count")] void GetBatteryCount([out, Description("Data")] uint16 Data);
	[WmiMethodId(73), Implemented, read, write, Description("Get Max Charge")] void GetMaxCharge([out, Description("Data")] uint8 Data);
	[WmiMethodId(80), Implemented, read, write, Description("Get PEG or SG")] void GetPEGorSG([out, Description("Data")] uint8 Data);
	[WmiMethodId(81), Implemented, read, write, Description("Get Nv Power Config")] void GetNvPowerConfig([out, Description("Data")] uint8 Data);
	[WmiMethodId(86), Implemented, read, write, Description("Get Thermal Sensor Data")] void GetThermalData([out, Description("Thermal1")] uint8 Thermal1, [out, Description("Thermal2")] uint8 Thermal2, [out, Description("Thermal3")] uint8 Thermal3);
	[WmiMethodId(87), Implemented, read, write, Description("Get Nv Thermal Target")] void GetNvThermalTarget([out, Description("Data")] uint8 Data);
	[WmiMethodId(88), Implemented, read, write, Description("Check Heavy Loading")] void CheckHeavyLoading([out, Description("Status")] uint8 Status);
	[WmiMethodId(89), Implemented, read, write, Description("Get Light Bar Status")] void GetLightBar([in, Description("Index")] uint8 Index, [out, Description("Status")] uint8 Status, [out, Description("Level")] uint8 Level, [out, Description("Red")] uint8 Red, [out, Description("Green")] uint8 Green, [out, Description("Blue")] uint8 Blue);
	[WmiMethodId(96), Implemented, read, write, Description("Get Deep Fan")] void GetDeepFan([out, Description("Temperature0")] uint8 Temperature0, [out, Description("Temperature1")] uint8 Temperature1, [out, Description("Temperature2")] uint8 Temperature2, [out, Description("Temperature3")] uint8 Temperature3, [out, Description("Temperature4")] uint8 Temperature4, [out, Description("Speed0")] uint8 Speed0, [out, Description("Speed1")] uint8 Speed1, [out, Description("Speed2")] uint8 Speed2, [out, Description("Speed3")] uint8 Speed3, [out, Description("Speed4")] uint8 Speed4);
	[WmiMethodId(97), Implemented, read, write, Description("Get Battery Health Status")] void GetBatteryHealth([out, Description("Data")] uint8 Data);
	[WmiMethodId(98), Implemented, read, write, Description("Get Fan Health Status")] void GetFanHealth([out, Description("Data")] uint8 Data);
	[WmiMethodId(99), Implemented, read, write, Description("Get Power On Time")] void GetPowerOnTime([out, Description("Year")] uint8 Year, [out, Description("Month")] uint8 Month, [out, Description("Day")] uint8 Day, [out, Description("Hour")] uint8 Hour, [out, Description("Minute")] uint8 Minute);
	[WmiMethodId(100), Implemented, read, write, Description("Get Charge Policy")] void GetChargePolicy([out, Description("Data")] uint16 Data);
	[WmiMethodId(101), Implemented, read, write, Description("Get Charge Stop")] void GetChargeStop([out, Description("Data")] uint16 Data);
	[WmiMethodId(103), Implemented, read, write, Description("Get 10 Steps Fan Status")] void GetStepFanStatus([out, Description("Data")] uint16 Data);
	[WmiMethodId(104), Implemented, read, write, Description("Get Fan Index Value")] void GetFanIndexValue([in, Description("Index")] uint8 Index, [out, Description("Temperature")] uint8 Temperture, [out, Description("Speed")] uint8 Value);
	[WmiMethodId(105), Implemented, read, write, Description("Get VR Mode Status")] void GetVRStatus([out, Description("Data")] uint16 Data);
	[WmiMethodId(106), Implemented, read, write, Description("Get Fixed Fan Status")] void GetFixedFanStatus([out, Description("Data")] uint16 Data);
	[WmiMethodId(107), Implemented, read, write, Description("Get Fixed Fan Speed")] void GetFixedFanSpeed([out, Description("Data")] uint16 Data);
	[WmiMethodId(108), Implemented, read, write, Description("Are Devices Exist")] void IsDeviceExist([out, Description("Data")] uint16 Data);
	[WmiMethodId(109), Implemented, read, write, Description("Get Battery Cycles1")] void getBattCyc1([out, Description("Data")] uint16 Data);
	[WmiMethodId(110), Implemented, read, write, Description("Get Battery Cycles")] void getBattCyc([out, Description("Data")] uint16 Data);
	[WmiMethodId(111), Implemented, read, write, Description("Get Fan PWM status")] void GetFanPWMStatus([out, Description("Data")] uint8 Data);
	[WmiMethodId(112), Implemented, read, write, Description("Get Fan Adjust status")] void GetFanAdjustStatus([out, Description("Data")] uint8 Data);
	[WmiMethodId(113), Implemented, read, write, Description("Getting Auto Fan Status")] void GetAutoFanStatus([out, Description("Data")] uint8 Data);
	[WmiMethodId(122), Implemented, read, write, Description("Getting the USB charge under Sleep")] void GetSleepUSBCharge([out, Description("Data")] uint8 Data);
[... cut ...]
	[WmiMethodId(244), Implemented, read, write, Description("Getting the Enable/Disable status of Onboard LAN")] void GetOnboardLANStatus([out, Description("Data")] uint8 Data);
[... 138 lines cut ...]
```

### Meeting Mode
```
SetDynamicBoostStatus(0)
SetNvPowerConfig(0)
SetWhisper(it depends of on other settings, false, 4UL)
SetCurrentFanStep(0)
SetFixedFanStatus(0)
SetStepFanStatus(0)
SetAutoFanStatus(0)
SetNvThermalTarget(1)
```

### Silence Mode
```
SetDynamicBoostStatus(0)
SetNvPowerConfig(0)
SetWhisper(it depends of on other settings, false, 4UL)
SetCurrentFanStep(0)
SetFixedFanStatus(0)
SetStepFanStatus(0)
SetAutoFanStatus(0)
SetNvThermalTarget(1)
```

Pyrestone, 2024-02-21:
@slavaandrejev That looks helpful. I think the aorus 15p shares many parts with the aero laptops (I think my battery reports as gigabyte aero for instance) so this could very well be useful.

However I still don't know what to do with that information. 
I'm still piecing together an understanding of what  WMI is and which parts of WMI and ACPI are in the motherboard/bios and what parts are in proprietary windows drivers or the linux kernel.
Sorry for taking so much of your time on this, but I just don't know how to apply the information provided.

slavaandrejev, 2024-02-22:
> I think the aorus 15p shares many parts with the aero laptops (I think my battery reports as gigabyte aero for instance) so this could very well be useful.

Can you install [WMI Explorer](https://github.com/vinaypamnani/wmie2/releases) and use it to get the list of methods for your laptop? Please publish it here; I am curious to see the differences. Here are the instructions:
 * Start WMI Explorer
 * Click "Connect"
 * Double click "ROOT\WMI"
 * Type WMIACPI in the "Quick Filter" field
 * Right click GB_WMIACPI_Get, choose "Show MOF"
 * Right click GB_WMIACPI_Set, choose "Show MOF"

> However I still don't know what to do with that information.

I can only point to the direction. You need to use ACPI to call methods from the ACPI's WMI object. You can check how to do it in the implementation of [wmidev_evaluate_method](https://elixir.bootlin.com/linux/latest/source/drivers/platform/x86/wmi.c#L369). Maybe you can do it using `acpi_call`. Otherwise, your only option is to write a small Linux driver for this. For the latter, you can use [asus_wmi_sensors.c](https://elixir.bootlin.com/linux/latest/source/drivers/hwmon/asus_wmi_sensors.c) as an example.

Wer-Wolf, 2024-02-22:
> Below are the lists for Gigabyte Aero 16 SKU P86YE5.
>
> ## Get methods
> ```
> [WMI, dynamic: ToInstance, provider("WmiProv"), Locale("MS\\0x409"), Description("Gigabyte WMI Get method"), [...]

Nice, that will help a lot.

ntzb, 2024-02-22:
> Nice, that will help a lot.

helps with what? support of ITE5571 in `lm-sensors`?

> I can only point to the direction. You need to use ACPI to call methods from the ACPI's WMI object. You can check how to do it in the implementation of [...]

I've done this, and describe it in the already mentioned github issue: https://github.com/zeule/asus-ec-sensors/issues/38#issuecomment-1510477372

Wer-Wolf, 2024-02-22:
The WMI MOF data will help in implementing access to the sensors on gigabyte notebooks, but sadly every `ITE5571` might need a special driver for the OEM interface.

Pyrestone, 2024-02-22:
@slavaandrejev thanks! I'll try that this evening.

@Wer-Wolf 
> The WMI MOF data will help in implementing access to the sensors on gigabyte notebooks, but sadly every `ITE5571` might need a special driver for the OEM interface.

So we need to go through the motherboard ACPI api to talk to the 5571 chip and can't bypass it to talk to it's address directly like the sensors-detect command does? and that's why we need a device-specific driver?

Wer-Wolf, 2024-02-22:
The thing is that many superio chips contain an embedded controller, and only this controller has access to the sensors. The interface to this embedded controller is manufacturer-specific.

Also trying to talk with the superio chip directly will have a great chance of conflicting with the ACPI firmware. Basically we would be fighting with ACPI over who has control over the superio chip.

slavaandrejev, 2024-02-22:
> @slavaandrejev thanks! I'll try that this evening.

Can you publish MOF for your laptop?

slavaandrejev, 2024-02-22:
@Pyrestone I created a skeleton for a WMI driver for Gigabyte: https://github.com/slavaandrejev/gigabyte-wmi

Pyrestone, 2024-02-22:
@slavaandrejev Here are the extracted MOF definitions. When cross-checking with the files from your repo, they are exactly identical except for the "SetPDWarmReset" method which is missing in mine.

<details>
<summary>WMI MOF ROOT\WMI:GB_WMIACPI_Get </summary>

```

[WMI, dynamic: ToInstance, provider("WmiProv"), Locale("MS\\0x409"), Description("Gigabyte WMI Get method"), guid("{ABBC0F6F-8EA1-11d1-00A0-C90629100000}")]
class GB_WMIACPI_Get
{
	[key, read] string InstanceName;
	[read] boolean Active;
	[WmiMethodId(70), Implemented, read, write, Description("Get CPU Fan Duty")] void GetCPUFanDuty([out, Description("Data")] uint8 Data);
	[WmiMethodId(71), Implemented, read, write, Description("Get GPU Fan Duty")] void GetGPUFanDuty([out, Description("Data")] uint8 Data);
	[WmiMethodId(72), Implemented, read, write, Description("Get Battery Cycle Count")] void GetBatteryCount([out, Description("Data")] uint16 Data);
	[WmiMethodId(73), Implemented, read, write, Description("Get Max Charge")] void GetMaxCharge([out, Description("Data")] uint8 Data);
	[WmiMethodId(80), Implemented, read, write, Description("Get PEG or SG")] void GetPEGorSG([out, Description("Data")] uint8 Data);
	[WmiMethodId(81), Implemented, read, write, Description("Get Nv Power Config")] void GetNvPowerConfig([out, Description("Data")] uint8 Data);
	[WmiMethodId(86), Implemented, read, write, Description("Get Thermal Sensor Data")] void GetThermalData([out, Description("Thermal1")] uint8 Thermal1, [out, Description("Thermal2")] uint8 Thermal2, [out, Description("Thermal3")] uint8 Thermal3);
	[WmiMethodId(87), Implemented, read, write, Description("Get Nv Thermal Target")] void GetNvThermalTarget([out, Description("Data")] uint8 Data);
	[WmiMethodId(88), Implemented, read, write, Description("Check Heavy Loading")] void CheckHeavyLoading([out, Description("Status")] uint8 Status);
	[WmiMethodId(89), Implemented, read, write, Description("Get Light Bar Status")] void GetLightBar([in, Description("Index")] uint8 Index, [out, Description("Status")] uint8 Status, [out, Description("Level")] uint8 Level, [out, Description("Red")] uint8 Red, [out, Description("Green")] uint8 Green, [out, Description("Blue")] uint8 Blue);
	[WmiMethodId(96), Implemented, read, write, Description("Get Deep Fan")] void GetDeepFan([out, Description("Temperature0")] uint8 Temperature0, [out, Description("Temperature1")] uint8 Temperature1, [out, Description("Temperature2")] uint8 Temperature2, [out, Description("Temperature3")] uint8 Temperature3, [out, Description("Temperature4")] uint8 Temperature4, [out, Description("Speed0")] uint8 Speed0, [out, Description("Speed1")] uint8 Speed1, [out, Description("Speed2")] uint8 Speed2, [out, Description("Speed3")] uint8 Speed3, [out, Description("Speed4")] uint8 Speed4);
	[WmiMethodId(97), Implemented, read, write, Description("Get Battery Health Status")] void GetBatteryHealth([out, Description("Data")] uint8 Data);
	[WmiMethodId(98), Implemented, read, write, Description("Get Fan Health Status")] void GetFanHealth([out, Description("Data")] uint8 Data);
	[WmiMethodId(99), Implemented, read, write, Description("Get Power On Time")] void GetPowerOnTime([out, Description("Year")] uint8 Year, [out, Description("Month")] uint8 Month, [out, Description("Day")] uint8 Day, [out, Description("Hour")] uint8 Hour, [out, Description("Minute")] uint8 Minute);
	[WmiMethodId(100), Implemented, read, write, Description("Get Charge Policy")] void GetChargePolicy([out, Description("Data")] uint16 Data);
	[WmiMethodId(101), Implemented, read, write, Description("Get Charge Stop")] void GetChargeStop([out, Description("Data")] uint16 Data);
	[WmiMethodId(103), Implemented, read, write, Description("Get 10 Steps Fan Status")] void GetStepFanStatus([out, Description("Data")] uint16 Data);
	[WmiMethodId(104), Implemented, read, write, Description("Get Fan Index Value")] void GetFanIndexValue([in, Description("Index")] uint8 Index, [out, Description("Temperature")] uint8 Temperture, [out, Description("Speed")] uint8 Value);
	[WmiMethodId(105), Implemented, read, write, Description("Get VR Mode Status")] void GetVRStatus([out, Description("Data")] uint16 Data);
	[WmiMethodId(106), Implemented, read, write, Description("Get Fixed Fan Status")] void GetFixedFanStatus([out, Description("Data")] uint16 Data);
	[WmiMethodId(107), Implemented, read, write, Description("Get Fixed Fan Speed")] void GetFixedFanSpeed([out, Description("Data")] uint16 Data);
	[WmiMethodId(108), Implemented, read, write, Description("Are Devices Exist")] void IsDeviceExist([out, Description("Data")] uint16 Data);
	[WmiMethodId(109), Implemented, read, write, Description("Get Battery Cycles1")] void getBattCyc1([out, Description("Data")] uint16 Data);
	[WmiMethodId(110), Implemented, read, write, Description("Get Battery Cycles")] void getBattCyc([out, Description("Data")] uint16 Data);
	[WmiMethodId(111), Implemented, read, write, Description("Get Fan PWM status")] void GetFanPWMStatus([out, Description("Data")] uint8 Data);
	[WmiMethodId(112), Implemented, read, write, Description("Get Fan Adjust status")] void GetFanAdjustStatus([out, Description("Data")] uint8 Data);
	[WmiMethodId(113), Implemented, read, write, Description("Getting Auto Fan Status")] void GetAutoFanStatus([out, Description("Data")] uint8 Data);
[... cut ...]
	[WmiDataId(1), read, write, Description("Read A ULONG value")] uint32 Data;
};

```

</details>

<details>

<summary>ROOT\WMI:GB_WMIACPI_Event</summary>

```

[WMI, dynamic: ToInstance, provider("WmiProv"), Description("Class containing event generated ULong data"), Locale("MS\\0x409"), guid("{ABBC0F72-8EA1-11d1-00A0-C90629100000}")]
class GB_WMIACPI_Event : WMIEvent
{
	[key, read] string InstanceName;
	[read] boolean Active;
	[WmiDataId(1), read, write, Description("WMI Event Data"), MAX(4)] uint8 Data[];
};

```

</details>

slavaandrejev, 2024-02-22:
> When cross-checking with the files from your repo, they are exactly identical except for the "SetPDWarmReset" method which is missing in mine.

Also, method 252 is `GetLightSensorValue` on Aero 16, but it is `CheckBIOSMode` on Aorus 15P.

slavaandrejev, 2024-02-22:
@Pyrestone Show the output for
```
sudo dmidecode -s system-product-name
sudo dmidecode -s system-sku-number
```

Pyrestone, 2024-02-23:
system-sku-number is not exposed. Here's the keys I do have:
```txt

bios-vendor: American Megatrends International, LLC.
bios-version: FB07
bios-release-date: 10/07/2021
system-manufacturer: GIGABYTE
system-product-name: AORUS 15P YD
system-version: 993AC
system-serial-number: <removed>
system-uuid: <removed>
system-family: AORUS
baseboard-manufacturer: GIGABYTE
baseboard-product-name: AORUS 15P YD
baseboard-version: Default string
baseboard-serial-number: Default string
baseboard-asset-tag: Default string
chassis-manufacturer: SYS_CHASSIS_MANUFACTURER_1
chassis-type: Notebook
chassis-version: y.y
chassis-serial-number: Default string
chassis-asset-tag: Default string
processor-family: Core i7
processor-manufacturer: Intel(R) Corporation
processor-version: 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz
processor-frequency: 2277 MHz
```

@slavaandrejev what BIOS version are you on? maybe that makes a difference in what WMI methods the board exposes.

Pyrestone, 2024-02-23:
Ah, I did find a SKU number in a different dmidecode command:
```
sudo dmidecode -t system | grep SKU
	SKU Number: X5LYD
```

Pyrestone, 2024-02-23:
@slavaandrejev 
I also found these resources which I can maybe go through on the weekend to get an idea on how to help with your driver repo:
Writing a WMI driver - an introduction:
https://lwn.net/Articles/391230/
Kernel reference WMI:
https://wiki.ubuntu.com/Kernel/Reference/WMI
Blog post about maybe turning MOF to .h files automatically ?:
http://blog.nietrzeba.pl/2011/12/mof-decompilation.html?m=1
Example driver for ASUS WMI methods (maybe as reference to your gigabyte version):
Kernel reference WMI:
https://github.com/torvalds/linux/blob/v5.17/drivers/platform/x86/asus-wmi.c

Wer-Wolf, 2024-02-23:
> @slavaandrejev I also found these resources which I can maybe go through on the weekend to get an idea on how to help with your driver repo: Writing a WMI driver - an introduction: [...]

Please use the new WMI api, otherwise the driver wont be accepted upstream.

Also you should take a look at the bmfdec tools and the wmi-bmof driver.

Pyrestone, 2024-02-23:
> Please use the new WMI api, otherwise the driver wont be accepted upstream.

Do you mean this one?
https://www.kernel.org/doc/html/next/driver-api/wmi.html

Pyrestone, 2024-02-23:
tbh, this documentation is not really informative for someone who hasn't written kernel drivers before:
https://docs.kernel.org/wmi/devices/wmi-bmof.html

Wer-Wolf, 2024-02-23:
> > Please use the new WMI api, otherwise the driver wont be accepted upstream.
> 
> Do you mean this one? https://www.kernel.org/doc/html/next/driver-api/wmi.html

Yes, this one.

A good starting point is to look at existing drivers, like the `dell-wmi-ddv` driver.

Pyrestone, 2024-02-23:
There's also this:
https://github.com/torvalds/linux/blob/v6.7/drivers/platform/x86/gigabyte-wmi.c

which seems to be a gigabyte-specific driver for some desktop motherboards, but it only does temperature sensors I think.

Wer-Wolf, 2024-02-23:
I agree, but please avoid having global variables (use the state container pattern instead).

JLP, 2024-05-20:
I have the same Trying family National Semiconductor/ITE... Yes Found unknown chip with ID 0x5570 output for KDE Slimbook V laptop (with AMD Ryzen 7 7840HS CPU).

Bush-cat, 2024-10-10:
Same Issue on my 7840HS Beelink GTR7 `Found unknown chip with ID 0x5571`

geheim27, 2024-12-13:
I have the same chip in an AiO PC (HP Omnistudio X 32). Just bought the device, really need to slow done the fan to be able to work on the device.

I tried to analyze the EC registers with RW-Everything, tried to understand the ACPI tables, also tried with notebook fan control ([nbfc](https://github.com/nbfc-linux/nbfc-linux))

What I have found out: the tool "HP Command Center" has a "silent option" This changes the EC register 0xE5 to the value 3 and slows down the fan a bit.

Temperatur1 (TZ00) goes to 0x16 (DTMP)
Temperatur2 (TZ01) goes to 0x0C (TTMP)

The following registers might change in line with the fan noice: 0xFE and 0x8B/0x8C

Register 0x96 holds usually the value 1. If i increase the value, the fan spins faster for some seconds. But value is not constant. It counts down to 1 again.
If I write a hex value greater or equal to 0x0B to 0x96, the fan stop. And sometimes the PC freezes. Somthing might be crashing here.

Maybe someone can help with the acpidump (exported with RW-Everything, can also do this from linux if needed). I have problems finding out more usefull stuff with it.

[AcpiTbls.txt](https://github.com/user-attachments/files/18124253/AcpiTbls.txt)

tkkost, 2025-03-12:
Hello. Now 2025. Does it support the IT5571?

> Trying family `National Semiconductor/ITE'...               Yes
> Found unknown chip with ID 0x5571

I bought a new Minisforum X7 Ti and can't add fans to lm-sensors.

11matt556, 2025-04-06:
> Can you install [WMI Explorer](https://github.com/vinaypamnani/wmie2/releases) and use it to get the list of methods for your laptop? Please publish it here; I am curious to see the differences. [...]

@slavaandrejev Do you have pointers on how one might find the ACPI calls if WMIACPI does not exist (Or if there is any fan control at all)? I have a different system (Minisforum V3) and I've exported the bmof and acpi dump (.dsl files) from Linux, but the only reference I've found to fans is a "ECFanTableIndex" in "ssdt7.dsl". From what I can gather, it seems the function call is "APX7" but I'm not sure where to go from here since "APX7" doesn't being up anything in WMI Explorer,.

I'm also not sure how to tell if fan control is even possible, since the vendor application doesn't have the fan controls separate from the power limit controls. Is there any other way to confirm if there is any fan control?

I've attached the ssdt7.dsl file with the "APX7" method in it, in case it helps.

Background: I have some experience working on Linux kernel drivers, but basically none with ACPI and WMI. If fan control is possible I feel like I could get something working with the kernel WMI interface, but I'm currently lost on how to find the WMI functions.

[ssdt7.dsl.txt](https://github.com/user-attachments/files/19623734/ssdt7.dsl.txt)

slavaandrejev, 2025-04-06:
> I'm currently lost on how to find the WMI functions

Do you have anything in `/sys/bus/wmi/devices/`?

11matt556, 2025-04-06:
> > I'm currently lost on how to find the WMI functions
> 
> Do you have anything in `/sys/bus/wmi/devices/`?

Yes. I have two devices there. 

```
ls /sys/bus/wmi/devices/ABBC0F5C-8EA1-11D1-A000-C90629100000
driver_override  expensive  guid  instance_count  modalias  notify_id  power  subsystem  uevent

ls /sys/bus/wmi/devices/05901221-D566-11D1-B2F0-00A0C9062910
bmof    driver_override  guid            modalias   power    subsystem
driver  expensive        instance_count  object_id  setable  uevent
```

 `05901221-D566-11D1-B2F0-00A0C9062910` is what I used to get the bmof information, since it matched [Wer-Wolfs comment](https://github.com/lm-sensors/lm-sensors/issues/400#issuecomment-1961169766), and the other one doesn't have a bmof file.

slavaandrejev, 2025-04-06:
Hmm... interesting. The first one it looks like is already [supported](https://github.com/torvalds/linux/blob/master/drivers/platform/x86/huawei-wmi.c) by the Linux kernel.

> I'm also not sure how to tell if fan control is even possible, since the vendor application doesn't have the fan controls separate from the power limit controls. Is there any other way to confirm if there is any fan control?

In case of my Gigabyte laptop, it turned out that their Control Center is written using .NET. I used dnSpy to check the calls that changed the performance settings. Maybe your vendor app is also written using .NET? Can you try?

11matt556, 2025-04-07:
> Hmm... interesting. The first one it looks like is already [supported](https://github.com/torvalds/linux/blob/master/drivers/platform/x86/huawei-wmi.c) by the Linux kernel.

I'm not sure, but my initial guess would be that Minisforum has different returns different DMI or device/vendor IDs such that the huaweu-wmi driver doesn't match, since Windows doesn't show anything as Huawei branded in device manager, hwinfo, etc. (I haven't looked again on Linux, but I don't recall seeing any Huawei devices in lspci and such when I first started investigating this). I suspect Minisforum probably uses it differently than Huawei though, because the features seemingly facilitated by that driver (Function keys, brightness, battery charge control, etc) all seem to work fine on my laptop. 

> In case of my Gigabyte laptop, it turned out that their Control Center is written using .NET. I used dnSpy to check the calls that changed the performance settings. Maybe your vendor app is also written using .NET? Can you try?

It unfortunately doesn't seem like it is .NET. dnSpy couldn't decompile it and I see malloc.h mentioned in some of the PE headers, so it's presumably either C or C++.

My next thought would be to try to trace all WMI events on Windows to see if I can identify any WMI calls potentially related to the application. It seems like [Win32 Tracing WMI Activity](https://learn.microsoft.com/en-us/windows/win32/wmisdk/tracing-wmi-activity) is probably the best option for this. 

Do you have any other ideas/suggestions?

slavaandrejev, 2025-04-07:
> Minisforum has different returns different DMI or device/vendor IDs

My thought was that you could compose your own driver by modifying `huawei_quirks` in `huawei-wmi.c` to match your system.

> It unfortunately doesn't seem like it is .NET

Shoot, it's a bummer, really sorry.

Just in case, did you check for `Name (_WDG, Buffer (...` in `dsdt.dsl`?

> My next thought would be to try to trace all WMI events on Windows

Yes, that is what I would do as well. Or use IDA (they have a [free version](https://hex-rays.com/ida-free) nowadays). If that Microsoft tracing you mentioned doesn't work, I would have intercepted WMI calls myself. WMI API is COM-based, and should be easily interceptable with a simple C++ program.

11matt556, 2025-04-08:
@slavaandrejev It doesn't appear to show in the event log tracing. I do see various WMI events, but none of them seem correlated with any activity I do of the vendor control app. Also, the process ID of the event nor the "ClientProcessId" reported in the text of the event match any of the process IDs associated with the vendor app (As reported by Process Explorer), and all of the WMI classes being accessed look like just standard Windows things, so I am guessing I'm probably just looking at normal background stuff Windows is doing.

**If there are no relevant WMI events in the trace, is there any reason to believe the WMI API would say anything different? Is it possible Minisforum didn't use WMI and instead have some other interface to the IT5571?** 

Also, do you know if the earlier comment from @intractabilis about it being equivalent to an IT8570E is accurate? (The link they posted is also dead now, but thankfully someone archived it in the wayback machine ([here](https://web.archive.org/web/20191224234652/https://dl.iranfso.com/DataSheet/IC/ITE/13161912-IT8570E.pdf)).

I've had a hard time figuring out any details about the "IT5571" hardware. It is kind of crazy that I can't find a datasheet or listing on component sites like Digikey, but if it is the same as the IT8570E then it seems to basically be an 8051 MCU with built in ACPI features a bunch of GPIO. 

If that is the case, is there any reason to assume Minisforum have implemented a WMI interface? (For context, I have some experience with the 8051 since my work is at a design/manufacturing company and some of our products used 8051-based MCUs. I don't have experience with it at any sort of granular/detailed level though, since that isn't my job)
Like does Windows require ACPI devices to use WMI or something?

I really hope I am wrong about this and they did implement a WMI interface, because a completely propriatary interface is probably beyond my time/ability to reverse engineer unless they left something blatently obvious in the decompiled program (I've not checked out IDA yet).

11matt556, 2025-04-08:
> Check if your vendor app process has `wbemcntl.dll`, `wbemprox.dll`, `wbemsvc.dll`, `wmiutils.dll`, or `wbemcomn.dll` loaded. 

I do not see any of those loaded in either Process Explorer or loading with [InspectExe](https://www.silurian.com/inspect/index.htm). 

> This will tell you if your vendor app uses WMI. If there is no WMI, I would search for `DeviceIoControl` calls using IDA. Then I would intercept calls to `DeviceIoControl` to see if I can extract anything useful.

Been busy with other things today so still not had a chance to figure out IDA, but InspectExe shows that the exe imports DeviceIoControl from kernel32, so that seems promising. I'll update with what (if anything) I am able to find out by intercepting DeviceIoControl.

11matt556, 2025-04-18:
@slavaandrejev I am still figuring out IDA, but it doesn't seem like DeviceIoControl is being used (Or at least it is not in the functions IDA recognized, nor does DeviceIoControl show up in any text search)

However, when I attach to the process of the vendor app and view the fan RPM I can see what I assume to be the calls responsible for getting the RPM.

This is the sort of message I see: in thr Output window in IDA:
```
Debugged application message: WaitECObf ReadIoPortByte EC_C_PORT SUC 
Debugged application message: ReadEcSpace EC_D_PORT data : 25, ret : 22 
```
The message prints each time the RPM value updates and it appears as though higher values are associated with higher RPM, so it all seems to check out as being the RPM value in some sort of format.

The only google result I got for ReadIoPortByte was [WinRing0](https://github.com/QCute/WinRing0/blob/master/drv/DriverApi.h), which is interesting since that has been in the news recently.  This might mean the vendor is using WinRing0 rather than normal DeviceIoControl, but I don't see other indications of WinRing0, so it might just a naming coincidence. It does seem a bit weird for it to be using IO Ports though... I thought pretty much everything was memory mapped these days?

I am not sure where to go next though because I cannot find any of these (ReadIoPortByte, WaitECObf, ReadEcSpace, etc) when I decompile the exe in IDA.

Am I missing something? Or is it just not guaranteed that IDA will be able to learn all of the function names from the exe?  

If it is the latter, where are the messages I get when attached to the process coming from? Is the 'output' sectio actually the debug output of the attached application, not neccessarily the output of IDA? In other words, are these actually debug messages from the attached process, presumably put in by the vendor to help with their internal development/debugging?

If it is debug put in by the application vendor, how might I go about identifying exactly where this call is located in the decompilation?

slavaandrejev, 2025-04-19:
OBF indicates they are using [ACPI Embedded Controller Interface](https://uefi.org/htmlspecs/ACPI_Spec_6_4_html/12_ACPI_Embedded_Controller_Interface_Specification/ACPI_Embedded_Controller_Interface_Specification.html). "IO port" doesn't mean it's necessarily an x86 CPU I/O port (port address space accessible via dedicated CPU pins).

11matt556, 2025-04-20:
@slavaandrejev This might be a dumb question, but what does OBF stand for in this context?

slavaandrejev, 2025-04-20:
No worries. It's a bit in the [embedded controller status register](https://uefi.org/htmlspecs/ACPI_Spec_6_4_html/12_ACPI_Embedded_Controller_Interface_Specification/embedded-controller-register-descriptions.html). OBF stands for "output buffer is full (data ready for host)". `WaitECObf` means they are waiting for the data to be ready.

11matt556, 2025-04-20:
@slavaandrejev Ah, ok. I wasn't sure where OBF was coming from. I didn't realize EC was Embedded Controller, so I had interpreted WaitECObf as "Wait ECO Buffer" rather than "Wait EC OBF" but the latter makes more sense. 

I've also found some symbol files that IDA said were missing and am rerunning the IDA analysis to see what it can tell me now.

slavaandrejev, 2025-04-21:
Great progress. I hope you will get to the root of it.

geheim27, 2025-04-26:
As mentioned before, I have the same chip in an HP all in one computer. Finally, I'm able to use NBFC to reduce the fan speed. It's far from perfect but at least some control is possible. the relevant register has to written every view hundred milliseconds to keep control over fan speed.

I also have the datasheet but could not find out more details within the document. On request, i could provide it via mail.

11matt556, 2025-04-26:
> I also have the datasheet but could not find out more details within the document. On request, i could provide it via mail.

You could upload it to the internet archive. I've tried to find a datasheet, but the closest I've got is https://web.archive.org/web/20191224234652/https://dl.iranfso.com/DataSheet/IC/ITE/13161912-IT8570E.pdf , which someone said is essentially the same chip as the IT5571, but I don't know how true that is.

geheim27, 2025-05-15:
Yes, I used just the registers from the EC controller. I used RW-Everything on windows and ec_probe on linux to watch and to write registers and a lot of try and error. ec_probe is part of nbfc (https://github.com/nbfc-linux/nbfc-linux).

I used a simple script which calls ec_probe every 100ms to overwrite a specific register and listened for a reaction of the fan. 

This is my current configuration (still pending for optimization):
`{
 "NotebookModel": "HP Omnistudio",
 "Author": "myself",
 "EcPollInterval": 500,
 "ReadWriteWords": false,
 "CriticalTemperature": 95,
 "CriticalTemperatureOffset": 5,
 "FanConfigurations": [
	{
	 "ReadRegister": 254,
	 "MinSpeedValueRead": 42,
	 "MaxSpeedValueRead": 239,
	 "WriteRegister": 148,
	 "MinSpeedValue": 8,
	 "MaxSpeedValue": 40,
	 "IndependentReadMinMaxValues": true,
	 "ResetRequired": false,
	 "FanSpeedResetValue": 50,
	 "FanDisplayName": "CPU fan",
	 "TemperatureThresholds": [
	  {
	   "UpThreshold": 35,
	   "DownThreshold": 34,
	   "FanSpeed": 0.0
	  },
	  {
	   "UpThreshold": 42,
	   "DownThreshold": 40,
	   "FanSpeed": 8.0
	  },
	  {
	   "UpThreshold": 44,
	   "DownThreshold": 41,
	   "FanSpeed": 10.0
	  },
	  {
	   "UpThreshold": 48,
	   "DownThreshold": 43,
	   "FanSpeed": 20.0
	  },
	  {
	   "UpThreshold": 52,
	   "DownThreshold": 46,
	   "FanSpeed": 25.0
	  },
	  {
	   "UpThreshold": 54,
	   "DownThreshold": 50,
	   "FanSpeed": 38.0
	  },
	  {
	   "UpThreshold": 58,
	   "DownThreshold": 53,
	   "FanSpeed": 41.0
	  },
	  {
	   "UpThreshold": 62,
	   "DownThreshold": 56,
	   "FanSpeed": 50.0
	  },
	  {
	   "UpThreshold": 64,
	   "DownThreshold": 60,
	   "FanSpeed": 60.0
	  },
	  {
	   "UpThreshold": 68,
	   "DownThreshold": 63,
	   "FanSpeed": 70.0
	  },
	  {
	   "UpThreshold": 72,
	   "DownThreshold": 66,
	   "FanSpeed": 80.0
	  },
	  {
	   "UpThreshold": 74,
	   "DownThreshold": 70,
	   "FanSpeed": 90.0
	  },
	  {
	   "UpThreshold": 78,
	   "DownThreshold": 73,
	   "FanSpeed": 100.0
	  },
	  {
	   "UpThreshold": 82,
	   "DownThreshold": 76,
	   "FanSpeed": 100.0
	  },
	  {
	   "UpThreshold": 84,
	   "DownThreshold": 80,
	   "FanSpeed": 100.0
	  },
	  {
	   "UpThreshold": 88,
	   "DownThreshold": 83,
	   "FanSpeed": 100.0
	  },
	  {
	   "UpThreshold": 90,
	   "DownThreshold": 86,
	   "FanSpeed": 100.0
	  }
	 ],
	 "FanSpeedPercentageOverrides": []
	}
 ],
 "RegisterWriteConfigurations": [
	{
	 "WriteMode": "Set",
	 "WriteOccasion": "OnInitialization",
	 "Register": 229,
	 "Value": 3,
	 "ResetRequired": true,
	 "ResetValue": 1,
	 "ResetWriteMode": "Set",
	 "Description": "CPU fan silent mode"
	}
 ]
}
`

chantsing, 2025-10-17:
I upload the datasheet of IT5571, Someone help to resolve it?
[IT5571_C_V0.3.3_20180709.pdf](https://github.com/user-attachments/files/22960854/IT5571_C_V0.3.3_20180709.pdf)@11matt556