lm-sensors/lm-sensors #405: Containerizing lm-sensors

bobbyrc, 2022-07-25:
Hi there! 

I'm attempting to create a containerized app that depends on lm-sensors. Right now the container can't find any sensors so I'm assuming that it either needs a runtime capability or maybe some volumes mapped so it has access to the right files on the host system. 

I was curious if there were any examples of lm-sensors being containerized that I could look at or if anyone might be able to advise on what config would be needed in docker to allow lm-sensors running in a container to access the host machine and read sensor data.

I really appreciate any help you're able to provide, and I hope you're having a great day!

Wer-Wolf, 2022-07-29:
lm-sensors needs access to the sysfs file system of the host. It could be that your container application restricts such access.

brauliobo, 2025-04-28:
what kind of mount of sysfs is required?