lm-sensors/lm-sensors #408: lm_sensors-3.6.0 Inaccurate PSU Temps

0pLuS0, 2022-08-18:
Hello,

As soon as I start up my computer, and run sensors within a 1-2 mins of the computer running, sensors will show my PSU is at around 80c.

The operating temps are only around 30c-50c

Sensors is inaccurately reporting these temps.

The motherboard was released June 2020, so I assumed in Linux there would be good sensor/kernel support by now, unless the sensors on the board are just not good, and that is the problem.

Is there anything that can be done, to get better more accurate PSU temps?

My system specs:

Slackware 15.0
lm_sensors-3.6.0
Asus ROG Strix B550-F Gaming motherboard
Corsair RM750x
Ryzen 5 5600X

This is the output of sensors
```
k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +31.0°C  
Tccd1:        +28.5°C  

nct6798-isa-0290
Adapter: ISA adapter
in0:                      328.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                      1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                      952.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                      248.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                      904.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                     376.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                       1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                     352.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                     256.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     1834 RPM  (min =    0 RPM)
fan2:                     1501 RPM  (min =    0 RPM)
fan3:                     1670 RPM  (min =    0 RPM)
fan4:                     1679 RPM  (min =    0 RPM)
fan5:                        0 RPM  (min =    0 RPM)
fan6:                        0 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +29.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +29.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +85.5°C    sensor = thermistor
AUXTIN1:                   +39.0°C    sensor = thermistor
AUXTIN2:                   -60.0°C    sensor = thermistor
AUXTIN3:                   +80.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +29.0°C  
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
PCH_CHIP_TEMP:              +0.0°C  
PCH_CPU_TEMP:               +0.0°C  
TSI0_TEMP:                 +31.1°C  
intrusion0:               ALARM
intrusion1:               ALARM
beep_enable:              disabled
```

THANKS

Wer-Wolf, 2022-08-21:
Are you sure that AUXIN0/2/3 measures the temperature of your PSU?
Because i see no way how the motherboard should be able to measure the temperature of your ATX PSU.

It could be that your motherboard vendor just didn't connect anything to AUXIN0/2/3, which causes
those channels to display bogus values.

If that is the case, then you can disable those channels inside sensors.conf.

0pLuS0, 2022-08-21:
Sorry for making this issue report.

My mistake, I just assumed there were these sensors, since sensors scanned and gave this output. I've never really spent much time with sensors, just the basics before.

To be honest, I've never even seen Temp Specs for what sensors are on this board.

Wer-Wolf, 2022-08-21:
This sensor chip likely monitors sensors on your motherboard, like cpu temp, chipset temp, fan speeds, etc.

0pLuS0, 2022-08-24:
> This sensor chip likely monitors sensors on your motherboard, like cpu temp, chipset temp, fan speeds, etc.

These typical things I'm aware of, but when I saw for the PSU, I wasn't sure.

No worries...

THANKS