lm-sensors/lm-sensors #411: Cannot control my fans, chip IT5570E

avirgos, 2022-09-07:
OS : Debian 11

Hello.
I want to control my fans because they make too much noise.
I used `sensors-detect` command but my chip is not recognized... 

```
# sensors-detect version 3.6.0
# System: GIGABYTE G5 GD [Not Applicable] (laptop)
# Kernel: 5.10.0-17-amd64 x86_64
# Processor: 11th Gen Intel(R) Core(TM) i5-11400H @ 2.70GHz (6/141/1)

Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x5570
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
```

I put this on my GRUB (`/etc/default/grub`) : ``GRUB_CMDLINE_LINUX="acpi_enforce_resources=lax"``.
Using chip ID, I found that my chip is IT5570E (https://github.com/coreboot/coreboot/blob/master/util/superiotool/ite.c) but this command ``modprobe it87 force_id=0x5570`` doesn't change anything.

`sensors` :
```
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +65.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +52.0°C  (high = +100.0°C, crit = +100.0°C)

nvme-pci-0200
Adapter: PCI adapter
Composite:    +40.9°C  (low  = -60.1°C, high = +89.8°C)
                       (crit = +94.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +56.0°C  (crit = +120.0°C)

nvme-pci-0500
Adapter: PCI adapter
Composite:    +45.9°C  (low  =  -0.1°C, high = +69.8°C)
                       (crit = +84.8°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          16.64 V  
curr1:         0.00 A
```

I cannot use `pwmconfig` command from `fancontrol` package.

```
# pwmconfig version 3.6.0
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

/usr/sbin/pwmconfig: There are no pwm-capable sensor modules installed
```

Any ideas ?

Thanks.

closed 2022-09-11 by avirgos

TheNeikos, 2022-10-12:
@viantoine Did you find a solution to your problem?

viantoine, 2022-10-12:
> @viantoine Did you find a solution to your problem?

Hello @TheNeikos,
I did not find any solutions to my problem at the level of lm-sensors.
I just cleaned my fans with a dusting spray and my fans don't make as much noise as before.
I can reopen this issue if you have a problem with this chip model.