lm-sensors/lm-sensors #414: lm_sensors coretemp-isa-0000 values below ambient temperature

reibuehl, 2022-09-22:
I am trying to setup lm_sensors for an older Zotac IONITX-U-E mainboard. The values I get from the w83667hg-isa-0a10 are all looking good and are in line with what I see in the BIOS. But the values I get from coretemp-isa-0000 are way off. They are around +9/+10°C for both cores while the ambient temperature is already 24°C and the CPU temperature is reported with ~45°C in the bios and from the w83667hg.

Is it possible that these are differences to the reported CPU temp? If so, is there a way to add CPU temp to it since both come from different sensors? Or is it more likely that there is a unknown offset that needs to be added?

I tried adding an offset of 40° with compute temp2 @+40,@+40 but that also increases the critical temperature value from 100 to 140°C which I don't want. Setting the critical threshold with set temp2_crit 100 did not work. It still showed 140.

This is the sensors -u output on the system:
`coretemp-isa-0000
Adapter: ISA adapter
Core 0:
  temp2_input: 8.000
  temp2_crit: 100.000
  temp2_crit_alarm: 0.000
Core 1:
  temp3_input: 7.000
  temp3_crit: 100.000
  temp3_crit_alarm: 0.000

w83667hg-isa-0a10
Adapter: ISA adapter
Vcore:
  in0_input: 1.096
  in0_min: 0.000
  in0_max: 1.744
  in0_alarm: 0.000
Memory Voltage:
  in1_input: 1.552
  in1_min: 1.352
  in1_max: 1.648
  in1_alarm: 0.000
3VCC:
  in2_input: 3.328
  in2_min: 2.976
  in2_max: 3.632
  in2_alarm: 0.000
V1.05:
  in4_input: 1.072
  in4_min: 0.904
  in4_max: 1.104
  in4_alarm: 0.000
V1.50:
  in5_input: 1.504
  in5_min: 1.352
  in5_max: 1.648
  in5_alarm: 0.000
3VSB:
  in7_input: 3.328
  in7_min: 2.976
  in7_max: 3.632
  in7_alarm: 0.000
Vbat:
  in8_input: 2.960
  in8_min: 2.704
  in8_max: 3.632
  in8_alarm: 0.000
SYS Fan:
  fan1_input: 2445.000
  fan1_min: 2860.000
  fan1_alarm: 1.000
  fan1_div: 8.000
CPU Fan:
  fan2_input: 1757.000
  fan2_min: 1308.000
  fan2_alarm: 0.000
  fan2_div: 8.000
M/B Temp:
  temp1_input: 33.000
  temp1_max: 55.000
  temp1_max_hyst: 52.000
  temp1_alarm: 0.000
  temp1_type: 4.000
  temp1_offset: 0.000
CPU Temp:
  temp2_input: 42.500
  temp2_max: 80.000
  temp2_max_hyst: 75.000
  temp2_alarm: 0.000
  temp2_type: 1.000
  temp2_offset: -16.000
`
[Zotac-IONITX-U-E.conf.txt](https://github.com/lm-sensors/lm-sensors/files/9624976/Zotac-IONITX-U-E.conf.txt)

Wer-Wolf, 2022-09-23:
Could you show me the output of `lscpu`?
I suspect that the coretemp driver has no offset informations for you cpu.

reibuehl, 2022-09-23:
Here is the output of lscpu:
`Architektur:                     x86_64
CPU Operationsmodus:             32-bit, 64-bit
Byte-Reihenfolge:                Little Endian
Adressgrößen:                    36 bits physical, 48 bits virtual
CPU(s):                          4
Liste der Online-CPU(s):         0-3
Thread(s) pro Kern:              2
Kern(e) pro Socket:              2
Sockel:                          1
NUMA-Knoten:                     1
Anbieterkennung:                 GenuineIntel
Prozessorfamilie:                6
Modell:                          28
Modellname:                      Intel(R) Atom(TM) CPU D525   @ 1.80GHz
Stepping:                        10
CPU MHz:                         1795.767
BogoMIPS:                        3591.53
L1d Cache:                       48 KiB
L1i Cache:                       64 KiB
L2 Cache:                        1 MiB
NUMA-Knoten0 CPU(s):             0-3
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Mmio stale data:   Not affected
Vulnerability Spec store bypass: Not affected
Vulnerability Spectre v1:        Not affected
Vulnerability Spectre v2:        Not affected
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Markierungen:                    fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts rep_good nopl cpuid aperfmperf pni dte
                                 s64 monitor ds_cpl tm2 ssse3 cx16 xtpr pdcm movbe lahf_lm dtherm
`

Wer-Wolf, 2022-09-23:
Try update your BIOS and microcode.
Which kernel version do you use?
Does the problem persist if the cpu reaches a high temperature (ca. 60 °C)?

reibuehl, 2022-09-23:
The BIOS is the latest that was released for this fairly old board. I use the Ubuntu 5.4.0-125-generic Kernel package.
I haven't seen any really high CPU temperatures yet. Can you give me a hint how I could get the temperature up without too much effort?

Wer-Wolf, 2022-09-23:
`stress --cpu 64`

reibuehl, 2022-09-27:
I couldn't get the CPU temperature over 48.5°C with the stress command.

reibuehl, 2022-09-27:
No, the the difference was constant. The increase in coretemp was linear to the increase in cpu temp.

reibuehl, 2022-09-27:
I guess I have to file the bug then against the Ubuntu kernel as I don't know if I manage to switch to a unmodified mainline kernel as required for a bug report on the kernel.
Thanks for your assistance!

closed 2022-09-27 by reibuehl