lm-sensors/lm-sensors #416: Asus X670E TUF Gaming

scoony, 2022-10-12:
Hello, I've got a Asus X670E TUF gaming Plus Wifi ... so far I can only get infos about my nVME.

Any way to help to get the support for this motherboard?

scoony, 2022-10-12:
It's a Nuvoton NCT6799D-R

MostHated, 2022-10-16:
Unfortunately, I get the same with an Asrock Taichi x670e. I am guessing we just have to wait a bit for x670 in general?

tymscar, 2022-10-17:
Very strange. I have a friend with an Asus X670E Hero and he can see the temps just fine for his 7950X.
I am on a Gigabyte X670E Aorus Master and I can see no temps for my 7950X. We are both on Arch Linux, with the same Kernel version (6.0.2).

MostHated, 2022-10-17:
I am glad you mentioned that. I didn't realize that I needed to be on kernel 6.0.x. I use Pop_OS and it is still on 5.19. I moved to 6.0 just now and wouldn't you know it, temps and such now show up. 
Thanks for that. :+1:

tymscar, 2022-10-17:
So is it only Gigabyte motherboards with x670e on 6.0.2 that dont work? Is there something else I could try?

scoony, 2022-10-17:
I've tried again and if i try to use the 6.0.0 or 6.0.1 the system won't boot... weird

Wer-Wolf, 2022-10-21:
Any error messages during boot?

MarkusH1975, 2022-10-26:
I have as well Asus X670E TUF Gaming Plus Wifi +  AMD Ryzen 7 7700X with integrated Graphics.
Running Ubuntu 22.04 LTS with Kernel 5.15.0-52.  (existing installation was used from old PC)

lm-sensors is detecting nearly nothing. No CPU temperature, no fan speed..... 

@scoony 
> I've tried again and if i try to use the 6.0.0 or 6.0.1 the system won't boot... weird

My system didn't boot as well. This was related due to old graphics driver of AMD. Try to use "nomodeset" in kernel parameters, booting should work, if this is the problem. 

After installing AMDGPU pro driver graphics is working fine. 
Attention official linked version is 
[https://repo.radeon.com/amdgpu-install/22.20.3/ubuntu/jammy/](https://repo.radeon.com/amdgpu-install/22.20.3/ubuntu/jammy/)
But there is a newer one (complete different version, but newer date!)
[https://repo.radeon.com/amdgpu-install/5.3/ubuntu/jammy/](https://repo.radeon.com/amdgpu-install/5.3/ubuntu/jammy/)

only 5.3 was working for me.

I don't know if AMDGPU pro driver already supports kernel 6.0.x.

scoony, 2022-10-26:
@MarkusH1975 

Thanks a lot, installed and now i can use Wayland.

Same for me, now i've got infos about iGPU and nVME thru lm-sensors, with 6.0.3 you can get the CPU temp but when i tried my iGPU wasn't working properly with this kernel so I had to put a graphic card. Didn't try with the iGPU driver you've provided.

starsforeveryone, 2022-11-12:
I have this motherboard and can confirm it's still not fully supported. :(

CPU temp, iGPU and nVME work. But there is no control over the fans (CPU fan and chassis fans), and the PSU's voltages are not shown either. In fact, lm-sensors doesn't even detect a chip when running sensors-detect.

I can confirm this board (ASUS TUF GAMING X670E-PLUS WIFI) has indeed a Nuvoton NCT6799D-R chip.

The FOSS Windows (!) software "LibreHardwareMonitor" seems to have access to a fix that enables fan control:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/846
Several users on the internet report that the nighly code from GitHub works now.

Would it be possible for someone from the lm-sensors team to adapt these changes to lm-sensors to get the board working?

chunkeey, 2022-11-14:
I have a ASRock X670E PG in my case the chip is a NCT6796D-S but it has the same chip ID (0xd802) as NCT6799D-R.
FYI: the nct6775 kernel module can be used with, even no real patch exists:
```sudo modprobe nct6775 force_id=0xd420``` (for the NCT6796D for a list see the kernel's nct6775-core.c) parameter. 
(and then just "sensors" to look at the raw values)

Though in my case at least two fans speed sensors are missing.

scoony, 2022-11-14:
@chunkeey I've tried your command on mine and nothing.

chunkeey, 2022-11-14:
> @chunkeey I've tried your command on mine and nothing.

Yea, the nct6775.c provides a reason for this. ASUS marks the Super-IO IO-Ports as being is "use" so the kernel module can't have access. You would need to add your board's identification to the list first:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hwmon/nct6775-platform.c#n1044

so it will use the ASUS-WMI interface to access the chip.

(ASRock allows direct io-access, so this isn't an issue there)

MostHated, 2022-11-15:
I just tried using "sudo modprobe nct6775 force_id=0xd420" on my Asrock  x670e Taichi and it worked wonders:

Running `sensors` is now jam-packed. Thanks for that!

chunkeey, 2022-11-15:
> Asrock x670e Taichi and it worked wonders:

That board certainly has an interesting SuperIO-Chip setup.  The [Manual for the ASRock X670E Taichi](https://download.asrock.com/Manual/X670E%20Taichi_English.pdf) has the Overview on page 11:
[image]

so, you should have two Super-IO Chips (NCT6796D-S + NCT6686D) in your sensor dump!

can you please share your ```sensors``` output and possibly issues that you see (i.e. are all the fans there and have the right fan speeds & pwm settings. Note: Voltages reading are less trivial (in other words: almost all are in the wrong range). This is because the chip's ADC can only measure in the 0 - 2.0V range, So if you take 12V, ASRock needs to wired up through resistor dividers(-networks) to get that down into the range. The NCT6796D Datasheet talks about this in their analog Section.)

MostHated, 2022-11-15:
Yeah, I only gave it a quick glance at first and noticed additional fans. 

Looking deeper into it, two sections did show up, it looks like one of them (nct6796-isa-0a20) is more in line with what I was expecting to see in terms of the number of fans, etc, but doesn't seem to be reporting properly. All voltages are either 2.04 or 4.08, all dans are seemingly an int.MaxValue, etc.

nct6796-isa-0290 is the one in which I saw additional fans show up, but it looks a bit off as well.

My actual setup is:

CPU_Fan1 is running my AIO's radiator fans (showing 0 RPM)  
CPU_Fan2 is my 3_amp port, which is set to pump mode and full speed, but I am not sure what that should be reporting actually, as I bought it new for this particular build.  
CHA_Fan1 is driving a deepcool pwm hub.
CHA_Fan2 and I believe 5 are driving independent 1500-4000RPM pwms

Here is a sensors dump:

<details>
<summary>Details</summary>

```log
nct6796-isa-0a20
Adapter: ISA adapter
Vcore:          2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in1:            2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
AVCC:           4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
+3.3V:          4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
in4:            2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in5:            2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in6:            2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
3VSB:           4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
Vbat:           4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
in9:            2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in10:           2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in11:           2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in12:           2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in13:           2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in14:           2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
fan1:         65535 RPM  (min =    0 RPM)  ALARM
fan2:         65535 RPM  (min =    0 RPM)  ALARM
fan3:         65535 RPM  (min =    0 RPM)  ALARM
fan4:         65535 RPM  (min =    0 RPM)  ALARM
fan5:         65535 RPM  (min =    0 RPM)  ALARM
fan6:         65535 RPM  (min =    0 RPM)  ALARM
fan7:         65535 RPM  (min =    0 RPM)
SYSTIN:         -1.0°C  (high =  -1.0°C, hyst =  -1.0°C)  sensor = CPU diode
CPUTIN:         -1.0°C  (high =  +0.0°C, hyst =  +0.0°C)  sensor = CPU diode
AUXTIN0:        -1.0°C    sensor = CPU diode
AUXTIN1:        -1.0°C    sensor = CPU diode
AUXTIN2:        -1.0°C    sensor = CPU diode
AUXTIN3:        -1.0°C    sensor = CPU diode
Virtual_TEMP:   -1.0°C  (high =  -1.0°C, hyst =  -1.0°C)  ALARM
Virtual_TEMP:   +0.0°C                                    ALARM
Virtual_TEMP:   +0.0°C                                    ALARM
Virtual_TEMP:   +0.0°C                                    ALARM
TSI0_TEMP:    +3892314.0°C  
TSI1_TEMP:    +3892314.0°C  
intrusion0:   ALARM
intrusion1:   ALARM
beep_enable:  enabled

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +28.0°C  

hidpp_battery_0-hid-3-c
Adapter: HID adapter
in0:           3.93 V  

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      787.00 mV 
[... 47 lines cut ...]
beep_enable:           disabled

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +63.5°C  
Tccd1:        +69.8°C  
Tccd2:        +48.9°C  

amdgpu-pci-5900
Adapter: PCI adapter
vddgfx:        1.38 V  
vddnb:         1.19 V  
edge:         +47.0°C  
PPT:          55.02 W  

nvme-pci-5300
Adapter: PCI adapter
Composite:    +42.9°C  (low  = -40.1°C, high = +83.8°C)
                       (crit = +87.8°C)
Sensor 1:     +61.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +42.9°C  (low  = -273.1°C, high = +65261.8°C)

```

</details>

chunkeey, 2022-11-15:
> Looking deeper into it, two sections did show up, it looks like one of them (nct6796-isa-0a20) is more in line with what I was expecting to see in terms of the number of fans, etc, but doesn't seem [...]

The nct6796-isa-0a20 could very well be the NCT6686D! The "force_id=" parameter makes the module ignore the ID mismatches. You should ``rmmod``` both nct6775 and nct6683 modules and then load the nct6683 module first so it hopefully is able to pick up the chip at 0x0a20 (check ```sensors```). And after that you can try to load nct6775 with the force_id and see if it still picks up the 0x290.

My ASRock X670E PG dump:

<details>

    nct6796-isa-0290
    Adapter: ISA adapter
    Vcore:                720.00 mV (min =  +0.00 V, max =  +1.74 V)
    in1:                    1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
    AVCC:                   3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
    +3.3V:                  3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
    in4:                    1.66 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
    in5:                    1.05 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
    in6:                    1.46 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
    3VSB:                   3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
    Vbat:                   3.23 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
    in9:                  904.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
    in10:                 608.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
    in11:                 608.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
    in12:                   1.04 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
    in13:                 896.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
    in14:                   1.26 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
    fan1:                  815 RPM  (min =    0 RPM)
    fan2:                    0 RPM  (min =    0 RPM)
    fan3:                  781 RPM  (min =    0 RPM)
    fan7:                    0 RPM  (min =    0 RPM)
    SYSTIN:                +26.5°C    sensor = thermistor
    CPUTIN:                +27.0°C  (high = +80.0°C, hyst = +75.0°C)
    AUXTIN0:                +2.0°C  (high = +67.0°C, hyst = +110.0°C)
    AUXTIN1:               +15.0°C    sensor = thermistor
    AUXTIN2:               +15.0°C    sensor = thermistor
    AUXTIN3:               +13.0°C    sensor = thermistor
    SMBUSMASTER 0:         +39.5°C  (high = +80.0°C, hyst = +75.0°C)
    PCH_CHIP_CPU_MAX_TEMP:  +0.0°C
    PCH_CHIP_TEMP:          +0.0°C
    PCH_CPU_TEMP:           +0.0°C
    TSI0_TEMP:             +39.9°C
    intrusion0:           ALARM
    intrusion1:           ALARM
    beep_enable:          disabled
`

Of some note is:
 - voltages for Vcore, 3V3, AVCC, 3VSB and Vbat have sensible values.
   The other voltage inputs are likely "limited to a maximum value of
   2.048V due to an internal setting of 8mV LSB (256 steps x 8mV = 2.048V)"
   [1] (Section 8.6 Analog Inputs - page 58)

 - temperatures reported by SYSTIN (mainboard), TSI0+SMBUSMASTER (CPU-CCDs)
   matches too. CPUTIN really seems to be the IO-DIE of the CPU.

 - fan1 rpms matches the AIO (2nd CPU fan).
   fan3 is the 2nd chassis fan.

   The motherboard has pin-headers for 2 CPU-fans and 4 Chassis/AUX-fans.
   This means that two are missing! The novuton documentation [1] says on
   page 3 that there are 6 fan controls + monitoring inputs.

   what's odd is that pwm1-5 values provided through the sysfs interface.
   and these are matching the fan curves that have been set in the bios.

 - intrusion is not connected, same with all PCH* sensors.
   AUXTIN123 are not moving. AUXTIN0 is moving with
 </details>

chunkeey, 2022-11-17:
@MostHated 

I see. Could be that they also invented a new ID for the nct6683... just like they did with the NCT6796D-S.
What really sucks is that your chips are hidden behind the heatsink/board armor, so you can't easily look
at those without disassembling your PC. 

Another way to look at the chips is though software with the `sensors-detect` utility.
ie. `sudo sensors-detect` (usually in /sbin, this is part of lm-sensors already).

This utility will either tell you which is the correct module or report the chip-id of
the unknown/unsupported sensor. You don't need to run the utility to completion.
The NCT6796D-S has 0xd802. The nct6683 should be ["0xc730"](https://github.com/torvalds/linux/blob/master/drivers/hwmon/nct6683.c#L14) but who knows.

watch-out for the part where it says:
"Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe. "

you can skip/break any step afterwards.

MostHated, 2022-11-17:
Here is what I was able to get out of it.

```log

Do you want to scan for Super I/O sensors? (YES/no): yes
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd802
    (logical device B has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd441

```

<details>
<summary>Full Output</summary>

```log

zsh ❯ sudo sensors-detect
sensors-detect version 3.6.0
System: Default string X670E Taichi [Default string]
Board: ASRock X670E Taichi
Kernel: 6.0.8-060008-generic x86_64
Processor: AMD Ryzen 9 7950X 16-Core Processor (25/97/2)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): Yes
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): yes
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd802
    (logical device B has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd441

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): yes
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): yes
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

```

</details>

kschat, 2022-11-23:
I'm experiencing the same issue as @MostHated. My motherboard is an ASRock B650E PG-ITX WiFi, which comes with the NCT6686D. After running `modprobe nct6683 force_id=0xd440`, I received the following output from `sensors-detect`:

```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd441
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
``` 

One thing I noticed is the `force_id` parameter seems to be ignored:

```
 —— sudo dmesg  | rg nct
[ 1442.064178] nct6683: Found NCT6686D or compatible chip at 0x2e:0xa20
[ 2438.203531] nct6683: unknown parameter 'force_id' ignored
[ 2438.203642] nct6683: Found NCT6686D or compatible chip at 0x2e:0xa20
```

starsforeveryone, 2022-11-24:
> @chunkeey I've tried your command on mine and nothing.

Same with me. On my ASUS TUF GAMING X670E-PLUS WIFI the command "sudo modprobe nct6775 force_id=0xd420" does nothing. I'm on latest Fedora 37 with latest Kernel version 6.1-rc6 installed for this experiment.

Seems that the Nuvoton NCT6799D-R chip is not recognized even by the latest mainline kernel. :-/

Phoronix News had an article about upcoming ASUS support for Nuvoton NCT6775/NCT6798D motherboards in Kernel 6.2, but sadly, nothing about NCT6799D-R.

starsforeveryone, 2022-11-27:
Anybody got any ideas? If we could get this chip running, we'd probably enable support for a lot of new AM5 motherboards on the market!

I'm willing to help testing

wdcocq, 2022-12-27:
> One thing I noticed is the `force_id` parameter seems to be ignored:
>
> ```
>  —— sudo dmesg  | rg nct
> [ 1442.064178] nct6683: Found NCT6686D or compatible chip at 0x2e:0xa20
> [ 2438.203531] nct6683: [...]

the nct6683 driver doesn't support `force_id`, but does support `force` (bool).
on my ASRock x670e Steel legend `modprobe nct6883 force=1` works

I still get `Found unknown chip with ID 0xd441`, but I do get the following values with `sensors` now:

```
nct6686-isa-0a20
Adapter: ISA adapter
VIN0:             912.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN1:             992.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN2:               1.01 V  (min =  +0.00 V, max =  +0.00 V)
VIN3:               1.68 V  (min =  +0.00 V, max =  +0.00 V)
VIN5:               1.26 V  (min =  +0.00 V, max =  +0.00 V)
VIN6:               1.12 V  (min =  +0.00 V, max =  +0.00 V)
VIN7:               1.34 V  (min =  +0.00 V, max =  +0.00 V)
VCC:                3.36 V  (min =  +0.00 V, max =  +0.00 V)
fan1:             1468 RPM  (min =    0 RPM)
fan2:             3508 RPM  (min =    0 RPM)
fan3:              438 RPM  (min =    0 RPM)
fan4:                0 RPM  (min =    0 RPM)
fan5:              329 RPM  (min =    0 RPM)
fan6:              337 RPM  (min =    0 RPM)
Thermistor 16:     +29.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
Thermistor 14:     +30.0°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
Thermistor 15:     +29.0°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
AMD TSI Addr 98h:  +39.0°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = AMD AMDSI
intrusion0:       OK
beep_enable:      disabled
```

starsforeveryone, 2022-12-28:
I'm happy to report that I've contacted Guenter Roeck from the Linux Kernel hwmon team about this issue, and that Nuvoton has kindly provided him with the necessary datasheet for the NCT6799D-R chip so that he can write a patch to the existing driver for the NCT67xx series. He'll be issueing the patch within the next couple of days.

I've also contacted Nuvoton as a customer and asked them about whether they'd be willing to work more closely together with the hwmon team in the future, e.g. providing them with the necessary datasheets whenever they release a new chip. Given the already extensive support for their chips in the Linux kernel, this would be a logical step and further add support for their hardware. I don't know whether Guenter contacted them directly, or whether they sent him the datasheet as a reply to my own E-Mail, but anyway, they do care about Linux support and that is a valuable thing to know.

I'll update you guys as soon as the patch is released.

starsforeveryone, 2022-12-30:
Patch is here and seems to be working as intended. I'll compile it and report whether it works for me, too. In the meantime, feel free to check it out! (You'll have to bake your own Linux kernel, though - download the patch as a .diff file and then apply it to the source code, after that compile it; there are a lot of tutorials out there on how to do it)

https://patchwork.kernel.org/project/linux-hwmon/patch/20221228135744.281752-1-linux@roeck-us.net/

starsforeveryone, 2022-12-31:
Sadly, for me and my ASUS motherboard, it doesn't work. ASUS seems to do its own thing again...
But for everyone who has the chip on a different vendor's board, this is good news! :)

SkyBeam, 2023-01-04:
> Sadly, for me and my ASUS motherboard, it doesn't work. ASUS seems to do its own thing again...

You might try to add the acpi_enforce_resources=lax kernel parameter.
I did personally apply the [patch](https://patchwork.kernel.org/project/linux-hwmon/patch/20221228135744.281752-1-linux@roeck-us.net/) (on my Gentoo Linux kernel 6.1.3 and added acpi_enforce_resources=lax kernel boot parameter.
After loading the module (modprobe nct6775) I can see the sensors:

```
nct6799-isa-0290
Adapter: ISA adapter
in0:                        1.41 V  (min =  +0.00 V, max =  +1.74 V)
in1:                      1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                        1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                      376.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                        1.66 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                     560.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                     560.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                     456.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                     400.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                      989 RPM  (min =    0 RPM)
fan2:                      846 RPM  (min =    0 RPM)
fan3:                      736 RPM  (min =    0 RPM)
fan4:                        0 RPM  (min =    0 RPM)
fan5:                        0 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +34.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +37.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +71.0°C    sensor = thermistor
AUXTIN1:                   +19.0°C    sensor = thermistor
AUXTIN2:                   +20.0°C    sensor = thermistor
AUXTIN3:                   +77.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +37.5°C
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C
PCH_CHIP_TEMP:              +0.0°C
PCH_CPU_TEMP:               +0.0°C
TSI0_TEMP:                 +48.8°C
intrusion0:               ALARM
intrusion1:               OK
beep_enable:              disabled
```

Mainboard: ASUS PRIME B650-PLUS

starsforeveryone, 2023-01-05:
I am the greatest idiot on the planet. Thank you! I did everything correctly and then forgot to load the module. Facepalm. Now everything works on my motherboard. :)

**Here comes the definite guide on how to get your sensors working: (Noob-friendly!)**
- Read up on how to compile a kernel for the specific Linux distribution you're using. Take your time to fully read and understand the process! Prepare your system by installing the necessary tools for compiling as described in the instructions for your distribution.
- Download the latest "stable" Linux kernel from https://www.kernel.org/ (or one fitting for your needs). Unpack it.
- Download the patch from [here](https://patchwork.kernel.org/project/linux-hwmon/patch/20221228135744.281752-1-linux@roeck-us.net/raw/).
- Open a Terminal and cd into the folder with your unpacked kernel.
- Copy your current kernel configuration file: `cp /boot/config-`uname -r`* .config`
- Open the file _Makefile_  in a text editor of your choice. Change the EXTRAVERSION parameter to something unique. It's there so that you'll recognize your self-compiled kernel in the boot menu later, and so that it won't name conflict with any other installed kernels on your system. Close the text editor.
- In the Terminal, apply the patch: `sudo patch -p1 < /path/to/your/patch.diff`
- Compile the Linux kernel: (I'm doing this on a CPU with 16 cores / 32 threads, that's why I added the -j 32 parameter; find out how many threads your CPU has and change the number 32 to your own thread count. You can find out by counting the number of "CPU cores" that are actually visible in your system, e.g. in GNOME System Monitor. Or you can look it up in the Intel / AMD product specs for your CPU model. It makes sense to look this up because then it'll utilize the full multi-threaded power of your CPU for the job, which makes it go a lot faster! Without the -j parameter, it'll be single-threaded.)
```
sudo make oldconfig -j 32 (just fight your way through any requests it has...)
sudo make bzImage -j 32
sudo make modules -j 32
```
- Install the Linux kernel:
```
sudo make modules_install
sudo make install
```
- Update the bootloader (I'm assuming you're using GRUB):
On Debian/Ubuntu: `sudo update-grub`
On Fedora: `sudo grub2-mkconfig -o /boot/efi/EFI/fedora/grub.cfg`
- Reboot your system. Be careful to pick your new kernel in the boot menu!
- Actually load the module by using the command: `sudo modprobe nct6775`
- If you run `sensors`, your sensors should be detected now!

If it works, add the text _nct6775_ to /etc/modules - this will load the module (=driver) automatically on boot.

If it doesn't work, you might need to add the aforementioned _acpi_enforce_resources=lax_ to your GRUB_CMDLINE_LINUX parameter in the file /etc/default/grub, and then update the bootloader again as described above. But read up on what this does, first!

The things you do for ASUS.... 🧐

**Much thanks to Guenter Roeck 👍 from the Linux hwmon driver team (who wrote the kernel module) &
much thanks to Nuvoton 👍 for providing the datasheet for the chip!**

kschat, 2023-01-05:
> the nct6683 driver doesn't support force_id, but does support force (bool).
> on my ASRock x670e Steel legend modprobe nct6883 force=1 works

Thanks! That works for my ASRock B650E PG-ITX WiFi, too.

MBilderbeek, 2023-04-20:
In my case:
# Board: ASUSTeK COMPUTER INC. ProArt X670E-CREATOR WIFI
# Processor: AMD Ryzen 9 7950X 16-Core Processor (25/97/2)

it helped a lot to do:
`modprobe nct6775 force_id=0xd420`

Next problem is to figure out what all these sensors mean... as I have no idea. Did someone ever make a mapping for these? lm-sensors config file most welcome! :)

DrMcCoy, 2023-04-24:
My board is a ASUS System Product Name/TUF GAMING X670E-PLUS WIFI, BIOS 1222 02/24/2023, seems to work somewhat with force_id=0xd420.

Some values, especially the voltages, seem off, though.

fan1, fan2, fan3 and fan6 are the 4 chassis fans. I'm not seeing the CPU fans, though, at all.
SYSTIN seems to be the motherboard temperature? At least that maps with what I see in the BIOS, I think.

Is there something I can do to help with the mapping?

Wer-Wolf, 2023-04-24:
sensors.conf can be used to do the mapping

KeithMyers, 2023-07-20:
> I have a ASRock X670E PG in my case the chip is a NCT6796D-S but it has the same chip ID (0xd802) as NCT6799D-R. FYI: the nct6775 kernel module can be used with, even no real patch exists: `sudo modprobe nct6775 force_id=0xd420` (for the NCT6796D for a list see the kernel's nct6775-core.c) . (and then just "sensors" to look at the raw values)
> 
> Though in my case at least two fans speed sensors are missing.

If you upgrade to the 6.5-rc kernels, you can get proper support for the nct6799D chip.  All the fans are exposed and not missing like they are with the 6796 chipID

MBilderbeek, 2023-08-23:
I'm running such a kernel:
Linux creator 6.5.0-0-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.5~rc6-1~exp1 (2023-08-15) x86_64 GNU/Linux

But which kernel module to load instead of nct6775?

KeithMyers, 2023-08-23:
The nct6775 kernel module covers all the Nuvoton chip variations so far produced. Just loading nct6775 in the modules.conf file is enough to get the sensors exposed from the SIO chip.

But have you verified that your motherboard actually uses a Nuvoton SIO chip?  They are many more SIO chip vendors like Fintek, ITE, National Semiconductor, Nuvoton, SMSC, VIA, and Winbond.

When you run sudo sensors-detect, it should find 'something' that looks like a SIO chip to the application.  Though it might not understand exactly how to utilize it and install the proper driver for it.

That was the situation with the newest Nuvoton 6798 and 6799 chips.  Had to wait for the HWMON tree maintainers to get new code into the nct6775 module to properly identify those chips and expose their sensors.  Took the later 6.5 kernels to finally accomplish this.

MBilderbeek, 2023-08-23:
```
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8883
    (logical device 4 has address 0x290, could be sensors)
```
Does that tell you which chip it is?

MBilderbeek, 2023-08-23:
Ah, what I did to try, I just removed the force in ` /etc/modprobe.d/nct6775.conf` and modprobed manually (without the force flag of course):
```
[23894.824100] nct6775: Found NCT6799D or compatible chip at 0x2e:0x290
```
This output is identical to before, but does seem to give me new sensors... so I guess that works:
```
nct6799-isa-0290
Adapter: ISA adapter
in0:                      952.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                      1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.38 V  (min =  +2.98 V, max =  +3.63 V)
in3:                        3.31 V  (min =  +2.98 V, max =  +3.63 V)
in4:                        1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                        1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                        1.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.38 V  (min =  +2.98 V, max =  +3.63 V)
in8:                        3.25 V  (min =  +2.70 V, max =  +3.63 V)
in9:                        3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                       1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                       1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                       1.29 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                       1.22 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                      895 RPM  (min =    0 RPM)
fan2:                      473 RPM  (min =    0 RPM)
fan3:                        0 RPM  (min =    0 RPM)
fan4:                        0 RPM  (min =    0 RPM)
fan5:                      881 RPM  (min =    0 RPM)
fan6:                      876 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +29.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +38.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +18.0°C    sensor = thermistor
AUXTIN1:                   +22.0°C    sensor = thermistor
AUXTIN2:                   +22.0°C    sensor = thermistor
AUXTIN3:                   +14.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +39.5°C  
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
PCH_CHIP_TEMP:              +0.0°C  
PCH_CPU_TEMP:               +0.0°C  
TSI0_TEMP:                 +50.8°C  
intrusion0:               OK
intrusion1:               ALARM
beep_enable:              disabled
```

The question is now what the  meaning is of all these names...

KeithMyers, 2023-08-23:
You should be getting k10temp by default also since it is already included in the kernel.  That gives you Tctl and Tdie temps.

vgropp, 2023-09-01:
this successfully enables `nct6799-isa-0290`  on `Asus ProArt X670E-CREATOR WIFI` with kernel `6.5.0-0-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.5~rc7-1~exp1`

cpu fan is reported as fan2, cpu-fan2 seems missing, but is not shown in bios either.

DrMcCoy, 2023-09-02:
I also see only one of my two CPU fans in lmsensors, even though two are shown in the BIOS for me.

vgropp, 2023-09-03:
@DrMcCoy you are correct. I did take another look. i can see optional cpu fan in BIOS, but its missing in BIOS Fan Utility to set the speed. As well as its missing in nct6799

KeithMyers, 2023-09-03:
They usually pair-link CPU Opt Fan to the CPU Fan.  Whatever speed and controls are set for the CPU Fan are applied to the CPU Opt header also. So the BIOS normally does not even bother displaying it in the Monitor section.

perroboc, 2023-09-23:
Just wanted to share that loading `nct6775` with `6.5.3-arch1-1` kernel, I get these results now

```console
$ cat /etc/modules-load.d/nct6775.conf
## Load nct6775
nct6775
$ sensors
nct6799-isa-0290
Adapter: ISA adapter
in0:                        1.43 V  (min =  +0.00 V, max =  +1.74 V)
in1:                        1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                        1.04 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                      416.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                        3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                       1.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                       1.12 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.04 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                     216.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                       2.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                        0 RPM  (min =    0 RPM)
fan2:                      631 RPM  (min =    0 RPM)
fan3:                      631 RPM  (min =    0 RPM)
fan4:                      636 RPM  (min =    0 RPM)
fan5:                      510 RPM  (min =    0 RPM)
fan6:                        0 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +24.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +39.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +66.0°C    sensor = thermistor
AUXTIN1:                    +6.0°C    sensor = thermistor
AUXTIN2:                   +19.0°C    sensor = thermistor
AUXTIN3:                   -60.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +39.5°C  
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
PCH_CHIP_TEMP:              +0.0°C  
PCH_CPU_TEMP:               +0.0°C  
TSI0_TEMP:                 +51.9°C  
intrusion0:               ALARM
intrusion1:               OK
beep_enable:              disabled
```

After messing around with speed profiles in BIOS and seeing results with `sensors`, I can tell this is how the fan speeds translate:

| UEFI       | sensors | comment |
|------------|---------|---------|
| `CPU`      | `fan2`  | seems OPT in included since it's the same profile?
| `CHA1`     | `fan1`
| `CHA2`     | `fan3`
| `CHA3`     | `fan4`
| `CHA4`     | `fan5`
| `W_PUMP+`  | `fan6`  | this is me guessing, since I don't have a water cooling pump
| `AIO PUMP` | `fan7`  | this is me guessing, since I don't have a water cooling pump

> They usually pair-link CPU Opt Fan to the CPU Fan. Whatever speed and controls are set for the CPU Fan are applied to the CPU Opt header also. So the BIOS normally does not even bother displaying it in the Monitor section.

Note that the UEFI menu does show both CPU and CPU OPT as different sensors, even showing different RPM values.

KeithMyers, 2023-09-23:
I concur.  This is the pairing I see on my Asus 670E Hero motherboard.

AIO Pump is Fan7 on my board.

Assume that W_Pump+ must be Fan 6 then as its the only one unaccounted for and unpopulated on my host.

esiqveland, 2023-10-14:
On my `ASUS TUF GAMING X670E-PLUS WIFI`,  I can confirm that `fan7` is `AIO_PUMP`.

This works for my mapping in sensors.conf: `/etc/sensors3.conf` on arch linux, or add it to something like `/etc/sensors.d/ASUS_TUF_X670E_PLUS_WIFI`

```
chip "nct6799-isa-0290"
    label fan1 "CHA1"
    label fan2 "CPU"
    label fan3 "CHA2"
    label fan4 "CHA3"
    label fan5 "CHA4"
    label fan6 "W_PUMP+"
    label fan7 "AIO_PUMP"
```

I manually had to load kernel module `modprobe nct6775`.

Thank you all for the great work on this!

MBilderbeek, 2023-11-05:
I have the nct6775 loaded (on kernel 6.5.8) and I do not have an AIO_PUMP (no water cooling connected), but still I see data for fan1, fan2 and fan6.
Maybe it's a bit different for the Asus ProArt X670E-WiFi, which I have?

Wer-Wolf, 2023-11-05:
Could be, can you access more sensor inside the BIOS setup menu?

MBilderbeek, 2023-11-05:
What do you mean exactly? I think all sensors are shown that appear in the BIOS menu (but I haven't checked recently), it's just that the mapping to these numbers seems different than what @esiqveland reported on my mobo.

Wer-Wolf, 2023-11-05:
Since @esiqveland  uses an `ASUS TUF GAMING X670E-PLUS WIFI` and you are using an `Asus ProArt X670E-WiFi`, the reason for the different mappings likely is that Asus wired the sensors differently. This is the norm for different mainboard models.

Thorazin, 2023-11-27:
Hello all, I also have an X670E Pro Art WiFi mainboard and when I do " modprobe nct6775 id=0xd420" in Terminal I am able to see all my sensors. However, after a reboot it is gone again. How do I make this permanent?

sensors
amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      477.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3200 RPM)
edge:         +30.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +35.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +49.0°C  (crit = +108.0°C, hyst = -273.1°C)
                       (emerg = +113.0°C)
PPT:          14.00 W  (cap = 315.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +47.1°C  
Tccd1:        +42.0°C  
Tccd2:        +36.0°C  

eno2-pci-0c00
Adapter: PCI adapter
PHY Temperature:  +44.0°C  
MAC Temperature:  +44.0°C  

nct6799-isa-0290
Adapter: ISA adapter
in0:                        1.06 V  (min =  +0.00 V, max =  +1.74 V)
in1:                      992.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.33 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                        1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                        1.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.26 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                        3.33 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                       1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                       1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                       1.29 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                       1.23 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                        0 RPM  (min =    0 RPM)
fan2:                      476 RPM  (min =    0 RPM)
fan3:                        0 RPM  (min =    0 RPM)
fan4:                      752 RPM  (min =    0 RPM)
fan5:                        0 RPM  (min =    0 RPM)
fan6:                        0 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +22.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +34.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +18.0°C    sensor = thermistor
AUXTIN1:                   +23.0°C    sensor = thermistor
AUXTIN2:                   +22.0°C    sensor = thermistor
AUXTIN3:                   +14.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +36.0°C  
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
PCH_CHIP_TEMP:              +0.0°C  
PCH_CPU_TEMP:               +0.0°C  
TSI0_TEMP:                 +47.1°C  
intrusion0:               OK
intrusion1:               ALARM
beep_enable:              disabled

nvme-pci-0400
Adapter: PCI adapter
Composite:    +36.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +38.9°C  (low  = -273.1°C, high = +65261.8°C)

Wer-Wolf, 2023-11-27:
I suggest to use `/etc/modules` for autoloading and `/etc/modprobe.d/` for configuring the module params.

Thorazin, 2023-11-27:
> I suggest to use `/etc/modules` for autoloading and `/etc/modprobe.d/` for configuring the module params.

Would I just add "nct6775" to /etc/modules?

Wer-Wolf, 2023-11-27:
Yes, and then create a nct6775.conf file under `/etc/modprobe.d/` containing `options nct6775 id=0xd420`.

Thorazin, 2023-11-27:
> Yes, and then create a nct6775.conf file under `/etc/modprobe.d/` containing `options nct6775 id=0xd420`.

Awesome, thank you.

KeithMyers, 2023-11-28:
Depending on your OS and kernel level, you don't have to force load the address in the newer kernels.  The support for the nct6799D chipset is already baked into the nct6775 driver.

All that is necessary is add the nct6775 module name to the modules.conf file.

Thorazin, 2023-11-28:
Ubuntu 23.10 and kernel 6.5 as up to date as possible

KeithMyers, 2023-11-28:
Using the same OS and kernel.  Asus X670E Crosshair Hero.  Didn't need to force load the address. Try it without the address, just the nct6775.

Thorazin, 2023-11-28:
Appears to be working fine, I deleted the file in /etc/modprobe.d/ and restarted, opened terminal and ran "sensors" and voila, still there.

KeithMyers, 2023-11-28:
Yes, the force load of the nct6796 address was only necessary back on < 6.5 kernels.
Use of that address was _mostly_ correct except for labels and Fan4 was missing.

krikun98, 2024-01-19:
Added a PR to automatically recognize `nct6799-isa-0290` through `lm-sensors`: https://github.com/lm-sensors/lm-sensors/pull/480

[7 more replies]