lm-sensors/lm-sensors #422: core order out of place ?

kub3let, 2022-12-25:


Wer-Wolf, 2022-12-27:
This seems to be an issue with the kernel driver. You could try to see if this bug still exists when running the latest kernel, and if yes, open a bugreport on bugzilla.kernel.org

mikeyo, 2022-12-27:
> For some reason the core order is out of place, this makes htop also show n/a temperatures
> 
> [image]
> 
> ```
> # sensors-detect version 3.6.0
> # Kernel: 6.1.0-1 x86_64
> # Processor: 13th Gen Intel(R) Core(TM) i5-13600K (6/183/1)
> 
> OS: Debian 11
> libsensors-config/stable,now 1:3.6.0-7 all [installed,automatic]
> libsensors5/stable,now 1:3.6.0-7 amd64 [installed,automatic]
> lm-sensors/stable,now 1:3.6.0-7 amd64 [installed]
> ```
> 
> ```
> root@host:~# sensors
> coretemp-isa-0000
> Adapter: ISA adapter
> Package id 0:  +36.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 0:        +33.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 4:        +32.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 8:        +31.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 12:       +33.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 16:       +32.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 20:       +31.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 24:       +35.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 25:       +35.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 26:       +35.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 27:       +35.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 28:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 29:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 30:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
> Core 31:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
> 
> nvme-pci-0100
> Adapter: PCI adapter
> Composite:    +38.9°C  (low  = -273.1°C, high = +84.8°C)
>                        (crit = +84.8°C)
> Sensor 1:     +38.9°C  (low  = -273.1°C, high = +65261.8°C)
> Sensor 2:     +35.9°C  (low  = -273.1°C, high = +65261.8°C)
> 
> nct6798-isa-02a0
> Adapter: ISA adapter
> in0:                   272.00 mV (min =  +0.00 V, max =  +1.74 V)
> in1:                     1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in2:                     3.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in3:                     3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in4:                   992.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
> in5:                   136.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
> in6:                   120.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
> in7:                     3.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in8:                     3.12 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in9:                   512.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
> in10:                   72.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
> in11:                   56.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
> in12:                    1.11 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> in13:                  144.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
> in14:                    1.43 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
> fan1:                     0 RPM  (min =    0 RPM)
> fan2:                  1090 RPM  (min =    0 RPM)
> fan3:                     0 RPM  (min =    0 RPM)
> fan4:                     0 RPM  (min =    0 RPM)
> fan5:                     0 RPM  (min =    0 RPM)
> fan7:                   680 RPM  (min =    0 RPM)
> SYSTIN:                +120.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
> CPUTIN:                 +36.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
> AUXTIN0:               +112.0°C    sensor = thermistor
> AUXTIN1:               +116.0°C    sensor = thermistor
> AUXTIN2:               +117.0°C    sensor = thermistor
> AUXTIN3:                 +3.0°C    sensor = thermistor
> PECI Agent 0:           +36.0°C  (high = +80.0°C, hyst = +75.0°C)
>                                  (crit = +100.0°C)
> PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
> PCH_CHIP_TEMP:           +0.0°C  
> PCH_CPU_TEMP:            +0.0°C  
> intrusion0:            ALARM
> intrusion1:            ALARM
> beep_enable:           disabled
> 
> nvme-pci-0300
> Adapter: PCI adapter
> Composite:    +37.9°C  (low  = -273.1°C, high = +84.8°C)
>                        (crit = +84.8°C)
> Sensor 1:     +37.9°C  (low  = -273.1°C, high = +65261.8°C)
> Sensor 2:     +34.9°C  (low  = -273.1°C, high = +65261.8°C)
> ```
> 
> how would i fix the order to represent core 0-19 and fix htop ?

Same for me although I have a 12900k. Curious how you got temps from sensors for the nct6798d chip. Are you using the lax kernel parameter?

kub3let, 2022-12-27:
@mikeyo I did not have to use lax with this build, simply running

```
sensors-detect --auto-tune
```

and then loading the kernel modules that sensors-detect told you to
```
modprobe nct6775
echo nct6775 >> /etc/modules
```

This can differ by motherboard afaik e.g. you might have another chip or need extra kernel parameters

mikeyo, 2022-12-28:
> @mikeyo I did not have to use lax with this build, simply running
>
> ```
> sensors-detect --auto-tune
> ```
>
> and then loading the kernel modules that sensors-detect told you to
>
> ```
> modprobe nct6775
> echo [...]

OK, doesnt work. What motherboard do you have? I still need to use the lax param for my Z690 ProArt/12900K

kub3let, 2022-12-28:
@mikeyo It's a Asrock Z690M-ITX/ax

After reading some more issues, it seems this has already been posted.

https://github.com/lm-sensors/lm-sensors/issues/376
https://github.com/lm-sensors/lm-sensors/issues/388
etc.

I will try to manually map the labels and see if that fixes htop.

Btop shows temps correctly but does not use lm-sensors afaik.

mikeyo, 2022-12-28:
> @mikeyo It's a Asrock Z690M-ITX/ax
> 
> After reading some more issues, it seems this has already been posted.
> 
> #376 #388 etc.
> 
> I will try to manually map the labels and see if that fixes htop.
> 
> Btop shows temps correctly but does not use lm-sensors afaik.

Thanks. Can you do me a fave and run it2cdetect -l and i2cdetect -y 0 assuming 0 is your smbus and post me the output?
I want to know if I can read DIMM temps using 
echo jc42 0x51 > /sys/bus/i2c/devices/i2c-0/new_device
echo jc42 0x53 > /sys/bus/i2c/devices/i2c-0/new_device

Temps show as 51.1 degrees for both modules and never change so I'm assuming jc42 is not reporting back correctly via sensors

 Mine is as follows----
i2c-3   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-10  i2c             i915 gmbus tc4                          I2C adapter
i2c-1   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-8   i2c             i915 gmbus tc2                          I2C adapter
i2c-6   i2c             i915 gmbus dpc                          I2C adapter
i2c-13  i2c             AUX USBC3/DDI TC3/PHY D                 I2C adapter
i2c-4   i2c             i915 gmbus dpa                          I2C adapter
i2c-11  i2c             i915 gmbus tc5                          I2C adapter
i2c-2   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-0   smbus           SMBus I801 adapter at efa0              SMBus adapter
i2c-9   i2c             i915 gmbus tc3                          I2C adapter
i2c-7   i2c             i915 gmbus tc1                          I2C adapter
i2c-5   i2c             i915 gmbus dpb                          I2C adapter
i2c-12  i2c             i915 gmbus tc6                          I2C adapter

i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- 49 -- 4b -- -- -- -- 
50: -- 51 -- 53 -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

kub3let, 2022-12-28:
No that's not working, I doubt the consumer ram that I got has temp sensors exposed, meminfo did not show temps as well.

I would expect this to require ecc memory and a W680 chip.

---
I changed the labels and htop now shows the temps correctly, however yet another problem since sensors only gives temps for the real 14 cores, but htop shows threads as well but there are no labels for them and they can't be duplicated in sensors.

```
cat << 'EOF' > /etc/sensors.d/ASRock_Z690M
chip "coretemp-isa-0000"
  label temp2 "Core 0"
  label temp3 "Core 1"
  label temp4 "Core 2"
  label temp5 "Core 3"
  label temp6 "Core 4"
  label temp7 "Core 5"
  label temp8 "Core 6"
  label temp9 "Core 7"
  label temp10 "Core 8"
  label temp11 "Core 9"
  label temp12 "Core 10"
  label temp13 "Core 11"
  label temp14 "Core 12"
  label temp15 "Core 13"
EOF

systemctl restart lm-sensors
```

[image]

Guess that's all you can do.

I also noticed that even /proc/cpuinfo prints wrong core id, so this might be reported wrongly by the cpu itself.

mikeyo, 2022-12-28:
> No that's not working, I doubt the consumer ram that I got has temp sensors exposed, meminfo did not show temps as well.
>
> I would expect this to require ecc memory and a W680 chip.
>
> I changed the [...]

If the mem temps are exposed in Windows Hwinfo/Aida64 then you should be able to read them in Linux?

Have you tited i7z to see if your cores/temps are reporting back correctly. Here is my output from my 12900k
Stock Intel speeds / Asus enhancements off and governer set as 'powersave'

Cpu speed from cpuinfo 3186.00Mhz
cpuinfo might be wrong if cpufreq is enabled. To guess correctly try estimating via tsc
Linux's inbuilt cpu_khz code emulated now
True Frequency (without accounting Turbo) 3186 MHz
  CPU Multiplier 32x || Bus clock frequency (BCLK) 99.56 MHz

Socket [0] - [physical cores=16, logical cores=24, max online cores ever=16]
  TURBO ENABLED on 16 Cores, Hyper Threading ON
  Max Frequency without considering Turbo 3285.56 MHz (99.56 x [33])
  Max TURBO Multiplier (if Enabled) with 1/2/3/4/5/6 Cores is  52x/51x/50x/50x/49x/49x
  Real Current Frequency 904.83 MHz [99.56 x 9.09] (Max of below)
        Core [core-id]  :Actual Freq (Mult.)      C0%   Halt(C1)%  C3 %   C6 %  Temp      VCore
        Core 1 [0]:       801.97 (8.05x)           1    99.9       0       0    22      0.7000
        Core 2 [2]:       800.36 (8.04x)           1    99.9       0       0    19      0.7000
        Core 3 [4]:       798.79 (8.02x)           1    99.8       0       0    21      0.7000
        Core 4 [6]:       795.88 (7.99x)           1     100       0       0    18      0.7000
        Core 5 [8]:       904.83 (9.09x)        3.63      98       0       1    21      0.6998
        Core 6 [10]:      859.69 (8.63x)        5.09    97.6       0       1    20      0.6998
        Core 7 [12]:      834.69 (8.38x)        3.59    98.1       0       1    22      0.6998
        Core 8 [14]:      810.78 (8.14x)           1    99.8       0       0    18      0.6998
        Core 9 [16]:      830.96 (8.35x)           1    1.64       0    98.3    21      0.6998
        Core 10 [17]:     795.40 (7.99x)           1    0.484      0    99.5    21      0.6998
        Core 11 [18]:     801.41 (8.05x)           1    0.457      0    99.5    21      0.6998
        Core 12 [19]:     797.51 (8.01x)           1    0.456      0    99.5    21      0.7000
        Core 13 [20]:     798.23 (8.02x)           1    0.397      0    99.6    20      0.7000
        Core 14 [21]:     796.99 (8.00x)           1    0.427      0    99.6    20      0.7000
        Core 15 [22]:     790.40 (7.94x)           1    0.476      0    99.5    20      0.7000
        Core 16 [23]:     794.70 (7.98x)           1    0.47       0    99.5    20      0.7000

kub3let, 2022-12-28:
Sorry I'm not planning on testing with windows, I will redo the cable management tomorrow and then reinstall for production.

i7z reports

[image]

It's on powersave using intel_pstate and idles at 22 watts from the wall, currently running proxmox on a raid1 m.2 array.

With a ambient temperature of 22°C it idles at 32°C and 58°C at full load (`stress -c 24` for 3 minutes) using a [NH-L12S](https://noctua.at/en/nh-l12s)

Did you undervolt ?

mikeyo, 2022-12-28:
> Sorry I'm not planning on testing with windows, I will redo the cable management tomorrow and then reinstall for production.
>
> i7z reports
>
> [image]
>
> It's on powersave using intel_pstate and idles at [...]

Nope, just turned off the Asus MCE in BIOS and left Intel stock values/settings alone. Also ran stress -c 24 for 3 minutes with Thermalright AXP120-X67 http://www.thermalright.com/product/axp120-x67/

(celcius)
Ambient = 20 
CPU idle = 21
stress -c 24 for 3mins = 99 nearly klilled my CPU lol. Freq hit 4.9ghz so going to dial down or disable Turbo.

kub3let, 2022-12-28:
You can set a power limit (PL1/PL2)

Usually it should just thermal throttle and be fine.

I also reduced PL2 to 15 seconds, so it down clocks faster, at PL2 it actually hits 75°C.

PL1 is at 4.8GHz and PL2 at 5.1GHz for me.

closed 2022-12-28 by kub3let