lm-sensors/lm-sensors #425: gpu_fan not detected in first boot

biji, 2023-01-24:
Wondering why `gpu_fan` is not detected in first boot

To reproduce:

1. Boot laptop from off
2. sensors
```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +51.0°C  (high = +100.0°C, crit = +100.0°C)

nvme-pci-e100
Adapter: PCI adapter
Composite:    +37.9°C  (low  =  -0.1°C, high = +76.8°C)
                       (crit = +79.8°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          11.98 V  

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +42.0°C  

asus-isa-0000
Adapter: ISA adapter
cpu_fan:     3600 RPM

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +52.0°C  (crit = +103.0°C)

```
3. reboot laptop , don't power off
4. sensors: 
```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +53.0°C  (high = +100.0°C, crit = +100.0°C)

nvme-pci-e100
Adapter: PCI adapter
Composite:    +36.9°C  (low  =  -0.1°C, high = +76.8°C)
                       (crit = +79.8°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          11.98 V  

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +41.0°C  

asus-isa-0000
Adapter: ISA adapter
cpu_fan:     3600 RPM
gpu_fan:     3500 RPM

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +58.0°C  (crit = +103.0°C)

```

Thanks

Wer-Wolf, 2023-01-25:
The driver providing fan control on your machine is `asus-wmi`, not `nvidia`. Since the driver relies on the system BIOS for interfacing with the sensors, maybe the problem can be solved by doing a BIOS update?

Also, whats your kernel version? If the BIOS update did not solve your problem, when maybe you can try to reload the kernel module and check if it detects the second fan.

Since this seems to be a driver problem, i would suggest to file a bugreport on [bugzilla](bugzilla.kernel.org) so the driver maintainers can  fix this issue.

biji, 2023-01-27:
- BIOS updated to latest
- reloading kernel module and its dependencies

Still give same result, only `cpu_fan`

Closing, because this is kernel driver issue

closed 2023-01-27 by biji