lm-sensors/lm-sensors #426: Can't control fans on Asus b550 f Gaming

wiiznokes, 2023-02-01:
lm-sensors recognizes all my fans, but I can't control them.

with `pwmconfig` :
```

Found the following fan sensors:
   hwmon6/fan1_input     current speed: 0 ... skipping!
   hwmon6/fan2_input     current speed: 2119 RPM
   hwmon6/fan3_input     current speed: 1039 RPM
   hwmon6/fan4_input     current speed: 1042 RPM
   hwmon6/fan5_input     current speed: 0 ... skipping!
   hwmon6/fan6_input     current speed: 0 ... skipping!
   hwmon6/fan7_input     current speed: 0 ... skipping!

```
The 3 fans match well.

```
Testing pwm control hwmon6/pwm5 ...
  hwmon6/fan2_input ... speed was 2119 now 2146
    no correlation
  hwmon6/fan3_input ... speed was 1039 now 1046
    no correlation
  hwmon6/fan4_input ... speed was 1042 now 1055
    no correlation
```
But it can't control speed.

I tried to load the `nct6775` module, as descibe [here](https://wiki.archlinux.org/title/lm_sensors#K10Temp_module), but nothing change.

I'm running Manjaro with the `6.1.7-1-MANJARO` kernel.

Wer-Wolf, 2023-02-03:
Can you show me which files are inside the`/sys/class/hwmon/hwmon6/` directory?
Also could you check if you can manually control the fans by writing to the corresponding sysfs files?

wiiznokes, 2023-02-20:
Hi, sorry for the wait, here is the list of files inside `/sys/class/hwmon/hwmon6/`: [hwmon6.txt](https://github.com/lm-sensors/lm-sensors/files/10786153/hwmon6.txt).
Also, I'm not sure about the command here, it will be something like `cat in3_max > fan3_input` ?

Wer-Wolf, 2023-02-20:
`echo <value between 0 - 255> > /sys/class/hwmon/hwmon6/pwmX`

Wer-Wolf, 2023-02-20:
With root privileges.

Wer-Wolf, 2023-02-20:
Can you tell me the name of your hwmon chip?
Its inside the `name` attribute.

wiiznokes, 2023-02-25:
Here is the name of my hwmon chip: `nct6798`
I made `sudo chown pwmX_enable`, idk if there is a better way, but just sudo didn't work.
So `echo 1 > /sys/class/hwmon/hwmon6/pwmX_enable`
then `echo 255 > /sys/class/hwmon/hwmon6/pwm3`
and then `echo 0 > /sys/class/hwmon/hwmon6/pwm3`.

But I did not see any change, however, it should be a fan of my case

Wer-Wolf, 2023-02-25:
First, you have to switch to superuser using `sudo su` (using only `sudo echo ...` is not enough).
Then you have execute `echo 1 > /sys/class/hwmon/hwmon6/pwmX_enable` if you want to be able to control
`echo 1 > /sys/class/hwmon/hwmon6/pwmX`, where `X` needs to be replaced with the number of the fan you want to control.
Looking at the [driver documentation](https://www.kernel.org/doc/html/next/hwmon/nct6775.html), it seems that you should try all fans in the range between 1 and 7.

Wer-Wolf, 2023-02-25:
And which kernel version are you using (output of `uname -r`)?

wiiznokes, 2023-05-02:
I changed distribution, I moved to Fedora 38. My kernel is `6.2.13-300.fc38.x86_64` now. I manage to control the fans with the new instructions, thank you!

Also, is there a way to control the fans of an Nvidia card with lm-sensor?

Wer-Wolf, 2023-05-03:
Not with he propietary driver i think.

wiiznokes, 2023-05-05:
Just a question, I'm trying to make software to control fan speeds. Only, I would like the logic to be calculated in my program, and fan speed to be set every second by my program, for example. Is there an easy way to do this using C or do I have to write in hwmon6 myself ? And will it cause performance issues to do that ?

Wer-Wolf, 2023-05-05:
I would use libsensors for that. Writing or reading  pwm attributes however is currently not supported, you have to do this yourself.

Wer-Wolf, 2023-05-05:
Regarding the performance: some hwmon chips are quite slow, however it should normally not cause any issues.

wiiznokes, 2023-05-09:
Too bad it's not merge #363 :smile:

wiiznokes, 2023-05-09:
Do you think I could use this without troubles ?

Wer-Wolf, 2023-05-09:
I think so, i tested the patches on my machine.

wiiznokes, 2023-05-09:
Hi, sorry to bother you again, but I can't use your patch.
I'm using the static library `libsensors.a`, but when I try to link it I get an error.
```
gcc -g -Wall -I. sensors_wrapper.o -o lsensorswrapper.so libsensors.a 
/usr/bin/ld : /usr/lib/gcc/x86_64-redhat-linux/13/../../../../lib64/crt1.o : dans la fonction « _start » :
(.text+0x1b) : référence indéfinie vers « main »
/usr/bin/ld : libsensors.a(access.ao) : dans la fonction « sensors_eval_expr » :
access.c:(.text+0x50b) : référence indéfinie vers « exp »
/usr/bin/ld : access.c:(.text+0x5ba) : référence indéfinie vers « log »
collect2: erreur: ld a retourné le statut de sortie 1
make: *** [Makefile:13: debug] Error 1
```
Looks like some `mathcalls.h` functions are not included and another problem with `crt1.o`

I know it's not related to the patch but if you have an idea to fix it, I don't know what to do.

Wer-Wolf, 2023-05-10:
The linker cant find your main funtion, check the function prototype. Regarding the issues with mathcalls: try the `-lm` option.

Wer-Wolf, 2023-05-10:
You can also use the sysfs interface directly, see [here](https://www.kernel.org/doc/Documentation/hwmon/sysfs-interface) for a description of the hwmon sysfs interface.

Wer-Wolf, 2023-05-10:
See [here](https://askubuntu.com/questions/332884/how-to-compile-a-c-program-that-uses-math-h) for the `-lm` option.

wiiznokes, 2023-05-11:
I finally only used `libsensors.so.5.0.0`. Mixing .a and .so was to much pain with C#.
You're patch works great on my machine though.

Now, I have the same problem as at the beginning, my program is not able to use `sensors_set_value` when it is run as a service (return code 4, I think it doesn't have enough privilege). Yet I put `User=root` in my `.service` file.

Anyway, thanks for your help man.

closed 2023-05-18 by wiiznokes