lm-sensors/lm-sensors #427: Weird nvme composite reading debian

gaccountg, 2023-02-11:
Hi,
I believe I'm getting weird readings from the command "sensors" for the nvme Composite output.
In this section, I have 2 sensors staying between 31°C and 39°C, plus a "Composite" that comes first.
The composite is usually equal to the value of the first sensor, except that it makes sudden jumps to 83°C
(raising an "ALARM" in the output). Then it comes back to 31°C, most often within less than couple of seconds.

I think this might be a problem with some wrong number type or wrong return value from the sensor, because the low and critical values
are wrong (absolute 0 for the low value, and max. integer value converted from Kelvin to °C for the high value!!). As I understand that composite is the result of some computation (averaging) between some data, maybe there could be  sort of issue here? All the more that the two sensors do not make such jumps, and stay in the 30-40°C range.

Does anyone have a similar experience? Thanks!

Hardware: motherboard ASRock B560M-HDV and NVMe M.2 Samsung SSD980.

Wer-Wolf, 2023-02-12:
The limit attributes are sometimes not initialized by the SSD controller, which causes such unusual  limit values. They are usually harmless.

Did you make sure that your SSD runs the latest firmware? And is the temperature jump always the same?

gaccountg, 2023-02-13:
Thanks for answering!

It seems indeed to always be the same value (83.something), although the jump did not occur in the last 2 tests I ran (which is weird as I used to always see it at least once), so I'm not sure. What is sure is that I never saw a value in-between, like in the range 45-79 say. It's just a direct jump in the 'critical zone' (between 83.something and 84.8) and then back to something in the 29-39 range.

What I learn from the tests is that the value of sensor 2 seems consistent with the drive activity: when doing some quick reading it gradually went up to 64°, then gradually back to 39° in 3/4 seconds after ending the reading: I believe this would be usual temperature?

As the motherboard has 2 slots, I believe sensor 1 is for the other, currently empty, slot. Could I check this by changing the slot on which the ssd is plugged? (I mean, do you by chance know whether it might confuse grub (partition is encrypted)?)

As for the firmware, I don't think I can use fwupdmgr, but maybe I can do it manually using files from the constructor's website?

Wer-Wolf, 2023-02-13:
Both sensors belong to the same SSD, so changing the slot should change nothing.

Which SSD model do you have?

gaccountg, 2023-02-13:
Ah, sorry I thought they might be on the motherboard. It's an NVMe M.2 Samsung SSD980.

Wer-Wolf, 2023-02-13:
Seems to be a [firmware issue](https://forum.proxmox.com/threads/smart-error-health-detected-on-host.109580/#post-475308).

AFAIK, samsung provides bootable iso images for firmware updates.

gaccountg, 2023-02-13:
Thanks a lot! I will try it soon and report here; it indeed seems to be the exact problem I have (yes I did not find this page).
Thanks also for the back-up warning (my back-up is usually quite up-to-date (rsync rules...)).

gaccountg, 2023-02-14:
Hello! I updated the firmware. I followed the method in [this link](https://blog.quindorian.org/2021/05/firmware-update-samsung-ssd-in-linux.html/), which I found on the webpage you had provided. Everything went smoothly, and no data has been lost. 
I will watch what happens but I'm confident I won't have these jumps anymore. I will close the issue in a few days to be sure things are cleared.

As a side note, since the drive was my boot drive, and it seems you need to unmount it prior to updating the firmware, I simply booted on the usb of my distrib (like for a fresh install; you could use that of any distrib) and followed the steps from this live system. It worked like a charm. (Bootable env rom Samsung seem to have been broken for years.)
A 'sudo smartctl -a /dev/your-drive-here' after shutting down & turning on again confirmed my system is now running the latest firmware.

Thanks again for your help!

gaccountg, 2023-02-24:
All is fine with the above, no weird reading anymore. Thanks a lot, issue solved.

closed 2023-02-24 by gaccountg