lm-sensors/lm-sensors #429: Adding support for Coral Edge TPU

mattncsu, 2023-02-18:
Hi, how would I go about adding the temperature from a dual edge tpu (https://coral.ai/products/m2-accelerator-dual-edgetpu) into sensors?

Temperature is available in millidegrees C at /sys/class/apex/apex_0/temp  where 0 could be 0 to the number of TPUs installed (a dual would have an apex_0 and apex_1, a pair of dual TPUs would have apex_0 through apex_3.

```
:cat /sys/class/apex/apex_0/temp
62300
```
for 62.3C

Thank you!

Wer-Wolf, 2023-02-19:
I suppose the driver for your TPU is not mainlined? In such a case, you have to ask the manufacturer of your TPU to add hwmon support to the driver, so sensors can find it.

Wer-Wolf, 2023-02-19:
I checked again, and your driver seems to be inside the mainline kernel, albeit as a staging (read beta) driver.

You should reach out to your manufacturer and ask them to turn this driver into a proper accel driver with hwmon support.

Wer-Wolf, 2023-02-19:
You can of course try to finish the kernel driver yourself, if your willing to start with linux kernel development.

mattncsu, 2023-02-19:
@Wer-Wolf Thanks for the feedback! No one wants me touching the linux kernel, my programming skills cap out at Arduino sketches and some basic python.

Wer-Wolf, 2023-02-19:
Thats ok, ask the manufacturer then.