lm-sensors/lm-sensors #430: Add support for `drm` class

ubunatic, 2023-02-22:
Is it possible to add support for `drm` sensors?
On my system (iGPU: Intel UHD 530) the GPU data seems to appear only in the generic `drm` class, and not on other busses.

```
for f in /sys/class/drm/card0/gt_*; do echo -n "$f: "; cat "$f"; done
/sys/class/drm/card0/gt_act_freq_mhz: 350
/sys/class/drm/card0/gt_boost_freq_mhz: 1100
/sys/class/drm/card0/gt_cur_freq_mhz: 483
/sys/class/drm/card0/gt_max_freq_mhz: 1100
/sys/class/drm/card0/gt_min_freq_mhz: 350
/sys/class/drm/card0/gt_RP0_freq_mhz: 1100
/sys/class/drm/card0/gt_RP1_freq_mhz: 350
/sys/class/drm/card0/gt_RPn_freq_mhz: 350
```

```
for f in /sys/class/drm/card0/gt/gt0/*; do echo -n "$f: "; cat "$f"; done
/sys/class/drm/card0/gt/gt0/id: 0
/sys/class/drm/card0/gt/gt0/punit_req_freq_mhz: 350
/sys/class/drm/card0/gt/gt0/rc6_enable: 1
/sys/class/drm/card0/gt/gt0/rc6_residency_ms: 26396254
/sys/class/drm/card0/gt/gt0/rps_act_freq_mhz: 350
/sys/class/drm/card0/gt/gt0/rps_boost_freq_mhz: 1100
/sys/class/drm/card0/gt/gt0/rps_cur_freq_mhz: 367
/sys/class/drm/card0/gt/gt0/rps_max_freq_mhz: 1100
/sys/class/drm/card0/gt/gt0/rps_min_freq_mhz: 350
/sys/class/drm/card0/gt/gt0/rps_RP0_freq_mhz: 1100
/sys/class/drm/card0/gt/gt0/rps_RP1_freq_mhz: 350
/sys/class/drm/card0/gt/gt0/rps_RPn_freq_mhz: 350
```

My system:
* 6.1.12-1-MANJARO x86_64
* Intel® Core™ i5-6500T, Mesa Intel® HD Graphics 530 (SKL GT2)

Wer-Wolf, 2023-02-26:
AFAIK, lm-sensors deals with sensors, not with frequency data. There where already some attempts to add frequency support to lm-sensors, but they where all not accepted upstream.

ubunatic, 2023-02-26:
Too bad. 😿  As a user, I am exposed to sensors readings in tools such as Freon:

[image]

So now, I'd rather have to ask each of the dozens of sensor GUIs to add support for GPU load metrics. Having these values exposed via an accepted standard would be much better for all these downstream components. For me it totally makes sense to add these core system gauges. 🤷🏼 

But thanks for letting me know.

I guess a good "sensors" GUI should have support for file based metrics anyway.

closed 2023-02-26 by ubunatic