lm-sensors/lm-sensors #432: sensors-detect killed NVMe SSD

supermaz, 2023-03-05:
Hi,

I want to report that sensors-detect just killed my brand new 1TB ADATA NVMe PCIe4 SSD.
I use gentoo linux, booted a Manjaro Live-CD, and during installation of Gentoo on an ext4 root FS, where you need to compile everything, I wanted to take a look at temperature values (its a complete new system, so I was curious). After starting sensors-detect from the live-cd during installation, I got strange errors during compile and then noticed via kernel messages that suddenly the ext4 was broken. No information about hardware failures etc, just that errors from ext4 (No space for directory leaf checksum). I haven't seen something like this in 15 years of linux usage.
OK, so I rebooted, started install again, this time took XFS as filesystem. Everything went smoothly, finally rebooted into Gentoo, installed a lot, then some final tweaks. So, I ran sensors-detect again. The first time I let it scan IPMI but that didn't even finished, was hanging. I restarted withtout IPMI etc., see code further down for details. After the last run my filesystem was completely gone... I rebooted and the BIOS took quite a long time to come up and the SSD was not found. From this on, every time the SSD is in, the boot takes very long without finding the SSD, other NVMe SSD works fine. So, this completely new SSD seems to be killed by sensors-detect.

I don't want to complain but to warn other users that this might happen.

Here is my run of sensors-detect (did it via SSH so I still have it):

```
workstation /usr/src/linux # sensors-detect 
# sensors-detect version 3.6.0
# System: ASRock B660M Pro RS
# Kernel: 6.2.2-gentoo x86_64
# Processor: 13th Gen Intel(R) Core(TM) i5-13500 (6/191/2)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Gentoo config file protection is active. Every file this program will
modify must be merged before the change will become active using
default Gentoo tools such as dispatch-conf, cfg-update, and etc-update.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT6798D Super IO Sensors'                   Success!
    (address 0x2a0, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x0101

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
^C
workstation /usr/src/linux # sensors-detect 
# sensors-detect version 3.6.0
[... 186 lines cut ...]

Driver `nct6775':
  * ISA bus, address 0x2a0
    Chip `Nuvoton NCT6798D Super IO Sensors' (confidence: 9)

Do you want to generate '/etc/modules-load.d/lm_sensors.conf'? (YES/no): 

 * IMPORTANT: At least 1 config file(s) in '/etc' need updating.
 * See the CONFIGURATION FILES and CONFIGURATION FILES UPDATE TOOLS
 * sections of the emerge man page to learn how to update config files.

Note: You only have created the required configuration to autoload the
required modules to use your sensors on boot. If you want to start using
your sensors right now you have to either load them now once manually, to
reboot this system or to execute the following command(s) _AFTER_
you have updated your config files:

  # systemctl restart systemd-modules-load

workstation /usr/src/linux # 
workstation /usr/src/linux # ll
-bash: /bin/ls: Input/output error
```

I don't think that there is much I can do about this, so I need to replace and probably will not run sensors-detect again. Or maybe you have an idea which function might have caused this problem. Let me know if you need more information.

Wer-Wolf, 2023-03-11:
I think NVMe SSD sensors are _not_ scaned for by sensors-detect, so i belive your SSD issues are caused by something else.

Indeed, the symptoms shown by your SSD (slow boot, data loss, ...) sound exactly like the SSD firmware was somehow corrupted.

AFAIK, sensors-detect cannot cause this, as it does not access the NVMe controller. Did you issued a secure erase? Because i know of some cases where a secure erase caused the SSD firmware to lock up.

Can you poste more system information please?

supermaz, 2023-03-12:
Thanks for the reply. I can only guess that it has something to do with these arbitrary writes from the logs.
And I'm pretty sure that it has something to do with sensors-detect as the system was running well and compiling under full load for nearly a day and the SSD broke in the same moment I used sensors detect in the middle of the installation. I also did never used secure erase as it was a brand new device.
There was nothing else running communicating with the hardware directly.
Again, I had SSD issues twice after sensors-detect, first time I think I did not run every test but I just broke the filesystem. Second time the SSD was completely lost (this time I copied the logs above).

It might be bad luck and a defective device from the start but the timely coincident would be really surprising (heavy load on the system for longer time).

System info:
- ASRock Mainboard with B660 chipset
- Intel 13500 CPU
- 2x16GB RAM
- Radeon 6600XT
- 1TB ADATA NVMe PCIe4 SSD.

First problem was with sensors-detect from current Manjaro Live-CD during installation via chroot (but started from the Live-CD FS) without overwriting default YES/NO answers.
Second time was already from the installed and running Gentoo system, logs above.

Wer-Wolf, 2023-03-12:
According to the sensors-detect logs, the utility was unable to load the kernel module i2c-dev, so most of the probing was not carried out.

After the IPMI hang, did the whole system freeze/reported any errors during reboot?

Wer-Wolf, 2023-03-12:
I did some research on the ext4 error message, and it seems to me that this message is usually harmless.

Since in both cases the SSD did malfunction after a period of high usage (compile/setup), could it be that it maybe overheated?

Wer-Wolf, 2023-03-12:
Forget my comment about your SSD overheating, it indeed seems that your SSD was faulty.

Because under gentoo, you where able to run sensors-detect muliple times. If sensors-detect indeed destroyed your SSD, then you would not have been able to do that.

supermaz, 2023-03-12:
There were no reboots between the runs of sensor-detect, indeed these 2 times of sensors-detect where done just immediately after another. The first time it hanged on IPMI, then CTRL-C and start again. This time it got threw to the end but after this the SSD was dead, you can see the error message at the end that it can't find the ls command anymore.

Wer-Wolf, 2023-03-12:
Can you provide me with the content of `/proc/ioports` and `/proc/iomem`? The only way sensors-detect could have destroyed your SSD is by somehow interacting with it over PCIe, but even this should be impossible.

supermaz, 2023-03-12:
Sure:

```
workstation ~ $ cat /proc/ioports 
0000-0000 : PCI Bus 0000:00
  0000-0000 : dma1
  0000-0000 : pic1
  0000-0000 : timer0
  0000-0000 : timer1
  0000-0000 : keyboard
  0000-0000 : keyboard
  0000-0000 : rtc_cmos
    0000-0000 : rtc0
  0000-0000 : dma page reg
  0000-0000 : pic2
  0000-0000 : dma2
  0000-0000 : fpu
  0000-0000 : pnp 00:00
  0000-0000 : pnp 00:00
  0000-0000 : pnp 00:00
  0000-0000 : pnp 00:00
  0000-0000 : serial
  0000-0000 : iTCO_wdt
  0000-0000 : pnp 00:01
0000-0000 : PCI conf1
0000-0000 : PCI Bus 0000:00
  0000-0000 : pnp 00:01
  0000-0000 : ACPI PM1a_EVT_BLK
  0000-0000 : ACPI PM1a_CNT_BLK
  0000-0000 : ACPI PM_TMR
  0000-0000 : ACPI PM2_CNT_BLK
  0000-0000 : pnp 00:02
  0000-0000 : ACPI GPE0_BLK
  0000-0000 : pnp 00:04
  0000-0000 : PCI Bus 0000:01
    0000-0000 : PCI Bus 0000:02
      0000-0000 : PCI Bus 0000:03
        0000-0000 : 0000:03:00.0
  0000-0000 : 0000:00:17.0
    0000-0000 : ahci
  0000-0000 : 0000:00:17.0
    0000-0000 : ahci
  0000-0000 : 0000:00:17.0
    0000-0000 : ahci
  0000-0000 : 0000:00:1f.4
    0000-0000 : i801_smbus
```

And 

```
workstation ~ $ cat /proc/iomem 
00000000-00000000 : Reserved
00000000-00000000 : System RAM
00000000-00000000 : Reserved
00000000-00000000 : System RAM
00000000-00000000 : Reserved
  00000000-00000000 : PCI Bus 0000:00
  00000000-00000000 : PCI Bus 0000:00
  00000000-00000000 : PCI Bus 0000:00
  00000000-00000000 : PCI Bus 0000:00
  00000000-00000000 : PCI Bus 0000:00
  00000000-00000000 : System ROM
00000000-00000000 : System RAM
00000000-00000000 : System RAM
00000000-00000000 : System RAM
00000000-00000000 : System RAM
00000000-00000000 : System RAM
00000000-00000000 : Reserved
00000000-00000000 : System RAM
00000000-00000000 : Reserved
00000000-00000000 : System RAM
00000000-00000000 : Reserved
00000000-00000000 : ACPI Tables
00000000-00000000 : ACPI Non-volatile Storage
00000000-00000000 : Reserved
00000000-00000000 : System RAM
00000000-00000000 : Reserved
00000000-00000000 : Reserved
00000000-00000000 : Reserved
00000000-00000000 : PCI Bus 0000:00
  00000000-00000000 : PCI Bus 0000:05
    00000000-00000000 : 0000:05:00.0
      00000000-00000000 : atlantic_mmio
    00000000-00000000 : 0000:05:00.0
    00000000-00000000 : 0000:05:00.0
      00000000-00000000 : atlantic_mmio
    00000000-00000000 : 0000:05:00.0
      00000000-00000000 : atlantic_mmio
  00000000-00000000 : PCI Bus 0000:01
    00000000-00000000 : PCI Bus 0000:02
      00000000-00000000 : PCI Bus 0000:03
        00000000-00000000 : 0000:03:00.0
        00000000-00000000 : 0000:03:00.0
        00000000-00000000 : 0000:03:00.1
          00000000-00000000 : ICH HD audio
    00000000-00000000 : 0000:01:00.0
  00000000-00000000 : PCI Bus 0000:04
    00000000-00000000 : 0000:04:00.0
      00000000-00000000 : nvme
  00000000-00000000 : 0000:00:1f.6
    00000000-00000000 : e1000e
  00000000-00000000 : 0000:00:17.0
    00000000-00000000 : ahci
  00000000-00000000 : 0000:00:17.0
    00000000-00000000 : ahci
  00000000-00000000 : 0000:00:17.0
    00000000-00000000 : ahci
  00000000-00000000 : 0000:00:1f.5
00000000-00000000 : PCI MMCONFIG 0000 [bus 00-ff]
  00000000-00000000 : pnp 00:03
00000000-00000000 : Reserved
00000000-00000000 : Reserved
  00000000-00000000 : IOAPIC 0
00000000-00000000 : Reserved
  00000000-00000000 : HPET 0
    00000000-00000000 : PNP0103:00
00000000-00000000 : Reserved
  00000000-00000000 : MSFT0101:00
00000000-00000000 : pnp 00:03
00000000-00000000 : pnp 00:03
00000000-00000000 : pnp 00:03
00000000-00000000 : pnp 00:03
00000000-00000000 : Local APIC
  00000000-00000000 : Reserved
00000000-00000000 : System RAM
  00000000-00000000 : Kernel code
  00000000-00000000 : Kernel rodata
  00000000-00000000 : Kernel data
  00000000-00000000 : Kernel bss
00000000-00000000 : RAM buffer
00000000-00000000 : PCI Bus 0000:00
  00000000-00000000 : PCI Bus 0000:01
    00000000-00000000 : PCI Bus 0000:02
      00000000-00000000 : PCI Bus 0000:03
        00000000-00000000 : 0000:03:00.0
        00000000-00000000 : 0000:03:00.0
  00000000-00000000 : 0000:00:1f.3
  00000000-00000000 : 0000:00:14.0
    00000000-00000000 : xhci-hcd
  00000000-00000000 : 0000:00:1f.3
    00000000-00000000 : ICH HD audio
  00000000-00000000 : 0000:00:14.2
  00000000-00000000 : 0000:00:1f.4
  00000000-00000000 : 0000:00:16.0
  00000000-00000000 : 0000:00:15.0
  00000000-00000000 : 0000:00:14.2
00000000-00000000 : 0000:03:00.0
```

I also thought that it should be impossible but then it would really be a great coincidence.

Wer-Wolf, 2023-03-12:
You need be root, otherwise the adresses are all zero

Wer-Wolf, 2023-03-12:
And maybe also the output of `lspci`?

supermaz, 2023-03-12:
Sorry! Haven't looked too close to the ouput...

```
workstation ~ $ sudo cat /proc/ioports 
0000-0cf7 : PCI Bus 0000:00
  0000-001f : dma1
  0020-0021 : pic1
  0040-0043 : timer0
  0050-0053 : timer1
  0060-0060 : keyboard
  0064-0064 : keyboard
  0070-0071 : rtc_cmos
    0070-0071 : rtc0
  0080-008f : dma page reg
  00a0-00a1 : pic2
  00c0-00df : dma2
  00f0-00ff : fpu
  0280-028f : pnp 00:00
  0290-029f : pnp 00:00
  02a0-02bf : pnp 00:00
  02c0-02df : pnp 00:00
  03f8-03ff : serial
  0400-041f : iTCO_wdt
  0680-069f : pnp 00:01
0cf8-0cff : PCI conf1
0d00-ffff : PCI Bus 0000:00
  164e-164f : pnp 00:01
  1800-1803 : ACPI PM1a_EVT_BLK
  1804-1805 : ACPI PM1a_CNT_BLK
  1808-180b : ACPI PM_TMR
  1850-1850 : ACPI PM2_CNT_BLK
  1854-1857 : pnp 00:02
  1860-187f : ACPI GPE0_BLK
  2000-20fe : pnp 00:04
  3000-3fff : PCI Bus 0000:01
    3000-3fff : PCI Bus 0000:02
      3000-3fff : PCI Bus 0000:03
        3000-30ff : 0000:03:00.0
  4020-403f : 0000:00:17.0
    4020-403f : ahci
  4040-4043 : 0000:00:17.0
    4040-4043 : ahci
  4050-4057 : 0000:00:17.0
    4050-4057 : ahci
  efa0-efbf : 0000:00:1f.4
```
and 

```
workstation ~ $ sudo cat /proc/iomem 
00000000-00000fff : Reserved
00001000-0009dfff : System RAM
0009e000-0009efff : Reserved
0009f000-0009ffff : System RAM
000a0000-000fffff : Reserved
  000a0000-000bffff : PCI Bus 0000:00
  00000000-00000000 : PCI Bus 0000:00
  00000000-00000000 : PCI Bus 0000:00
  00000000-00000000 : PCI Bus 0000:00
  000e0000-000effff : PCI Bus 0000:00
  000f0000-000fffff : System ROM
00100000-59133017 : System RAM
59133018-59142857 : System RAM
59142858-59151017 : System RAM
59151018-5916f057 : System RAM
5916f058-5a595fff : System RAM
5a596000-5a5d6fff : Reserved
5a5d7000-5f303fff : System RAM
5f304000-5f304fff : Reserved
5f305000-61d17fff : System RAM
[... 76 lines cut ...]
```
and

```
00:00.0 Host bridge: Intel Corporation Device 4640 (rev 02)
00:01.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x16 Controller #1 (rev 02)
00:06.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 (rev 02)
00:14.0 USB controller: Intel Corporation Alder Lake-S PCH USB 3.2 Gen 2x2 XHCI Controller (rev 11)
00:14.2 RAM memory: Intel Corporation Alder Lake-S PCH Shared SRAM (rev 11)
00:15.0 Serial bus controller: Intel Corporation Alder Lake-S PCH Serial IO I2C Controller #0 (rev 11)
00:16.0 Communication controller: Intel Corporation Alder Lake-S PCH HECI Controller #1 (rev 11)
00:17.0 SATA controller: Intel Corporation Alder Lake-S PCH SATA Controller [AHCI Mode] (rev 11)
00:1c.0 PCI bridge: Intel Corporation Alder Lake-S PCH PCI Express Root Port #1 (rev 11)
00:1f.0 ISA bridge: Intel Corporation Device 7a86 (rev 11)
00:1f.3 Audio device: Intel Corporation Alder Lake-S HD Audio Controller (rev 11)
00:1f.4 SMBus: Intel Corporation Alder Lake-S PCH SMBus Controller (rev 11)
00:1f.5 Serial bus controller: Intel Corporation Alder Lake-S PCH SPI Controller (rev 11)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (17) I219-V (rev 11)
01:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Upstream Port of PCI Express Switch (rev c1)
02:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Downstream Port of PCI Express Switch
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 23 [Radeon RX 6600/6600 XT/6600M] (rev c1)
03:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21/23 HDMI/DP Audio Controller
04:00.0 Non-Volatile memory controller: Sandisk Corp Device 5017 (rev 01)
05:00.0 Ethernet controller: Aquantia Corp. AQC107 NBase-T/IEEE 802.3bz Ethernet Controller [AQtion] (rev 02)
```

Wer-Wolf, 2023-03-12:
Your nvme ssd is using memory mapped io, while sensors only probes ioports. This means that there could be no way sensors-detect somehow touched the ssd controller.

It really seems to be that your drive was faulty, maybe you could contact AData and ask for a replacement drive, and if similar problems where reported previously.

supermaz, 2023-03-12:
The SSD is already on retour, the one currently in is a new one, this time a WD Black.

Wer-Wolf, 2023-03-12:
Ok. I also wanted to ask you if you enabled the i2c-dev module for your gentoo kernel build, because sensors-detect complained that it could not find it.

supermaz, 2023-03-12:
At the moment I'm not in the mood to run it again ;) At least not until my replacement is there and I can remove the current working SSD before.

Wer-Wolf, 2023-03-12:
I just asked since if you enabled it and the module still could not be found, thenthis might point at data corruption issues prior to running sensors-detect.

youngunix, 2023-03-19:
I'm having perhaps the same issue on my Gentoo machine. When I run the command `watch -n 1 sensors` the NVMe SSD activity spikes up and it seems pretty aggressive. I'm not sure if that's the best way to monitor the temps and fan speeds but it is faster and easier to run. 

Is there a way to run it completely from RAM using tmpfs?

Wer-Wolf, 2023-03-19:
Can you try and check if this also happens when you read the nvme sensor chip under `/sys/class/hwmon/hwmonX` directly? If yes, then maybe you can bkacklist this sensor chip. I would recommend you to check for firmware updates for your SSD.

Wer-Wolf, 2023-04-10:
Any updates on this?

youngunix, 2023-04-11:
I found the device folder but that just took me into a rabbit hole, because the folder `hwmonX` just keeps repeating it self!
Do you know what specific file I'm supposed to read the temps from?