lm-sensors/lm-sensors #438: ASRock X470 M-ITX & Nuvoton NCT5562D

celeroncool, 2023-04-06:
Hello,

I'm having problems reading correct sensor data from my ASRock X470 M-ITX board.

Using `sudo modprobe nct6775` gets me the following voltages & values:
```
nct6792-isa-0290
Adapter: ISA adapter
in0:                   672.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                     0.00 V  (min =  +0.00 V, max =  +0.00 V)
in2:                     3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                     3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                     0.00 V  (min =  +0.00 V, max =  +0.00 V)
in5:                     0.00 V  (min =  +0.00 V, max =  +0.00 V)
in6:                   744.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                     3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                     3.25 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                     1.80 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                  184.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                  136.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                    1.84 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                    1.70 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                    0.00 V  (min =  +0.00 V, max =  +0.00 V)
fan1:                  2571 RPM  (min =    0 RPM)
fan2:                  1175 RPM  (min =    0 RPM)
fan3:                   732 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                +117.0°C  (high =  +0.0°C, hyst =  +0.0°C)  sensor = thermistor
CPUTIN:                 -62.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +40.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
AUXTIN1:               +107.0°C    sensor = thermistor
AUXTIN2:               +106.0°C    sensor = thermistor
AUXTIN3:               -128.0°C    sensor = thermistor
SMBUSMASTER 0:          +29.0°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
TSI0_TEMP:              +29.2°C  
intrusion0:            OK
intrusion1:            ALARM
beep_enable:           disabled
```

What I have figured out so far is:
fan1 = Chassis fan 1/ water pump header
fan2 = CPU fan header
fan3 = Chassis fan 2
fan4&5 = not connected at all
AUXTIN = Maybe motherboard (does not change)
SMBUSMASTER 0 = CPU temp
TSI0_TEMP = CPU temp

AUXTIN1,2,3 = bogus
PCH_CHIP_CPU_MAX_TEMP, PCH_CHIP_TEMP, PCH_CPU_TEMP = bogus

in4,5,14 = bogus, or atleast reports 0V

Where this gets interesting, is that the markings on the chip says that it is Nuvoton NCT5562D, which seems like not even a real chip, atleas I cannot find ANY datasheets for the device.
Closest I can find is NCT5569D...

Any help and guidance on what to do next?

celeroncool, 2023-04-11:
Well, atleast some of the voltages are read, but the scaling etc is all wrong.
I'm getting a Ryzen 7 5700x in the mail today, I can then upgrade my bios, to see if maybe the readings get better with later bios.

Wer-Wolf, 2023-04-16:
Its often the case that the voltage scaling needs to be adjusted, and some voltage channels might need to be masked.

This can be done using sensors.conf