lm-sensors/lm-sensors #441: Asus Tuf Gaming B550-Plus WIFI II

nathan-stewart, 2023-04-23:
I'm trying to figure out the sensors on this motherboard. sensors-detect finds the  nct6798-isa-0290, I've found the fan mapping through plug and play:

label fan2 "CPU FAN"
    set fan2_min 300
label fan7 "CPU_OPT"
label fan6 "AIO"
# Always show zeros
label fan1 "CHA_FAN1"
    set fan1_min 450
label fan3 "CHA_FAN2"
label fan4 "CHA_FAN3"
ignore fan5

The thermistor port is AUXTIN2. But AUXTIN0 and AUXTIN3 report slowly changing values in the upper 80's low 90's, even while BIOS is not ramping the fans. I can't find what these are reporting - perhaps a voltage regulator?  I'm not sure if there's a scaling error, they're totally bogus, or if I should be alarmed. 

Does anyone have a working conf file for this motherboard?

wolcen, 2023-05-10:
My understanding is that these are auxiliary temp inputs that are not normally connected (they float very high in my case as well).

I've been trying (unsuccessfully) to add them an ignore entry. Other things ignore fine, but not "AUXTIN0" nor "auxtin0" or any other random thing I've tried. :shrug:

wolcen, 2023-05-10:
Clarification: they don't all always float high. AUXTIN2 is clearly a little cooler than ambient here:
```
AUXTIN0:                   +77.5°C    sensor = thermistor
AUXTIN1:                   +50.0°C    sensor = thermistor
AUXTIN2:                   -62.0°C    sensor = thermistor
AUXTIN3:                   +74.0°C    sensor = thermistor
```

wolcen, 2023-05-10:
In any case - let's not be surprised if this gets closed unceremoniously since the lm-sensors devs are not the people that would hold the answers we're looking for (ultimately, motherboard vendors should be able to tell us such things, but I won't hold my breath).

I do wish that the ignore would work however. I'm not getting why it would not - it's a direct strcmp between the ignore I put in and the sensor label...makes no sense. The labels for the sensors strings seem normal https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hwmon/nct6775-core.c?h=v6.4-rc1#n619

Idunno...

nathan-stewart, 2023-05-10:
Asus wanted to RMA when I asked.  Regardless, lm-sensors  doesn't have a discussion section, just issues. So I made an issue.  Thanks for that input -hadn't considered they were floating inputs.