lm-sensors/lm-sensors #445: sensors-detect doesn't detect temperature sensor in NETGEAR ReadyNAS Duo v2

LimakMiRose, 2023-05-16:
This seems to be closely related to https://github.com/lm-sensors/lm-sensors/issues/30 - all symptoms are exactly the same but the hardware is different.

It's ARMv5 architecture. Most detailed information about hardware in this device that I've found is here: http://www.natisbad.org/NAS/ and according to the temperature sensor chapter on this site (http://www.natisbad.org/NAS/#hw-temperature-sensor) it ought to work out of box... But it doesn't and my knowledge regarding drivers and modules in kernel is shamelessly low. Therefor I am asking for help or some kind of guidance how to make it work. Final goal is to display CPU temperature in htop - afaik it bases on sensors command.

sensors-detect command output:
```cmd
limak@syrenka-nas:~$ sudo sensors-detect
[sudo] password for limak:
# sensors-detect version 3.6.0
# DMI data unavailable, please consider installing dmidecode 2.7
# or later for better results.
# Kernel: 6.1.0-0.deb11.7-marvell armv5tel
# Processor: Feroceon 88FR131 rev 1 (v5l)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no):
modprobe: FATAL: Module cpuid not found in directory /lib/modules/6.1.0-0.deb11.7-marvell
Failed to load module cpuid.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no):
Sorry, no supported PCI bus adapters found.

Next adapter: mv64xxx_i2c adapter (i2c-0)
Do you want to scan it? (YES/no/selectively):

Sorry, no sensors were detected.
Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
https://hwmon.wiki.kernel.org/device_support_status for driver status.
```

Temperature values available by the driver:
```cmd
limak@syrenka-nas:~$ cat /sys/class/thermal/thermal_zone0/type
kirkwood_thermal
limak@syrenka-nas:~$ cat /sys/class/thermal/thermal_zone0/temp
41100
limak@syrenka-nas:~$ cat /sys/class/thermal/thermal_zone0/temp
41834
```

Configuration in /boot directory:
```cmd
limak@syrenka-nas:~$ grep 'CONFIG_THERMAL=\|CONFIG_HWMON=' /boot/config-*
/boot/config-5.10.0-22-marvell:CONFIG_HWMON=m
/boot/config-5.10.0-22-marvell:CONFIG_THERMAL=y
/boot/config-6.1.0-0.deb11.7-marvell:CONFIG_HWMON=m
/boot/config-6.1.0-0.deb11.7-marvell:CONFIG_THERMAL=y
```

Kernel modules status:
```cmd
limak@syrenka-nas:~$ lsmod
Module                  Size  Used by
binfmt_misc             9001  1
cpufreq_powersave       1276  0
cpufreq_userspace       3149  0
cpufreq_conservative     3829  0
cpufreq_ondemand        5935  0
af_packet              34159  2
ofpart                  4270  0
cmdlinepart             2978  0
evdev                  10701  1
xhci_pci                5062  0
ehci_orion              3573  0
ehci_hcd               52043  1 ehci_orion
xhci_hcd              172518  1 xhci_pci
marvell                18699  1
orion_nand              2297  0
usbcore               169371  4 ehci_orion,ehci_hcd,xhci_pci,xhci_hcd
nand                   90289  1 orion_nand
marvell_cesa           30331  0
sg                     20575  0
mv643xx_eth            26967  0
orion_wdt               6879  0
g762                    7667  0
mvmdio                  4995  0
nandcore               18521  1 nand
usb_common              7772  3 ehci_hcd,usbcore,xhci_hcd
watchdog               20642  1 orion_wdt
libdes                 16319  1 marvell_cesa
nls_base                5931  1 usbcore
bch                     9641  1 nandcore
mtd                    51444  10 nandcore,ofpart,nand,cmdlinepart,orion_nand
gpio_keys               8807  0
hwmon                  15221  1 g762
kirkwood_thermal        2250  0
leds_gpio               3321  0
fuse                   93364  1
configfs               25895  1
sunrpc                366177  1
ip_tables              11265  0
x_tables               19070  1 ip_tables
ipv6                  444656  41
autofs4                27919  2
ext4                  612953  1
crc16                   1425  1 ext4
mbcache                 4571  1 ext4
jbd2                   94904  1 ext4
crc32c_generic          1990  2
sd_mod                 34010  3
t10_pi                  6779  1 sd_mod
crc64_rocksoft          2115  1 t10_pi
crc64                   5188  1 crc64_rocksoft
crc_t10dif              2027  1 t10_pi
crct10dif_generic       1737  1
crct10dif_common        1406  2 crct10dif_generic,crc_t10dif
sata_mv                27446  2
libata                222377  1 sata_mv
scsi_mod              178630  3 sd_mod,libata,sg
scsi_common             3295  3 libata,sg,scsi_mod
rtc_rs5c372             9501  0
```

What am I missing to achieve the goal?

Wer-Wolf, 2023-05-17:
You might also need to enable `THERMAL_HWMON`.

Wer-Wolf, 2023-05-17:
I assume you are using debian, right? In this case, you should file a bugreport against the package `linux-image-marvell` saying that `CONFIG_HWMON` and `CONFIG_THERMAL_HWMON` should both be set to `y`.

LimakMiRose, 2023-05-18:
> I suspect you have to recompile the kernel with `CONFIG_HWMON` set to `y`, otherwise the thermal subsystem containing the thermal sensor entry cannot register it as a hwmon sensor too.

I see. Thanks. Time for some Googling now. I'll come back with results.

> Which linux distribution are you using?

It's Debian 11 (bullseye).

> I assume you are using debian, right? In this case, you should file a bugreport against the package `linux-image-marvell` saying that `CONFIG_HWMON` and `CONFIG_THERMAL_HWMON` should both be set to `y`.

I'll Google how to and then do it but after first I'll confirm the solution on my device.