lm-sensors/lm-sensors #446: Support for MSI MPG Z790 EDGE WIFI (MS-7D91)

mindrunner, 2023-05-22:
Cannot read/control fan speed and temps from my board MSI MPG Z790 EDGE WIFI (MS-7D91).

What might be missing here?

```
❯ sudo dmidecode | grep "Product Name"
	Product Name: MS-7D91
	Product Name: MPG Z790 EDGE WIFI (MS-7D91)
```

```
❯ sudo dmesg | grep nct6683
[   23.519227] nct6683: Found NCT6687D or compatible chip at 0x4e:0xa20
[11213.015369] nct6683: Found NCT6687D or compatible chip at 0x4e:0xa20
[11236.056639] nct6683: Found NCT6687D or compatible chip at 0x4e:0xa20
[11271.279765] nct6683: Found NCT6687D or compatible chip at 0x4e:0xa20
```

```
❯ sudo sensors-detect
[...]
Found `Nuvoton NCT6687D eSIO'                               Success!
    (address 0xa20, driver `nct6683')
[...]
Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Driver `nct6683':
  * ISA bus, address 0xa20
    Chip `Nuvoton NCT6687D eSIO' (confidence: 9)
[...]
```

```
❯ cat /etc/conf.d/lm_sensors
[...]
HWMON_MODULES="coretemp nct6683"
```

```
❯ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +48.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +40.0°C  (high = +80.0°C, crit = +100.0°C)
Core 4:        +41.0°C  (high = +80.0°C, crit = +100.0°C)
Core 8:        +42.0°C  (high = +80.0°C, crit = +100.0°C)
Core 12:       +40.0°C  (high = +80.0°C, crit = +100.0°C)
Core 16:       +42.0°C  (high = +80.0°C, crit = +100.0°C)
Core 20:       +41.0°C  (high = +80.0°C, crit = +100.0°C)
Core 24:       +42.0°C  (high = +80.0°C, crit = +100.0°C)
Core 28:       +38.0°C  (high = +80.0°C, crit = +100.0°C)
Core 32:       +45.0°C  (high = +80.0°C, crit = +100.0°C)
Core 33:       +45.0°C  (high = +80.0°C, crit = +100.0°C)
Core 34:       +45.0°C  (high = +80.0°C, crit = +100.0°C)
Core 35:       +45.0°C  (high = +80.0°C, crit = +100.0°C)
Core 36:       +42.0°C  (high = +80.0°C, crit = +100.0°C)
Core 37:       +42.0°C  (high = +80.0°C, crit = +100.0°C)
Core 38:       +42.0°C  (high = +80.0°C, crit = +100.0°C)
Core 39:       +42.0°C  (high = +80.0°C, crit = +100.0°C)
Core 40:       +43.0°C  (high = +80.0°C, crit = +100.0°C)
Core 41:       +43.0°C  (high = +80.0°C, crit = +100.0°C)
Core 42:       +43.0°C  (high = +80.0°C, crit = +100.0°C)
Core 43:       +43.0°C  (high = +80.0°C, crit = +100.0°C)
Core 44:       +40.0°C  (high = +80.0°C, crit = +100.0°C)
Core 45:       +40.0°C  (high = +80.0°C, crit = +100.0°C)
Core 46:       +40.0°C  (high = +80.0°C, crit = +100.0°C)
Core 47:       +40.0°C  (high = +80.0°C, crit = +100.0°C)

nvme-pci-0600
Adapter: PCI adapter
Composite:    +40.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +40.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +50.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +105.0°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:            N/A  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +53.9°C  (low  =  -5.2°C, high = +89.8°C)
                       (crit = +93.8°C)

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      787.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3500 RPM)
edge:         +53.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +56.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +58.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
PPT:          30.00 W  (cap = 215.00 W)
```

Wer-Wolf, 2023-06-03:
Can you provide the full dmesg output?

mindrunner, 2023-06-03:
Turned out, that I needed the `nct6687` module which (for unknown) reason did not make it upstream, yet.

Relevant Lines (nct6683 module is blacklisted):
```
[   15.136100] nct6687: loading out-of-tree module taints kernel.
[   15.136114] nct6687: module verification failed: signature and/or required key missing - tainting kernel
[   16.143967] nct6687: Found NCT6687D or compatible chip at 0x004e:0x0a20
[   16.149130] nct6687 nct6687.2592: NCT6687D EC firmware version 1.0 build 11/01/22
```

```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.10 V  (min = +11.90 V, max = +12.26 V)
+5V:             5.10 V  (min =  +5.09 V, max =  +5.12 V)
+3.3V:           0.00 V  (min =  +0.00 V, max =  +3.34 V)
CPU Soc:         1.32 V  (min =  +0.72 V, max =  +1.43 V)
CPU Vcore:     560.00 mV (min =  +0.54 V, max =  +0.57 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            2.27 V  (min =  +2.14 V, max =  +3.22 V)
Chipset:       788.00 mV (min =  +0.78 V, max =  +0.80 V)
CPU SA:        896.00 mV (min =  +0.88 V, max =  +0.91 V)
Voltage #2:      1.48 V  (min =  +1.48 V, max =  +1.49 V)
AVCC3:           3.36 V  (min =  +3.34 V, max =  +3.37 V)
AVSB:            0.00 V  (min =  +0.00 V, max =  +3.34 V)
VBat:            0.00 V  (min =  +0.00 V, max =  +1.06 V)
CPU Fan:          0 RPM  (min =    0 RPM, max = 1524 RPM)
Pump Fan:      1013 RPM  (min =    0 RPM, max = 3252 RPM)
System Fan #1:  317 RPM  (min =    0 RPM, max = 1520 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:  320 RPM  (min =    0 RPM, max = 1465 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:  312 RPM  (min =    0 RPM, max = 1449 RPM)
CPU:            +51.0°C  (low  = +32.0°C, high = +99.0°C)
System:         +48.5°C  (low  = +28.5°C, high = +49.0°C)
VRM MOS:        +53.5°C  (low  = +25.0°C, high = +55.0°C)
PCH:            +60.0°C  (low  = +37.0°C, high = +62.0°C)
CPU Socket:     +46.5°C  (low  = +24.5°C, high = +55.5°C)
PCIe x1:         +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
```

mindrunner, 2023-06-03:
Happy to provide more information if needed somehow, but I consider it solved for my side. Seems like not everything is reported properly, but I can monitor critical temps, read fan speed and control them via pwm, which is all I need!

Wer-Wolf, 2023-06-04:
Strange, the `nct6683` driver should support your model. Can you create a bugreport on [bugzilla](https://bugzilla.kernel.org)  regarding the missing support for your motherboard?

mindrunner, 2023-06-24:
https://bugzilla.kernel.org/show_bug.cgi?id=217591