lm-sensors/lm-sensors #447: Gigabyte MB MC62-G40 + AMD Treadripper PRO 5975WX  - No CPU/Fans detected

funky28, 2023-05-28:
Hi, I have a Gigabyte motherboard MC62-G40 with a AMD Treadripper PRO 5975WX  running Ubuntu 22.04. After running sensors -detect, only the NVME drives are detected, no CPU temperature or system fans.

Reading online, I found several information regarding the it87 module and how it has to be manually installed and loaded. I am unclear what chipset my MB uses (or how to find out... not much came up on google), therefore I decided to give that avenue a try. I followed the guides specified [here](https://github.com/lm-sensors/lm-sensors/issues/154), [here](https://forum.level1techs.com/t/configuring-lm-sensors-on-x570-aorus-master-to-get-cpu-temps-fans-etc/166170) and [here - section 6.13](https://wiki.archlinux.org/title/Lm_sensors?ref=techhut.tv).

In summary, I did the following:

- Added `quiet splash intel_pstate=disable acpi_enforce_resources=lax `to my grub - rebooted
- Cloned [it87](https://github.com/rpavlik/it87)
- After installing the proper dependencies, ran `sudo make dkms`
- Forced it87 by creating /etc/modprobe.d/it87.conf with options it87 force_id=0x8628 and added it87 in /etc/modules-load.d/it87.conf (with module name it87)

After that, running the command `sudo modprobe it87 `doest do anything (where as before the installation it said: `modprobe: ERROR: could not insert 'it87': No such device)`. I forgot exactly how I came up with this, but at some point I noticed the the the module name was: it87-unknown.

Running sensors -detect yielded the same results as before installing the it87 module:
```
# sensors-detect version 3.6.0
# System: GIGABYTE MC62-G40-00 [0100]
# Kernel: 5.19.0-42-generic x86_64
# Processor: AMD Ryzen Threadripper PRO 5975WX 32-Cores (25/8/2)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

```
Since then, I removed the it87 module by running: `sudo make dkms_clean` as it provided no benefit.

Searching online for my specific MB and CPU combination did not yield a lot of pings, and thus in am unsure what is my chipset  (so that I can figure out what modules I need) and how to get lm-sensors to report my system fans and CPU temperatures.

Any help is greatly appreciated.

Thanks

Wer-Wolf, 2023-06-03:
Your motherboard uses a ast2600 bmc for system management, it should support the IPMI interface. Unfortunatly, lm-sensors does not provide support for sensors exposed over IPMI yet, you might want to try other tools like ipmitool.

The k10temp driver should normally support your amd cpu, you could try the latest kernel (6.3.X) and open a bugreport on [bugzilla](bugzilla.kernel.org) if k10temp still does not support your cpu.

funky28, 2023-06-03:
Thank you so much for the reply! after creating this post, I was able to install the IPMItool and get the die temp as reported by the BMC (which by the way, Gigabyte provides a mean see all of it via their BMC access on a dedicated LAN interface). I really wanted to get each core temp as reported by the CPU since with other machines/servers that I have I have noticed a couple of degree difference between the IPMI die temp and the CPU reported temp.

I don't want to venture too much in updating the kernel to > 6.3.x as I am unsure what it will do to the other things I have running on the server (KVM, Docker, Backups, Repetitive tasks, etc....), perhaps will try with an old HD/SSD and use it as a test to see what would happen without having to break my current system.

I guess, short of a better/less invasive fix, I will have to stay with the IPMI temp reporting and hope that I can run some test on a dummy machine to try and see how reliable the kernel update would be to my system.

Thanks for the help!