lm-sensors/lm-sensors #448: NCT6775 module causes an ACPI conflict

Gleis1234, 2023-06-12:
I added the nct6775 module, because I wanted to control the fans on my ASUS B550M-A board. In syslog an ACPI conflict is stated.
`Jun 12 17:36:02 magickubuntu12 kernel: [    6.287911] nct6775: Found NCT6798D or compatible chip at 0x2e:0x290
Jun 12 17:36:02 magickubuntu12 kernel: [    6.287917] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000290-0x0000000000000299 (\AMW0.SHWM) (20210730/utaddress-204)
Jun 12 17:36:02 magickubuntu12 kernel: [    6.287923] ACPI: OSL: Resource conflict; ACPI support missing from driver?
`
I followed https://github.com/lm-sensors/lm-sensors/issues/134#issuecomment-932969738 to switch off ACPI. 
The suggested acpi_enforce_resources=lax does not help, but acpi=off solved the conflict and the sensor values became visible.
But now my NVIDIA GPU could not start because of missing interupts.
Is there another way to solve this conflict?

Wer-Wolf, 2023-06-14:
Which kernel version do you use? On recent kernels, a special hwmon driver might be available which does not rely on dangerous acpi hacks.

Gleis1234, 2023-06-14:
Thank for your answer. I have an 5.15.0-73 kernel installed. Where is the new driver available?

Wer-Wolf, 2023-06-14:
Idk for sure, but it might be promising to try the latest kernel.

Gleis1234, 2023-06-15:
I installed linux-hwe-5.19-generic and now I can see all vaues. Thanx a lot for the info. It helped a lot.

closed 2023-06-15 by Gleis1234