lm-sensors/lm-sensors #451: No fans for MSI z790 Tomahawk

HathewayWill, 2023-06-19:
output from sensors-detect

```
sudo sensors-detect
# sensors-detect version 3.6.0
# System: Micro-Star International Co., Ltd. MS-7D91 [4.0]
# Board: Micro-Star International Co., Ltd. MAG Z790 TOMAHAWK WIFI (MS-7D91)
# Kernel: 5.19.0-45-generic x86_64
# Processor: 13th Gen Intel(R) Core(TM) i9-13900K (6/183/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): yes
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): yes
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd592

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): yes
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): yes
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

[... 179 lines cut ...]
Core 41:       +45.0°C  (high = +80.0°C, crit = +100.0°C)
Core 42:       +45.0°C  (high = +80.0°C, crit = +100.0°C)
Core 43:       +45.0°C  (high = +80.0°C, crit = +100.0°C)
Core 44:       +46.0°C  (high = +80.0°C, crit = +100.0°C)
Core 45:       +46.0°C  (high = +80.0°C, crit = +100.0°C)
Core 46:       +46.0°C  (high = +80.0°C, crit = +100.0°C)
Core 47:       +46.0°C  (high = +80.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +105.0°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +29.0°C  

nvme-pci-0200
Adapter: PCI adapter
Composite:    +34.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +34.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +35.9°C  (low  = -273.1°C, high = +65261.8°C)
```

cpu fan not showing up

Wer-Wolf, 2023-06-23:
Which kernel version are you using? I belief support for chip id `0xd592` was added recently to the `nct6683` driver.

HathewayWill, 2023-06-24:
5.19.0-45-generic

I had to install it from scratch using this method
https://github.com/Fred78290/nct6687d

Wer-Wolf, 2023-06-24:
Try to manually load the `nct6683` driver using modprobe: `sudo modprobe nct6683`.
If the driver refuses to load, try `sudo modprobe nct6683 force=true`.

If you require the `force` option, then could you please share the output of `dmesg`?

HathewayWill, 2023-06-25:
> Try to manually load the `nct6683` driver using modprobe: `sudo modprobe nct6683`. If the driver refuses to load, try `sudo modprobe nct6683 force=true`.
> 
> If you require the `force` option, then could you please share the output of `dmesg`?
[dmesg.log](https://github.com/lm-sensors/lm-sensors/files/11860832/dmesg.log)

HathewayWill, 2023-07-04:
> Try to manually load the `nct6683` driver using modprobe: `sudo modprobe nct6683`. If the driver refuses to load, try `sudo modprobe nct6683 force=true`.
> 
> If you require the `force` option, then could you please share the output of `dmesg`?

This seems to work for me @Wer-Wolf 

```
sudo apt install git  -y
sudo -s
cd
git clone https://github.com/Fred78290/nct6687d
cd nct6687d
make install
sudo sh -c 'echo "nct6687 force=true" >> /etc/modules'
```

HathewayWill, 2023-07-12:
> Maybe the upcoming 6.5 kernel does have the necessary MSI customer ID enabled to allow the `nct6683` driver to work on your hardware.

@Wer-Wolf is there any documentation on the release?

llyyr, 2023-07-17:
you don't need to use `force=1` anymore, as the necessary customer ID was added in 6.5 kernel. however I'm not sure how to teach lmsensors/sensors-detect to load nct6683 driver for nct6687d sensor

`sudo modprobe nct6683` manually on 6.5, or `sudo modprobe nct6683 force=1` on older kernels should work

HathewayWill, 2023-07-17:
> you don't need to use `force=1` anymore, as the necessary customer ID was added in 6.5 kernel. however I'm not sure how to teach lmsensors/sensors-detect to load nct6683 driver for nct6687d sensor [...]

@llyyr @Wer-Wolf 

I just noticed this:
```
workhorse@workhorse-MS-7D91:~/Desktop$ uname -r
5.19.0-46-generic

workhorse@workhorse-MS-7D91:~/Desktop$ uname -a
Linux workhorse-MS-7D91 5.19.0-46-generic #47~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jun 21 15:35:31 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

```

llyyr, 2023-07-17:
This has actually already been fixed, but a new release needs to be cut

https://github.com/lm-sensors/lm-sensors/commit/ac03cccaf92bdb2661d6e8bdde6ea78f60527b0a

This should load the right driver on kernel 6.5

HathewayWill, 2023-07-17:
> This has actually already been fixed, but a new release needs to be cut
> 
> [ac03ccc](https://github.com/lm-sensors/lm-sensors/commit/ac03cccaf92bdb2661d6e8bdde6ea78f60527b0a)
> 
> This should load the right driver on kernel 6.5

@llyyr  Am I not using the right kernal?

llyyr, 2023-07-17:
sorry I don't know anything about Ubuntu or LTS releases. It should still work if you do `sudo modprobe nct6687 force=1`

Wer-Wolf, 2023-07-18:
You can also add the necessary modprobe config files manually, see [here](https://manpages.ubuntu.com/manpages/xenial/man5/modules-load.d.5.html).

HathewayWill, 2023-07-18:
Not really sure how to do this since I'm very new to Ubuntu environment.

Could you walk me through it with steps?

Wer-Wolf, 2023-07-19:
Create the file `nct6683.conf` inside `/etc/modules-load.d/` with the following content:
`nct6683 force=1`

This should automatically load the driver during boot.