lm-sensors/lm-sensors #453: Thinkpad X390 fan is always 0 RPM.

likengmx, 2023-06-27:
Thinkpad X390 yoga
fan1: always 0 RPM (although it is running)
On Windows there is a program NBFC and a profile for Lenovo X390 and you can read fan speed and control it.

NBFC: [https://github.com/hirschmann/nbfc](url)
Profile X390: [https://github.com/riccardik/nbfcprofiles/blob/main/LENOVO%20X390.xml](url)

Interesting, profile uses a register 148 to write and 149 for read.
Maybe this will serve as a clue to see what needs to be modified in lm-sensors.

Wer-Wolf, 2023-07-05:
Can you share the output of `sensors`?

likengmx, 2023-07-05:
```

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

thinkpad-isa-0000
Adapter: ISA adapter
fan1:           0 RPM
temp1:         +1.0°C  
temp2:         +1.0°C  
temp3:         +4.0°C  
temp4:         +0.0°C  
temp5:       +111.0°C  
temp6:       +111.0°C  
temp7:         +0.0°C  
temp8:         +0.0°C  
temp9:        +64.0°C  
temp10:       +64.0°C  
temp11:       -80.0°C  
temp12:        +0.0°C  
temp13:        +0.0°C  
temp14:        +0.0°C  
temp15:       +16.0°C  
temp16:        +0.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +47.0°C  (high = +100.0°C, crit = +100.0°C)

nvme-pci-0300
Adapter: PCI adapter
Composite:    +29.9°C  (low  = -273.1°C, high = +83.8°C)
                       (crit = +84.8°C)
Sensor 1:     +29.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +29.9°C  (low  = -273.1°C, high = +65261.8°C)

ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +32.0°C  

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +45.0°C  

BAT0-acpi-0
Adapter: ACPI interface
in0:          11.62 V
```

Wer-Wolf, 2023-07-11:
Maybe you can create a bugreport on [bugzilla](https://bugzilla.kernel.org) so the maintainer of the kernel driver can help you with this?

likengmx, 2023-07-11:
https://bugzilla.kernel.org/show_bug.cgi?id=217657

hacxman, 2024-03-19:
bump. whats wrong? kernel bug has zero response. i need to stop fan whining