lm-sensors/lm-sensors #455: Cannot run pwmconfig (There are no pwm-capable sensor modules installed)

RedStarITA, 2023-07-28:
Hi, I'm trying to control the fan speed in my Laptop(Victus 15-fa0xxxnl. I've already installed lm_sensors and run `sudo sensor` detect. However, everytime i try to run `sudo pwnconfig `i get this output:

```
# pwmconfig version 3.6.0+git
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

/usr/bin/pwmconfig: There are no pwm-capable sensor modules installed
```

and, the output for when i do `sensors `is:

```
p-isa-0000
Adapter: ISA adapter
fan1:        2588 RPM
fan2:        2417 RPM

BAT0-acpi-0
Adapter: ACPI interface
in0:          17.25 V  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +29.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 8:        +33.0°C  (high = +100.0°C, crit = +100.0°C)
Core 12:       +31.0°C  (high = +100.0°C, crit = +100.0°C)
Core 16:       +32.0°C  (high = +100.0°C, crit = +100.0°C)
Core 17:       +33.0°C  (high = +100.0°C, crit = +100.0°C)
Core 18:       +33.0°C  (high = +100.0°C, crit = +100.0°C)
Core 19:       +33.0°C  (high = +100.0°C, crit = +100.0°C)
Core 20:       +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 21:       +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 22:       +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 23:       +34.0°C  (high = +100.0°C, crit = +100.0°C)

nvme-pci-0200
Adapter: PCI adapter
Composite:    +32.9°C  (low  =  -0.1°C, high = +79.8°C)
                       (crit = +81.8°C)
Sensor 1:     +32.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +38.0°C  

```

Can someone help me about this issue? I've tried so many times(Even re-installed Arch)...

Wer-Wolf, 2023-07-31:
Can you find out which driver is controlling your "p" sensor chip?

RedStarITA, 2023-07-31:
Why do you mean? I've done "`Sudo lsmod | grep pwm`" and the output is none.

RedStarITA, 2023-07-31:
> Can you find out which driver is controlling your "p" sensor chip?

Sorry bro, i just saw that i've deleted an "h", so it's not "p" sensor chip, but "hp" sensor chip

Wer-Wolf, 2023-07-31:
Oh, then your hwmon chip is likely handled by the `hp-wmi` driver. AFAIK, this driver can only force the fans to run at max speed, the hardware interface does not allow for setting the pwm values directly.

Wer-Wolf, 2023-07-31:
You might be able to set the thermal profile using this driver thou.

RedStarITA, 2023-08-01:
How do i do that? Setting the fans to max speed is excatly what i do on Windows with the Omen Gaming Hub software

Wer-Wolf, 2023-08-01:
Writing `0` to `/sys/class/hwmon/hwmonX/pwm1_enable`, where X is the number of the hwmon chip being named "hp".

Wer-Wolf, 2023-08-01:
You might also be able to modify the acpi platform profile.

RedStarITA, 2023-08-01:
> You might also be able to modify the acpi platform profile.

Thank you so much bro, modifying the variable in pwm1_enable to 0 was enough to let the fans kick in, first time i've ever heard so much noise and power coming from my fans on linux, thank really really so much, now i can finally return to game on linux without worries od overheating😆

Wer-Wolf, 2023-08-01:
Great to hear :)

closed 2023-08-02 by RedStarITA