lm-sensors/lm-sensors #459: pwmconfig: fans not returned to normal speed

rpigott, 2023-08-22:
Hi.

I was trying to mitigate an issue I'm having where it appears that my fans stop spinning under load soon after they reach maximum speed. I had not previously run pwmconfig.

When running pwmconfig, I get the following outputs:

<details><summary>pwmconfig</summary>

```
# pwmconfig version 3.6.0+git
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

Found the following devices:
   hwmon0 is acpitz
   hwmon1 is nvme
   hwmon2 is nvme
   hwmon3 is dell_smm
   hwmon4 is coretemp
   hwmon5 is iwlwifi_1

Found the following PWM controls:
   hwmon3/pwm1           current value: 255
   hwmon3/pwm2           current value: 255
   hwmon3/pwm3           current value: 255

Giving the fans some time to reach full speed...
Found the following fan sensors:
   hwmon3/fan1_input     current speed: 4356 RPM
   hwmon3/fan2_input     current speed: 1369 RPM
   hwmon3/fan3_input     current speed: 1483 RPM

Warning!!! This program will stop your fans, one at a time,
for approximately 5 seconds each!!!
This may cause your processor temperature to rise!!!
If you do not want to do this hit control-C now!!!
Hit return to continue: 
Testing pwm control hwmon3/pwm1 ...
  hwmon3/fan1_input ... speed was 4356 now 1136
    It appears that fan hwmon3/fan1_input
    is controlled by pwm hwmon3/pwm1
Would you like to generate a detailed correlation (y)? [n]
  hwmon3/fan2_input ... speed was 1369 now 1506
    no correlation
  hwmon3/fan3_input ... speed was 1483 now 1455
    no correlation

Testing pwm control hwmon3/pwm2 ...
  hwmon3/fan1_input ... speed was 4356 now 4347
    no correlation
  hwmon3/fan2_input ... speed was 1369 now 1368
    no correlation
  hwmon3/fan3_input ... speed was 1483 now 1485
    no correlation

No correlations were detected.
There is either no fan connected to the output of hwmon3/pwm2,
or the connected fan has no rpm-signal connected to one of
the tested fan sensors. (Note: not all motherboards have
the pwm outputs connected to the fan connectors,
check out the hardware database on http://www.almico.com/forumindex.php)

Did you see/hear a fan stopping during the above test (n)? [n]
Testing pwm control hwmon3/pwm3 ...
  hwmon3/fan1_input ... speed was 4356 now 4338
    no correlation
  hwmon3/fan2_input ... speed was 1369 now 1367
    no correlation
  hwmon3/fan3_input ... speed was 1483 now 1506
    no correlation

No correlations were detected.
There is either no fan connected to the output of hwmon3/pwm3,
or the connected fan has no rpm-signal connected to one of
the tested fan sensors. (Note: not all motherboards have
the pwm outputs connected to the fan connectors,
check out the hardware database on http://www.almico.com/forumindex.php)

Did you see/hear a fan stopping during the above test (n)? [n]
Testing is complete.
Please verify that all fans have returned to their normal speed.

The fancontrol script can automatically respond to temperature changes
of your system by changing fanspeeds.
Do you want to set up its configuration file now (y)? [n] 
```
</details>

My trouble is that after pwmconfig has exited, following the line

> Please verify that all fans have returned to their normal speed

they have not returned to normal speed. Fan1 is still running at its maximum speed. This fan quickly ramps up to high speed when pwmconfig starts, is disabled for 5 seconds during the test, and then returns to full speed and remains that way... forever. Re-running pwmconfig can disable it again for 5 seconds during the test before it ramps up again.

Curiously, a reboot does _not_ change the fan speed at all. It remains at high speed during the entire reboot process. Powering off the PC finally stops the fan when the power supply clicks it off. On the next boot it seems normal.

What can be done to correct the fan behavior if it does not return to normal speed? And possibly, how might I approach the original problem I observed that the fan apparently stops when under heavy load?

Wer-Wolf, 2023-09-08:
Can you try to load the dell-smm-hwmon kernel module with the "fan_max" option set to 4?
Use the following commands as root:

    modprobe -r dell-smm-hwmon
    modprobe dell-smm-hwmon fan_max=4

Wer-Wolf, 2023-09-08:
Can you tell me how the fan behaves now when set to 255?
You can set the fan speed of fan X using `echo <speed> > /sys/class/hwmon/hwmon3/pwmX` as root.

It seems your your hwmon driver (dell-smm-hwmon) fails to notice that each fan supports 4 fan states (off, low, high, auto), with your fan starting in state 3 (auto). The driver however only uses fan states up to 2 (high), so it cannot switch back to automatic mode.

Do the fanX_min/fanX_max/fanX_target attributes appear under "/sys/class/hwmon/hwmon3/" if you changed the "fan_max" option?

rpigott, 2023-09-11:
Hi thanks for the response.

The fan min/max values do change (same workstation, but a new gpu was installed that changed the fan numbers):

```
$ grep -H $ /sys/class/hwmon/hwmon4/fan1_{min,max}  
/sys/class/hwmon/hwmon4/fan1_min:0
/sys/class/hwmon/hwmon4/fan1_max:3000
$ sudo modprobe -r dell-smm-hwmon
$ sudo modprobe dell-smm-hwmon fan_max=3
$ grep -H $ /sys/class/hwmon/hwmon4/fan1_{min,max}
/sys/class/hwmon/hwmon4/fan1_min:0
/sys/class/hwmon/hwmon4/fan1_max:4950
```

The behavior at pwm 255 also changes:

```
$ sudo modprobe dell-smm-hwmon # no fan_max=3
$ echo 255 | sudo tee /etc/class/hwmon/hwmon4/pwm1
$ sudo modprobe -r dell-smm-hwmon && sudo modprobe dell-smm-hwmon fan_max=3
$ echo 255 | sudo tee /etc/class/hwmon/hwmon4/pwm1
```

Without fan_max=3,  fans ramp up to ~4300rpm according to sensors. Anything >= 192 seems to work, less than that the fans go low.

With fan_max=3, the fans are auto I think. They drop immediately and ramp up slowly to ~3000rpm if I begin to build linux, then ramp back down after about 1 minute while the (still running) build completes.

With this parameter, speed < 128 seems low, 128 <= speed < 213 is high.

Wer-Wolf, 2023-09-12:
Ok, what happens if you load the module with "fan_max=4"? Do the min/max attrs disappear?

rpigott, 2023-09-12:
Yes, they become 0:

```
$ sudo modprobe -r dell-smm-hwmon          
$ sudo modprobe dell-smm-hwmon fan_max=4
$ grep -H $ /sys/class/hwmon/hwmon4/fan1_{min,max}
/sys/class/hwmon/hwmon4/fan1_min:0
/sys/class/hwmon/hwmon4/fan1_max:0
```

Wer-Wolf, 2023-09-12:
And what happens if you set `pwmX` to 255 in this case?

rpigott, 2023-09-12:
It is not permitted:
```
$ sudo modprobe dell-smm-hwmon fan_max=4
# echo 255 > /sys/class/hwmon/hwmon4/pwm1
echo: write error: invalid argument
```

dm17, 2025-02-19:
Same here. It just outputs "Please verify that all fans have returned to their normal speed" - but why does it output this message? Does it give us some way to verify? No - they're just left at max speed.