lm-sensors/lm-sensors #460: Wrong reading from power_meter-acpi power sensor (4.29 MW)

urbinek, 2023-09-12:
I've found same issue on tracker [lm_sensors-3.4 get 4.29 MW Megawatt](https://github.com/lm-sensors/lm-sensors/issues/130), but in my case it seems to be persistent. 

### Readings
```Bash
# sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +51.0 C  (high = +80.0 C, crit = +90.0 C)
Core 0:        +51.0 C  (high = +80.0 C, crit = +90.0 C)
Core 1:        +50.0 C  (high = +80.0 C, crit = +90.0 C)
Core 2:        +50.0 C  (high = +80.0 C, crit = +90.0 C)
Core 3:        +49.0 C  (high = +80.0 C, crit = +90.0 C)
Core 4:        +49.0 C  (high = +80.0 C, crit = +90.0 C)
Core 5:        +51.0 C  (high = +80.0 C, crit = +90.0 C)
Core 6:        +49.0 C  (high = +80.0 C, crit = +90.0 C)
Core 7:        +50.0 C  (high = +80.0 C, crit = +90.0 C)

power_meter-acpi-0
Adapter: ACPI interface
power1:        4.29 MW (interval =   1.00 s
```
Obviously I don't have small reactor attached :)
### hwmon test from [[olysonek](https://github.com/olysonek)](https://github.com/lm-sensors/lm-sensors/issues/130#issuecomment-420333543)
```Bash
# cat /tmp/test_hwmon.sh
for i in /sys/class/hwmon/hwmon*/{,device/}name; do
        if [ "$(cat $i)" = power_meter ]; then
                dir=$(dirname $i)
                file="$dir/power1_input"
                if [ -f "$file" ]; then
                        cat "$file"
                        break
                fi
                file="$dir/power1_average"
                if [ -f "$file" ]; then
                        cat "$file"
                        break
                fi
        fi
done

# bash test_hwmon.sh
4294967295000
```
### DSDT dump [[olysonek](https://github.com/olysonek)](https://github.com/lm-sensors/lm-sensors/issues/130#issuecomment-420631603)
```Bash
# cat /sys/firmware/acpi/tables/DSDT > /tmp/dsdt.dat
```
[dsdt.zip](https://github.com/lm-sensors/lm-sensors/files/12583981/dsdt.zip)
### Package info
```Bash
# dnf info lm_sensors.x86_64
Last metadata expiration check: 2:05:17 ago on Tue Sep 12 07:48:35 2023.
Installed Packages
Name         : lm_sensors
Version      : 3.4.0
Release      : 23.20180522git70f7e08.el8
Architecture : x86_64
Size         : 445 k
Source       : lm_sensors-3.4.0-23.20180522git70f7e08.el8.src.rpm
Repository   : @System
From repo    : baseos
Summary      : Hardware monitoring tools
URL          : http://github.com/lm-sensors/lm-sensors/
License      : GPLv2+ and Verbatim and MIT
Description  : The lm_sensors package includes a collection of modules for general SMBus
             : access and hardware monitoring.
```
### Loaded drivers
```Bash
# lsmod | grep acpi
acpi_ipmi              16384  0
acpi_pad              180224  0
acpi_power_meter       20480  0
ipmi_msghandler       114688  4 ipmi_devintf,ipmi_si,acpi_ipmi,ipmi_ssif
```
```Bash
# modinfo acpi_power_meter
filename:       /lib/modules/4.18.0-513.el8.x86_64/kernel/drivers/hwmon/acpi_power_meter.ko.xz
license:        GPL
description:    ACPI 4.0 power meter driver
author:         Darrick J. Wong <darrick.wong@oracle.com>
rhelversion:    8.9
srcversion:     60AC415BBE246C69A3410F7
alias:          acpi*:ACPI000D:*
depends:
intree:         Y
name:           acpi_power_meter
vermagic:       4.18.0-513.el8.x86_64 SMP mod_unload modversions
sig_id:         PKCS#7
signer:         Red Hat Enterprise Linux kernel signing key
sig_key:        1F:81:C5:59:B5:5A:8E:68:72:AB:6F:96:38:BB:B5:E7:C6:5B:E7:76
sig_hashalgo:   sha256
signature:      17:C0:C9:45:57:9B:26:2F:48:1E:06:32:81:15:D7:4B:1B:F9:13:49:
                A8:F1:A0:8C:12:5C:D7:5F:D7:11:81:63:A6:18:CE:DA:76:39:17:1C:
                9D:8D:D7:62:A8:63:75:52:B2:21:E9:01:3F:3A:09:E6:11:2B:50:BC:
                A8:E3:DC:41:91:3E:0B:E2:A3:C9:B7:43:46:11:2B:C7:BC:68:C0:E3:
                B6:57:29:03:48:24:1D:56:2F:CA:B9:25:D9:99:31:90:1F:45:04:C7:
                30:0E:2F:6D:12:45:A8:13:C3:1C:3A:11:10:39:85:4C:4D:30:87:CB:
                18:AA:B6:C3:55:CE:95:15:99:1E:DC:77:C7:FB:0E:04:CA:E2:8C:8F:
                67:F7:0A:69:C2:92:26:13:4A:F1:8B:14:86:07:B7:72:61:05:9A:2B:
                CB:71:8D:BE:1A:21:12:17:DA:36:91:14:BB:43:D5:DF:02:56:09:7D:
                FE:C6:61:69:13:D3:DD:8E:B6:0A:09:2E:D2:B2:E6:46:66:51:BD:37:
                0F:97:D8:EC:9D:F0:6E:38:A6:B7:60:AF:9E:3D:A0:18:52:07:8B:4E:
                9A:95:D2:9D:CD:9D:3E:C1:89:67:1D:7B:4A:48:0C:5E:78:12:92:DF:
                46:45:6B:5A:33:42:B8:95:F8:20:0F:11:12:2B:A8:EC:E5:76:60:DB:
                98:D1:E0:4D:2D:EE:9D:BA:E1:C0:1E:F1:3C:02:F2:02:05:34:06:0B:
                9C:C2:BD:8A:71:29:6E:B2:45:D0:AA:F0:B3:87:C5:20:57:E9:0B:5A:
                BF:B7:7B:AE:C3:E5:89:AF:13:FB:F3:EF:16:A4:BA:F5:BA:32:93:69:
                55:CD:6B:85:0A:06:30:7C:AB:F7:E9:F3:A9:7C:A8:1A:08:01:F3:90:
                8A:7A:F4:33:91:D7:6E:20:9A:56:E7:38:24:CB:7B:A5:D0:D9:DC:7F:
                CB:02:40:CF:71:8F:1D:62:E9:BB:1E:5E:09:5D:EC:1B:97:38:E4:F2:
                BC:04:4E:B9
parm:           force_cap_on:Enable power cap even it is unsafe to do so. (bool)
```
### OS logs
```Bash
# dmesg | grep power -C4
[   12.417391] EDAC MC1: Giving out device to module skx_edac controller Skylake Socket#0 IMC#1: DEV 0000:64:0c.0 (INTERRUPT)
[   12.441598] intel_rapl_common: Found RAPL domain package
[   12.441612] intel_rapl_common: Found RAPL domain dram
[   12.441617] intel_rapl_common: DRAM domain energy unit 15300pj
[   12.879140] power_meter ACPI000D:00: Found ACPI power meter.
[   12.906481] XFS (nvme0n1p3): Mounting V5 Filesystem
[   12.912618] XFS (nvme0n1p3): Ending clean mount
[   13.816231] power_meter ACPI000D:00: Found ACPI power meter.
[   13.816262] power_meter ACPI000D:00: Ignoring unsafe software power cap!
[   17.035012] Adding 33554428k swap on /dev/zd16.  Priority:-2 extents:1 across:33554428k SSFS
[   17.040071] GPT:Primary header thinks Alt. header is not at the end of the disk.
[   17.040074] GPT:134217727 != 146800639
[   17.040076] GPT:Alternate GPT header not at the end of the disk.
```
```Bash
# dmesg | grep -i 'power\|acpi' -C4 > tmp/dmesg.log
```
[dmesg.log](https://github.com/lm-sensors/lm-sensors/files/12584005/dmesg.log)
### Platform
```Bash
# cat /etc/centos-release
CentOS Stream release 8

# cat /etc/os-release
NAME="CentOS Stream"
VERSION="8"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="8"
PLATFORM_ID="platform:el8"
PRETTY_NAME="CentOS Stream 8"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:8"
HOME_URL="https://centos.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux 8"
REDHAT_SUPPORT_PRODUCT_VERSION="CentOS Stream"
```
```Bash
# uname -a
Linux xen.urbinek.eu 4.18.0-513.el8.x86_64 #1 SMP Fri Aug 25 14:33:28 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
``` 
### Hardware
[SuperMicro X11DPI-NT](https://www.supermicro.com/en/products/motherboard/x11dpi-nt)

Wer-Wolf, 2023-09-12:
The power meter device is not described inside the DSDT. Could you share a complete dump of all ACPI tables using acpidump?

mmiarka, 2023-09-13:
@Wer-Wolf here-> [acpidump.zip](https://github.com/lm-sensors/lm-sensors/files/12595324/acpidump.zip)

Wer-Wolf, 2023-09-18:
According to the ACPI power meter standard, 0xffffffff is returned to signal an error. This seems to happen in this case, since 4294967295000 / 1000 = 0xffffffff.

Can you try to tweak the sysfs attributes of the hwmon device? Maybe the average interval is too high. Or maybe you need to perform a BIOS update/some BIOS configuration?

urbinek, 2023-09-18:
> Or maybe you need to perform a BIOS update/some BIOS configuration?
There is nothing much on ACPI configuration:
[image]

Bios, IPMI and OS upgrade didn't help

> Can you try to tweak the sysfs attributes of the hwmon device? Maybe the average interval is too high. 
How can i change them? Never done that

Wer-Wolf, 2023-09-18:
Alright, could you share the values of all sysfs attrs under /sys/class/hwmon/hwmonX, with hwmonX being the hwmon device of the power meter (hwmonX/name should be helpful here)?

urbinek, 2023-09-18:
```Bash
# grep . /sys/class/hwmon/hwmon1/ -r
/sys/class/hwmon/hwmon1/power/runtime_active_time:0
/sys/class/hwmon/hwmon1/power/runtime_status:unsupported
grep: /sys/class/hwmon/hwmon1/power/autosuspend_delay_ms: Input/output error
/sys/class/hwmon/hwmon1/power/runtime_suspended_time:0
/sys/class/hwmon/hwmon1/power/control:auto
```

Wer-Wolf, 2023-09-18:
I need the sysfs files of the hwmon1 directory, not the underlying directories.

mmiarka, 2023-09-19:
There is not much there
```
# ls -lah /sys/class/hwmon/hwmon1/
razem 0
drwxr-xr-x 3 root root    0 09-18 19:53 .
drwxr-xr-x 3 root root    0 09-18 19:53 ..
lrwxrwxrwx 1 root root    0 09-19 08:00 device -> ../../../ACPI000D:00
drwxr-xr-x 2 root root    0 09-18 19:53 power
lrwxrwxrwx 1 root root    0 09-18 19:53 subsystem -> ../../../../../../class/hwmon
-rw-r--r-- 1 root root 4,0K 09-18 19:53 uevent
# file /sys/class/hwmon/hwmon1/uevent
/sys/class/hwmon/hwmon1/uevent: empty
```

hwmon0 is temperature sensor
```
# ls /sys/class/hwmon/hwmon0/
device/           temp2_crit        temp3_max         temp5_label       temp7_input       temp9_crit_alarm
name              temp2_crit_alarm  temp4_crit        temp5_max         temp7_label       temp9_input
power/            temp2_input       temp4_crit_alarm  temp6_crit        temp7_max         temp9_label
subsystem/        temp2_label       temp4_input       temp6_crit_alarm  temp8_crit        temp9_max
temp1_crit        temp2_max         temp4_label       temp6_input       temp8_crit_alarm  uevent
temp1_crit_alarm  temp3_crit        temp4_max         temp6_label       temp8_input
temp1_input       temp3_crit_alarm  temp5_crit        temp6_max         temp8_label
temp1_label       temp3_input       temp5_crit_alarm  temp7_crit        temp8_max
temp1_max         temp3_label       temp5_input       temp7_crit_alarm  temp9_crit
```

Wer-Wolf, 2023-09-19:
Oh wait, the power meter seems to use the old hwmin sysfs api. Can you show me the content of the "device" directory?

urbinek, 2023-09-19:
Sure, no problem 

```
# ls -lah /sys/class/hwmon/hwmon1/device/
razem 0
drwxr-xr-x  5 root root    0 09-18 19:53 .
drwxr-xr-x 44 root root    0 09-18 19:53 ..
lrwxrwxrwx  1 root root    0 09-18 19:53 driver -> ../../../../bus/acpi/drivers/power_meter
-r--r--r--  1 root root 4,0K 09-18 19:55 hid
drwxr-xr-x  3 root root    0 09-18 19:53 hwmon
drwxr-xr-x  2 root root    0 09-18 19:53 measures
-r--r--r--  1 root root 4,0K 09-18 19:55 modalias
-r--r--r--  1 root root 4,0K 09-19 08:00 name
-r--r--r--  1 root root 4,0K 09-18 19:55 path
lrwxrwxrwx  1 root root    0 09-18 19:53 physical_node -> ../../../platform/ACPI000D:00
drwxr-xr-x  2 root root    0 09-18 19:53 power
-r--r--r--  1 root root 4,0K 09-18 19:55 power1_accuracy
-r--r--r--  1 root root 4,0K 09-19 08:00 power1_average
-rw-r--r--  1 root root 4,0K 09-19 08:00 power1_average_interval
-r--r--r--  1 root root 4,0K 09-18 19:55 power1_average_interval_max
-r--r--r--  1 root root 4,0K 09-18 19:55 power1_average_interval_min
-rw-r--r--  1 root root 4,0K 09-18 19:55 power1_average_max
-rw-r--r--  1 root root 4,0K 09-18 19:55 power1_average_min
-r--r--r--  1 root root 4,0K 09-18 19:55 power1_is_battery
-r--r--r--  1 root root 4,0K 09-18 19:55 power1_model_number
-r--r--r--  1 root root 4,0K 09-18 19:55 power1_oem_info
-r--r--r--  1 root root 4,0K 09-18 19:55 power1_serial_number
-r--r--r--  1 root root 4,0K 09-18 19:55 status
lrwxrwxrwx  1 root root    0 09-19 08:00 subsystem -> ../../../../bus/acpi
-rw-r--r--  1 root root 4,0K 09-19 13:02 uevent
-r--r--r--  1 root root 4,0K 09-18 19:55 uid
```
```
# grep . /sys/class/hwmon/hwmon1/device/  -r
/sys/class/hwmon/hwmon1/device/uevent:DRIVER=power_meter
/sys/class/hwmon/hwmon1/device/uevent:MODALIAS=acpi:ACPI000D:
/sys/class/hwmon/hwmon1/device/power1_average_interval:1000
/sys/class/hwmon/hwmon1/device/power1_accuracy:97.500%
/sys/class/hwmon/hwmon1/device/power/runtime_active_time:0
/sys/class/hwmon/hwmon1/device/power/runtime_status:unsupported
grep: /sys/class/hwmon/hwmon1/device/power/autosuspend_delay_ms: Input/output error
/sys/class/hwmon/hwmon1/device/power/runtime_suspended_time:0
/sys/class/hwmon/hwmon1/device/power/control:auto
/sys/class/hwmon/hwmon1/device/uid:0
/sys/class/hwmon/hwmon1/device/power1_average_interval_min:1000
/sys/class/hwmon/hwmon1/device/hid:ACPI000D
/sys/class/hwmon/hwmon1/device/power1_oem_info:Meter measures total domain
/sys/class/hwmon/hwmon1/device/power1_average:4294967295000
/sys/class/hwmon/hwmon1/device/power1_model_number:Intel(R) Node Manager
/sys/class/hwmon/hwmon1/device/path:\_SB_.PMT1
/sys/class/hwmon/hwmon1/device/power1_is_battery:0
/sys/class/hwmon/hwmon1/device/power1_average_interval_max:3600000
/sys/class/hwmon/hwmon1/device/status:15
/sys/class/hwmon/hwmon1/device/power1_average_min:unknown
/sys/class/hwmon/hwmon1/device/power1_average_max:unknown
/sys/class/hwmon/hwmon1/device/name:power_meter
/sys/class/hwmon/hwmon1/device/hwmon/hwmon1/power/runtime_active_time:0
/sys/class/hwmon/hwmon1/device/hwmon/hwmon1/power/runtime_status:unsupported
grep: /sys/class/hwmon/hwmon1/device/hwmon/hwmon1/power/autosuspend_delay_ms: Input/output error
/sys/class/hwmon/hwmon1/device/hwmon/hwmon1/power/runtime_suspended_time:0
/sys/class/hwmon/hwmon1/device/hwmon/hwmon1/power/control:auto
/sys/class/hwmon/hwmon1/device/modalias:acpi:ACPI000D:
```

urbinek, 2023-09-19:
``` 
# echo 10000 >  /sys/class/hwmon/hwmon1/device/power1_average_interval
```
```
# date ; sensors ; sleep 30 ; date ; sensors
Tue Sep 19 13:48:48 CEST 2023
[...]
power_meter-acpi-0
Adapter: ACPI interface
power1:        4.29 MW (interval =  10.00 s)

Tue Sep 19 13:49:18 CEST 2023
[...]
power_meter-acpi-0
Adapter: ACPI interface
power1:        4.29 MW (interval =  10.00 s)
```

Wer-Wolf, 2023-09-19:
Ok, then it seems that the power readings of the power meter are not updated by the ACPI firmware.
Can you share the results of `grep . /sys/firmware/acpi/interrupts/ -r`?

urbinek, 2023-09-19:
```
# grep . /sys/firmware/acpi/interrupts/ -r
/sys/firmware/acpi/interrupts/gpe2F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe23:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe5D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe51:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe1F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe13:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe4D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe41:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe7B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe0F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe03:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe3D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe31:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe6B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe2D:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe21:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe5B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe1D:       0         disabled     unmasked
/sys/firmware/acpi/interrupts/ff_pwr_btn:       0  EN     enabled      unmasked
/sys/firmware/acpi/interrupts/gpe78:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe11:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe4B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe0D:       0         disabled     unmasked
/sys/firmware/acpi/interrupts/gpe68:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe01:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe3B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe58:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe2B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/ff_rt_clk:       0         disabled     unmasked
/sys/firmware/acpi/interrupts/ff_pmtimer:       0     STS invalid      unmasked
/sys/firmware/acpi/interrupts/gpe48:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe1B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe76:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe38:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe0B:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe66:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe28:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe56:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe18:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe46:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe74:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe08:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe36:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe64:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe26:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/error:       0
/sys/firmware/acpi/interrupts/gpe54:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe16:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/sci:       3
/sys/firmware/acpi/interrupts/gpe44:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe7E:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe72:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe06:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe34:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe6E:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe62:       0  EN     enabled      unmasked
/sys/firmware/acpi/interrupts/gpe24:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe5E:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe52:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe14:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe4E:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe42:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe7C:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe70:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe04:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe3E:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe32:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe6C:       3  EN     enabled      unmasked
[... 45 lines cut ...]
/sys/firmware/acpi/interrupts/gpe17:       0     STS invalid      unmasked
/sys/firmware/acpi/interrupts/gpe45:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe7F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe73:       0         invalid      unmasked
[... cut ...]
/sys/firmware/acpi/interrupts/gpe71:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe05:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe3F:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe33:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe6D:       0         disabled     unmasked
/sys/firmware/acpi/interrupts/ff_slp_btn:       0         invalid      unmasked
/sys/firmware/acpi/interrupts/gpe61:       0  EN     enabled      unmasked
```

Wer-Wolf, 2023-09-19:
Ok, the interrupt used by the power meter is enabled, but it seems that the power reading is still not updated by the device firmware.

Are there any setup options inside the BIOS regarding the BMC/Monitoring/Intel Node Manager?

I think you need to consult the device manufacturer, since only he knows how the device firmware works. I will perpare some changes to the linux driver so that an error is returned if the power meter returns 0xffffffff.

Wer-Wolf, 2023-09-24:
I prepared the necessary changes: [acpi-power-meter.zip](https://github.com/lm-sensors/lm-sensors/files/12709791/acpi-power-meter.zip).

Could you load the kernel module by executing `make` and then `sudo insmod acpi_power_meter.ko` after unloading the old module with `sudo modprobe -r acpi_power_meter`? Does `sensors` now show an error instead of those 4,29 MW?

urbinek, 2023-09-27:
@Wer-Wolf is that typo or i am doing something wrong?
```Bash
# make
make -C /lib/modules/`uname -r`/build M=`pwd` modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-513.el8.x86_64'
  CC [M]  /root/lm-sensors-git-460/acpi-power-meter/acpi_power_meter.o
/root/lm-sensors-git-460/acpi-power-meter/acpi_power_meter.c: In function 'read_domain_devices':
/root/lm-sensors-git-460/acpi-power-meter/acpi_power_meter.c:605:9: error: implicit declaration of function 'acpi_get_acpi_dev'; did you mean 'acpi_get_pci_dev'? [-Werror=implicit-function-declaration]
   obj = acpi_get_acpi_dev(element->reference.handle);
         ^~~~~~~~~~~~~~~~~
         acpi_get_pci_dev
/root/lm-sensors-git-460/acpi-power-meter/acpi_power_meter.c:605:7: warning: assignment to 'struct acpi_device *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
   obj = acpi_get_acpi_dev(element->reference.handle);
       ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:324: /root/lm-sensors-git-460/acpi-power-meter/acpi_power_meter.o] Error 1
make[1]: *** [Makefile:1619: _module_/root/lm-sensors-git-460/acpi-power-meter] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-513.el8.x86_64'
make: *** [Makefile:5: all] Error 2
```

Wer-Wolf, 2023-09-27:
Could you use a more recent kernel?

urbinek, 2023-09-27:
I am afraid that 4.18 is all i can get now. I am planning to upgrade my setup but i don't have enough time for that.

As for issue itself, I've contacted with SuperMicro support and it seems that, while board have powe meter, values itself are provided via PSU SMBus I2C cable (which i dont have)

 So if there is no i2c input, acpi must be defaulting to some overflowed value

Wer-Wolf, 2023-09-27:
Oh well, this would explain why ACPI defaults to the error value 0xffffffff.

I will try to backport the change for you to test, just give some time.

Wer-Wolf, 2023-11-23:
Sorry for taking so long, i totally forgot about this issue.
Can you still test the fix?
[acpi-power-meter.zip](https://github.com/lm-sensors/lm-sensors/files/13452731/acpi-power-meter.zip)

urbinek, 2023-11-24:
Hi @Wer-Wolf, no problem i also forgot about it :) 

It seems that with this driver value is N/A

```Bash
# make
make -C /lib/modules/`uname -r`/build M=`pwd` modules
make[1]: Wejście do katalogu '/usr/src/kernels/4.18.0-518.el8.x86_64'
  CC [M]  /root/acpi-power-meter/acpi-power-meter/acpi_power_meter.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /root/acpi-power-meter/acpi-power-meter/acpi_power_meter.mod.o
  LD [M]  /root/acpi-power-meter/acpi-power-meter/acpi_power_meter.ko
make[1]: Opuszczenie katalogu '/usr/src/kernels/4.18.0-518.el8.x86_64'
```
```Bash
# ls
acpi_power_meter.c  acpi_power_meter.ko  acpi_power_meter.mod.c  acpi_power_meter.mod.o  acpi_power_meter.o  Makefile  modules.order  Module.symvers
```
```Bash
modprobe -r acpi_power_meter
insmod acpi_power_meter.ko
```
```Bash
# sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +53.0°C  (high = +80.0°C, crit = +90.0°C)
Core 0:        +53.0°C  (high = +80.0°C, crit = +90.0°C)
Core 1:        +52.0°C  (high = +80.0°C, crit = +90.0°C)
Core 2:        +53.0°C  (high = +80.0°C, crit = +90.0°C)
Core 3:        +51.0°C  (high = +80.0°C, crit = +90.0°C)
Core 4:        +52.0°C  (high = +80.0°C, crit = +90.0°C)
Core 5:        +53.0°C  (high = +80.0°C, crit = +90.0°C)
Core 6:        +52.0°C  (high = +80.0°C, crit = +90.0°C)
Core 7:        +52.0°C  (high = +80.0°C, crit = +90.0°C)

power_meter-acpi-0
Adapter: ACPI interface
power1:           N/A  (interval =   1.00 s)
```

Wer-Wolf, 2023-11-24:
Excellent, the modified driver now returns an error instead of returning 4.29 MW.
I will submit the necessary changes upstream, maybe they even get be backported by your distribution.

Wer-Wolf, 2023-12-09:
The fix has been accepted. You might need to ask your distribution maintainer to backport it.

Firefishy, 2026-10-06:
There is also a bug here where ACPI_UINT64_MAX should also be ignored. ACPI specification permits both 32-bit (ACPI_UINT32_MAX) but uses 64-bit (ACPI_UINT64_MAX) for DSDT revision 2+.

Fix 18.45 TW bug ;-)

Wer-Wolf, 2026-10-06:
Good catch, i will prepare the necessary patch.

Firefishy, 2026-10-07:
@Wer-Wolf For reference I see the 18.45 TW bug soon after a reboot in our following systems: HP ProLiant DL360 Gen9, DL380 Gen9,  DL360 Gen10 & DL360 Gen10 Plus systems. I can likely reliably trigger the bug, and I would be happy to test the kernel fix.

Wer-Wolf, 2026-10-07:
I CCed you on the patch email, but i can also provide you with a kernel module if necessary.

Firefishy, 2026-10-07:
For the HPE ProLiant systems I am now confident the misreporting happens when PSUs are hot-swapped. I will try to replicate & test the patch using a similar dev system I currently have physical access to.

Wer-Wolf, 2026-10-19:
The patch was accepted upstream.