lm-sensors/lm-sensors #462: ASUS PRIME B650M-A II

31KM, 2023-09-21:
I just got my hand on that board and tried to setup lm_sensors for it, but nothing was recognized. Digging around I found that it has a Nuvoton NCT6779D chip. `modprobe`ing `nct6775` made the sensors available:
``` 
nct6799-isa-0290
Adapter: ISA adapter
in0:                        1.28 V  (min =  +0.00 V, max =  +1.74 V)
in1:                      1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                      1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                        1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                      512.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                       1.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                       1.12 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                     1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                       1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                      716 RPM  (min =    0 RPM)
fan2:                      620 RPM  (min =    0 RPM)
fan3:                      676 RPM  (min =    0 RPM)
fan4:                      665 RPM  (min =    0 RPM)
fan5:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +39.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +43.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +58.5°C    sensor = thermistor
AUXTIN1:                   +12.0°C    sensor = thermistor
AUXTIN2:                   +19.0°C    sensor = thermistor
AUXTIN3:                   +25.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +43.0°C
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C
PCH_CHIP_TEMP:              +0.0°C
PCH_CPU_TEMP:               +0.0°C
TSI0_TEMP:                 +54.0°C
intrusion0:               ALARM
intrusion1:               ALARM
beep_enable:              disabled
```

Modifying the `Nuvoton NCT/5532D/NCT6779D Super IO Sensors` entry in  `/usr/bin/sensors-detect`  like this
```
		name => "Nuvoton NCT5532D/NCT6779D Super IO Sensors",
		driver => "nct6775",
		devid => 0xD802, # <--
		devid_mask => 0xFFFF, # <--
		logdev => 0x0b,
		features => FEAT_IN | FEAT_FAN | FEAT_TEMP,
```
also made the script recognize it.

krikun98, 2024-01-19:
Hi! https://github.com/lm-sensors/lm-sensors/pull/480 should be a proper fix here.

N0TScrooge, 2025-02-11:
> also made the script recognize it.

@31KM 
tell me, was you able to adjust the correct display of 12V 5V ?