lm-sensors/lm-sensors #463: MSI x670e Carbon Wifi no sensors detected

gokomer, 2023-10-03:
Hello guys. sensors-detect couldn't find any sensors and I've no idea what to do. Any pointers?

```
# sensors-detect version 3.6.0
# System: Micro-Star International Co., Ltd. MS-7D70 [1.0]
# Board: Micro-Star International Co., Ltd. MPG X670E CARBON WIFI (MS-7D70)
# Kernel: 6.5.5-200.fc38.x86_64 x86_64
# Processor: AMD Ryzen 7 7800X3D 8-Core Processor (25/97/2)
```

Wer-Wolf, 2023-10-10:
What is the output of `sensors`?

ffjdm, 2023-10-28:
Here is the output for the same motherboard:
`amdgpu-pci-1400
Adapter: PCI adapter
vddgfx:      755.00 mV 
vddnb:         1.30 V  
edge:         +40.0°C  
PPT:          34.01 W  

nvme-pci-0100
Adapter: PCI adapter
Composite:    +34.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +34.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +43.6°C  
Tccd1:        +30.6°C  
Tccd2:        +32.5°C`

Wer-Wolf, 2023-12-11:
Is the driver from the mainline kernel? Does it load without the force option?

Can you share the output of `acpidump`?

Wer-Wolf, 2023-12-11:
Also please note that sensors-detect does not work with secure boot enabled.

linuxtek-canada, 2023-12-12:
I disabled Secure Boot, and now I can see the module loading properly:

```
$ lsmod | grep nct
nct6687                32768  0
```

Curiously, running `sensors-detect` still returns nothing found.
But running `sensors` is actually reporting all of the values now.

If it's helpful, I've put a gist with the output of all of there commands, plus `acpidump` [here](https://gist.githubusercontent.com/linuxtek-canada/ec3f5a9d6303f2ad8e683511bdaaa75b/raw/1d7e3935c42a23c8a3363e2b879ee1ddd98602b8/gistfile1.txt).