lm-sensors/lm-sensors #468: Renaming certain sensors do nothing

HackMeMan, 2023-11-09:
On my ThinkPad-T440p, the chip `thinkpad-isa-0000` shows many values, including values from `temp3` to `temp8,` all showing 0 degrees, which can easily be ignored. However, a value named `GPU` on this same chip also always show 0 degrees, cannot be ignored in a config file. This is also the case for other values such as `Core 0`,`Core 1`, and `Package ID 0`, all can't be renamed, only this time they return a syntax error. This is probably because the names include spaces, so it's understandable but should be fixed. The `GPU` case, on the other hand, I do not understand.
Here's my config file:
```
chip "thinkpad-isa-0000"
  ignore GPU
  ignore temp3
  ignore temp4
  ignore temp5
  ignore temp6
  ignore temp7
  ignore temp8
  label fan1 Fan
chip "coretemp-isa-0000"
  label "Core 0" "Core 1"
chip "acpitz-acpi-0"
  label temp1 "PSU"
chip "BAT0-acpi-0"
  label in0 Battery
```

Here's the `sensors` output:
```
thinkpad-isa-0000
Adapter: ISA adapter
Fan:            0 RPM
CPU:          +48.0°C  
GPU:           +0.0°C  

BAT0-acpi-0
Adapter: ACPI interface
Battery:      11.69 V  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +50.0°C  (high = +84.0°C, crit = +100.0°C)
Core 0:        +50.0°C  (high = +84.0°C, crit = +100.0°C)
Core 1:        +47.0°C  (high = +84.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
PSU:          +48.0°C  (crit = +200.0°C)
```

Wer-Wolf, 2023-11-14:
Keep in mind that "Core X" is already an label of the temperature channel, so you have to use the original "tempX" value when referencing that channel.

HackMeMan, 2023-11-23:
That's great, but the issue for GPU remains

Wer-Wolf, 2023-11-23:
The same goes for the channel with the label "GPU".

dekeonus, 2024-08-25:
> That's great, but the issue for GPU remains

I realise this response is quite late, but you should be able to extract the actual name of **GPU** using something like the following:

`sensors -u thinkpad-isa-0000 coretemp-isa-0000` will get you the actual "feature names" 

output on a nearby system for a coretemp device:
```shellsession
$ sensors -u coretemp-isa-0000
coretemp-isa-0000
Adapter: ISA adapter
Core 0:
  temp2_input: 30.000
  temp2_crit: 100.000
  temp2_crit_alarm: 0.000
Core 1:
  temp3_input: 31.000
  temp3_crit: 100.000
  temp3_crit_alarm: 0.000
Core 2:
  temp4_input: 38.000
  temp4_crit: 100.000
  temp4_crit_alarm: 0.000
Core 3:
  temp5_input: 37.000
  temp5_crit: 100.000
  temp5_crit_alarm: 0.000

```
I  can see that the temperature feature names are temp2_??? to temp5_???
So if I use this sensors.conf snippet
```shell
chip "coretemp-*"
    # No temp1 on this cpu
    #label temp1 "does not exist"
    label temp2 "t.cpu_core1" # "Core 0"
    label temp3 "t.cpu_core2" # "Core 1"
    label temp4 "t.cpu_core3" # "Core 2"
    label temp5 "t.cpu_core4" # "Core 3"
```
I get this output
```shellsession
$ sensors -u coretemp-isa-0000
coretemp-isa-0000
Adapter: ISA adapter
t.cpu_core1:
  temp2_input: 30.000
  temp2_crit: 100.000
  temp2_crit_alarm: 0.000
t.cpu_core2:
  temp3_input: 31.000
  temp3_crit: 100.000
  temp3_crit_alarm: 0.000
t.cpu_core3:
  temp4_input: 38.000
  temp4_crit: 100.000
  temp4_crit_alarm: 0.000
t.cpu_core4:
  temp5_input: 37.000
  temp5_crit: 100.000
  temp5_crit_alarm: 0.000

$ sensors coretemp-isa-0000
coretemp-isa-0000
Adapter: ISA adapter
t.cpu_core1:  +30.0°C  (crit = +100.0°C)
t.cpu_core2:  +31.0°C  (crit = +100.0°C)
t.cpu_core3:  +38.0°C  (crit = +100.0°C)
t.cpu_core4:  +37.0°C  (crit = +100.0°C)

```